Proxmark/proxmark3 #39: What is this iclass_dump.bin file?

ikarus23, 2015-01-09:
A binary file in a source folder?
What is its purpose?

Commit: 6f101995b633112d092b4f61b9fb2345f85ba353
File: https://github.com/Proxmark/proxmark3/blob/master/client/loclass/iclass_dump.bin

holiman, 2015-01-09:
It's a file which contains the dump from an elite system. It is used for 'hf iclass loclass t', a self-test which verifies that the loclass crypto algos are intact/correct.

closed 2015-01-09 by holiman

ikarus23, 2015-01-09:
Thanks for clearing that up!

OT: Are the traces (https://github.com/Proxmark/proxmark3/tree/master/traces)  also for self testing? Or just for "scientific" analysis?

holiman, 2015-01-09:
My impression is that they are for testing, but not as in "verify functionality via self-test", but instead example dumps of various tags, to be used in lack of real tags when doing development.