Proxmark/proxmark3 #42: Erros while trying to falsh my proxmark

ikarus23, 2015-01-10:
Hi,
I have some trouble flashing my proxmark.

My proxmark setup:
```
proxmark3> hw version
#db# Prox/RFID mark3 RFID instrument                 
#db# bootrom: master/v1.0.0-13-ge17437f-suspect 2014-06-26 11:25:03                 
#db# os: master/v1.0.0-13-ge17437f-suspect 2014-06-26 11:25:04                 
#db# HF FPGA image built on 2014/ 6/19 at 21:26: 2                 
uC: AT91SAM7S512 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory   
```
My system setup:
Linux (64 Bit)
```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (devkitARM release 43) 4.9.2
```

There are different errors and I can't really reproduce them exactly.

```
./client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0001497e->0x0001497e) [R  ] @0xb4
1: V 0x00118000 P 0x00118000 (0x000186f9->0x000186f9) [R X] @0x14a38
2: V 0x00200000 P 0x001306f9 (0x000029f8->0x000029f8) [RWX] @0x2d138
Note: Extending previous segment from 0x186f9 to 0x1b0f1 bytes

Waiting for Proxmark to appear on /dev/ttyACM0...... Found.
Error: Unknown Proxmark mode
```

or 

```
./client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0001497e->0x0001497e) [R  ] @0xb4
1: V 0x00118000 P 0x00118000 (0x000186f9->0x000186f9) [R X] @0x14a38
2: V 0x00200000 P 0x001306f9 (0x000029f8->0x000029f8) [RWX] @0x2d138
Note: Extending previous segment from 0x186f9 to 0x1b0f1 bytes

Waiting for Proxmark to appear on /dev/ttyACM0. Found.
Error: Unknown Proxmark mode
*** stack smashing detected ***: ./client/flasher terminated
[1]    8577 abort (core dumped)  ./client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf
```

or 

```
./client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000b30->0x00000b30) [RWX] @0x298

Waiting for Proxmark to appear on /dev/ttyACM0........................ Found.
Error: Unknown Proxmark mode
*** stack smashing detected ***: ./client/flasher terminated
[1]    8304 abort (core dumped)  ./client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf
```

or 

```
./client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000b30->0x00000b30) [RWX] @0x298

Waiting for Proxmark to appear on /dev/ttyACM0. Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to reappear on /dev/ttyACM0............... Found.
Note: Your bootloader does not understand the new START_FLASH command
      It is recommended that you update your bootloader

*** stack smashing detected ***: ./client/flasher terminated
[1]    8362 abort (core dumped)  ./client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf
```

Am I doing something wrong?
Whats up with the "stack smashing detected"? Is there something wrong with the code?

Thanks for any help!

holiman, 2015-01-15:
Try this: 

	> gdb --args ./client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf
	GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
	Copyright (C) 2014 Free Software Foundation, Inc.
	License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
	This is free software: you are free to change and redistribute it.
	There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
	and "show warranty" for details.
	This GDB was configured as "x86_64-linux-gnu".
	Type "show configuration" for configuration details.
	For bug reporting instructions, please see:
	<http://www.gnu.org/software/gdb/bugs/>.
	Find the GDB manual and other documentation resources online at:
	<http://www.gnu.org/software/gdb/documentation/>.
	For help, type "help".
	Type "apropos word" to search for commands related to "word"...
	Reading symbols from ./proxmark3...done.
	(gdb) run

That it, first load it with appropriate argumentes, then `run` to start execution. This way, you should able to see the exact stack trace of the stack smash.

ikarus23, 2015-01-28:
Today I suddenly realized what is wrong...
I recently killed my old Linux installation and setup a new one. With the new one I got a new `modem-manager` (without a blacklist rule). Silly me for not realizing this earlier. I followed the old Wiki and created a `/etc/udev/rules.d/77-mm-usb-device-blacklist.rules` file with the following content
```python
# proxmark3 - http://www.proxmark.org/
ATTRS{idVendor}=="2d2d" ATTRS{idProduct}=="504d", ENV{ID_MM_DEVICE_IGNORE}="1"
```
Now everything is back to normal.

closed 2015-01-28 by ikarus23

holiman, 2015-01-28:
You don't need the wiki: [it's included](https://github.com/Proxmark/proxmark3/blob/master/driver/77-mm-usb-device-blacklist.rules) . Also, I made a better version which isn't in main yet, putting the pm3 at `/dev/pm3-1` instead of `/dev/ttyACMX` [here](https://github.com/Proxmark/proxmark3/commit/3fd26a683ddc9e166392f722c23ab21fb618ce08)