Proxmark/proxmark3 #43: bug in 14443a anti-collision code

jnferguson, 2015-01-12:
When a collision occurs while operating via the command "hf 14a read" or a couple of other code paths, the iso14443a_select_card() function enters the following code path (cleaned up for line wraps/etc):

	if (Demod.collisionPos) {			
		memset(uid_resp, 0, 4);
		uint16_t uid_resp_bits = 0;
		uint16_t collision_answer_offset = 0;
		// anti-collision-loop:
		while (Demod.collisionPos) {
			Dbprintf("Multiple tags detected. Collision after Bit %d", Demod.collisionPos);
			for (uint16_t i = collision_answer_offset; i < Demod.collisionPos; i++, uid_resp_bits++) {	
				uint16_t UIDbit = (resp[i/8] >> (i % 8)) & 0x01;
				uid_resp[uid_resp_bits & 0xf8] |= UIDbit << (uid_resp_bits % 8);
			}
			uid_resp[uid_resp_bits/8] |= 1 << (uid_resp_bits % 8);					
			uid_resp_bits++;
			// construct anticollosion command:
			sel_uid[1] = ((2 + uid_resp_bits/8) << 4) | (uid_resp_bits & 0x07);  	
			for (uint16_t i = 0; i <= uid_resp_bits/8; i++) {
				sel_uid[2+i] = uid_resp[i];
			}
			collision_answer_offset = uid_resp_bits%8;
			ReaderTransmitBits(sel_uid, 16 + uid_resp_bits, NULL);
			if (!ReaderReceiveOffset(resp, collision_answer_offset)) return 0;
		}
		// finally, add the last bits and BCC of the UID
		for (uint16_t i = collision_answer_offset; i < (Demod.len-1)*8; i++, uid_resp_bits++) {
			uint16_t UIDbit = (resp[i/8] >> (i%8)) & 0x01;
			uid_resp[uid_resp_bits/8] |= UIDbit << (uid_resp_bits % 8);
		}

	} else { [...]

This results in an invalid response from the proxmark, for instance, consider:
two cards with UIDs of 0x04 0x62 0x1f 0x1a 0x94 0x35 0x80 and 0x04 0x42 0x23 0xd2 0x0b 0x35 0x80 respectively, and a collision occurs at bit 21, eliciting the following response: 0x88 0x04 0x62 0x3f 0xfd, then the proxmark responds with: 0x93 0x46 0x88 0x00 0x20 which is incorrect.

The correct response is : 0x93 0x46 0x88 0x04 0x22

This is due to the line:

uid_resp[uid_resp_bits & 0xf8] |= UIDbit << (uid_resp_bits % 8);

Which not only results in an invalid response, but also corrupts stack memory (id est with a collision at bit 21 we will eventually write to uid_resp[16], which is actually 4 bytes in length).This can be fixed by modifying the line to read as follows:

uid_resp[i / 8]  |= UIDbit << (i % 8);

With this line changed, instead of hanging after the debug output informing us of the collision due to the UID not corresponding to the cards UID, we successfully complete the transaction:

[image]

holiman, 2015-01-12:
Wow, now that's a great bug report! Thanks a lot. Care to submit a pull-request? Otherwise, I'll fix it in the next few days.

jnferguson, 2015-01-12:
My local copy of the repository is all messed up; I'm in the process of creating a reader/sniffer that will probably merge additions into both the firmware and client application; when I get to a point where the code is presentable I'll definitely contribute back. cheers/thanks for the great product.

pwpiwi, 2015-01-13:
Well spotted. But it still should be `uid_resp_bits`. I.e. the fixed line should be
`uid_resp[uid_resp_bits / 8] |= UIDbit << (uid_resp_bits % 8);`

jnferguson, 2015-01-13:
Fair enough/correct; it would be nice to get rid of the state as much as possible given that multiple collisions could occur and uid_resp_bits never resets and corresponds to an index into a 4 byte buffer. I'm actually not sure what happens with triple length UIDs if a collision occurs at >40 bits, 14443a-3 seems to assume that this never happens.

Even so, consider a collision at bit 39, I guess maybe the right overall fix is to switch to uid_resp_bits and then maybe uid_resp as large as the longest possible UID response, which i *think* is going to be 5 bytes long, but as I said, I have no idea what happens with triple length UIDs as even the authors of the standard seem to not have considered instances where collisions occur at >40 bits.

I need to drink coffee, this isnt coming out right, but now i want to see how commercial readers deal with bizarre sets of collisions that in practice shouldnt happen (collision at bit 39 followed a collision at 21, followed by a collision at... an attack you could implement in a device like a proxmark :) )

pwpiwi, 2015-01-13:
It isn't that bad. 
- Multiple collisions can be handled (this is why uid_resp_bits only resets before and not within the while loop). But I think this is only theoretical: I was never able to see more than 2 cards with the PM3. 
- The UIDs are transferred and collision is handled in 4 Byte groups. Collision therefore can only appear at UID bit positions 1 to 32 and the 4 Byte buffer is enough - even for 7 Byte and 10 Byte UIDs.

jnferguson, 2015-01-14:
I am inclined to agree with you most principally because its a development device and because that a collision can in fact go backwards won't ever occur in natural circumstances.

I also admit I haven't read the demodulation code to discern how its working or if it pre-parses things to ensure that you actually only get a 4 byte buffer, or other similar such bizarre circumstances.

holiman, 2015-01-15:
Fixed as of 758f1fd

closed 2015-01-15 by holiman

jnferguson, 2019-01-22:
Hi,

I'm sitting here looking at the code and I apologize I am less familiar with it than I was in 2015, but is this fix correct? Should it still entail a bitwise and to keep the index in bounds?

uid_resp[uid_resp_bits/8] |= UIDbit << (i % 8);
vs
uid_resp[(uid_resp_bits/8) % 4] |= UIDbit << (i % 8);

Briefly reading through it, it appears that successive collisions could still cause it to go out of bounds? I didn't pay much attention to it at the time, its a proxmark and so relatively low-risk, but in using it to describe something about something else this occurred to me.

pwpiwi, 2019-01-30:
> I'm sitting here looking at the code and I apologize I am less familiar with it than I was in 2015

Me too. :smile:

Your fix should be enough to prevent an overflow. However, it will not produce correct results (well, what is a correct result when someone tries to fool the reader with collisions going backwards...). Coming back to your previous comments, it would be best to get rid of the constantly growing `uid_resp_bits` and make the while loop stateless.