Proxmark/proxmark3 #47: Qt version detection in client\Makefile

iceman1001, 2015-01-22:
The Qt used is normally the old 4.6.2,   but if you, as I,  tries with the latest Qt 5.3.1  you will run into some compiling issues in the client\makefile  

After I´ve been looking into the Makefile magic, I think this is a possible solution..
When running the command  "/qt/bin/moc -v"   you get the current version of Qt.
The returned string is very different from Qt4 and Qt5.  The fix tries to find "MOC 5" in the returned string if found it assumes Qt5.   Otherwise it just thinks its old Qt4. 

# QT version,  4 or 5
qtplatform = $(shell $(MOC) -v)
ifeq (, $(findstring MOC 5,$(qtplatform)))
   CXXFLAGS = -I$(QTDIR)/include -I$(QTDIR)/include/QtCore -I$(QTDIR)/include/QtGui -I$(QTDIR)/include/QtWidgets  -I/mingw/include
   QTLDLIBS = -L$(QTDIR)/lib  -lQt5Core -lQt5Gui -lQt5Widgets 
else
   CXXFLAGS = -I$(QTDIR)/include -I$(QTDIR)/include/QtCore -I$(QTDIR)/include/QtGui
   QTLDLIBS = -L$(QTDIR)/lib -lQtCore4 -lQtGui4
endif

pwpiwi, 2015-02-13:
Doesn't work for me (using MinGW on Win8.1). `MOC`isn't set and the moc executable is not in my PATH.

Why not looking if the Qt5Core library exists? If yes, use it, else use QtCore4?

By the way: what are the advantages of using Qt5?

iceman1001, 2015-02-13:
How do you manage to compile the client if you don't use the moc?  I'm open for better ways of detecting the installed version of Qt.  
I suppose there is no obvious reasons for using Qt5, other than if you download it today, thats the version the Qt team is pushing out.   The Pm3 compile environment consists of very outdated components.  It would be nice to have it up-to-date.

pwpiwi, 2015-03-10:
Please try my pull request #75

iceman1001, 2015-03-11:
Great! It looks like it works.  Or at least it works for me in my mingw-environment.

closed 2015-03-13 by pwpiwi