Proxmark/proxmark3 #62: Reworking how the graphing functions

holiman, 2015-02-09:
I am currently rewriting the graph framework. I want to provide a better UI for experimenting with signals. 

Here's how it looks right now:
[image]

What you can't see here, is that if you pull the slider to the left (increasing 'window'), the red graph updates in real time. The same thing with dir threshold:

[image]

Once you're happy with the settings, you can press apply, and the red signal is copied into the main signal buffer (graphbuffer):
[image]

This does not really affect the console-commands, they work as usual. I've had to restructure them a bit though; e.g. dirthreshold and autocorr look like this:

	int CmdDirectionalThreshold(const char *Cmd)
	{
		int8_t upThres = param_get8(Cmd, 0);
		int8_t downThres = param_get8(Cmd, 1);

		printf("Applying Up Threshold: %d, Down Threshold: %d\n", upThres, downThres);

		directionalThreshold(GraphBuffer, GraphBuffer,GraphTraceLen, upThres, downThres);
		RepaintGraphWindow();
		return 0;
	}

That is, they parse the commands, then call an "api"-function which does the actual data operation. These data-operations all have the signature (const int* in, const int*out, size_t len, ... parameters..). That is, they take one read-buffer and one write-buffer. In the console-case, in and out are the same. 

I think there are probably a few more things which could fit into this (changes from analog -> analog which depends on parameters which the user may want to fiddle with). 
So that's for analog signals -> analog signals. 

I have some thoughts on also doing another UI for functions which map analog->digital, i.e; decoders. The UI would show the digital out, represented both in bitstream-form, and hex-form (in four ways, shifted one bit each), and perhaps ascii form. Perhaps highlighting potential parity bits. And then the user could slide stuff like clock or starting point or whatever, and in real time see what makes the most sense. 

This is still work in progress, but if anyone wants to tag along, feel free. 

And I do think we're approaching a major version number increase soon.

holiman, 2015-02-09:
Oh, the "Sticky" button was supposed to just snapshot the red signal (turn it grey), so the user could fiddle on with other signals and it would remain. I haven't implemented it, as it's a bit finicky to dynamically keep track of the number of signal buffers, and I'm not sure it's worth it. The usecase may not be that relevant.

iceman1001, 2015-02-10:
I really like this idea of yours.  This will be such an enchancement to the Pm3.

When it comes to the  Analog -> Digital,  the new remade api from @marshmellow42  will come in hand.
To be able to switch between different decoders real time. 

And I also agree with the major version number increase.  A new release (v2.0.0) maybe?

holiman, 2015-04-08:
I've put this on hold a bit, I'll take another stab once the LF-parts stabilize from the major overhauls that @marshmellow42 is doing...

marshmellow42, 2017-04-12:
taking this where holiman left off and adjusting it a bit to include the demod yields:
[image]
still working out some bugs.  if anyone knows QT Designer well assistance would be welcomed!

pwpiwi, 2017-04-12:
I was already wondering when you finally would arrive here. :grin:

marshmellow42, 2017-04-12:
Heh, every demod needs adjusting...  And I have no clue how to work QT..  but I might as well learn...

marshmellow42, 2017-04-12:
It is up on my fork in the graphwork branch.  Should I attempt to merge it into the Master's graphwork branch?  Too many bugs for the master at this point...

marshmellow42, 2017-04-12:
It currently should work with all data rawdemod cmds...  Working on the rest still.

marshmellow42, 2017-04-13:
I reused holimans code from that branch as a starting point.  There are a few bugs in his code (like you couldn't reopen the plot window if you closed it) and now a few in my additions that I'm trying to work out (like the main app is unable to close the overlays form when it exits).

pwpiwi, 2017-04-13:
Then I would agree to your proposal and merge to proxmark3/graphwork first. Even without bugs it will be difficult enough to merge with master later.

marshmellow42, 2017-04-13:
that branch is so old, can i just blow it away and pull in my new code (it contains all holiman's code with new updates to work with current master)?

pwpiwi, 2017-04-13:
Sure. I didn't think that you are closer to master than holiman's graphwork.

marshmellow42, 2017-04-13:
for now i created a graphwork2 branch and merged my fork/branch into it.  (it is current with master + graph stuff)
i'm struggling to identify why it won't quit the overlays form when the proxmark application closes... (it hangs instead)  any help would be appreciated!

iceman1001, 2017-04-13:
Its just as well to drop the old graphwork branch and take marshmellows as basis for a new one.

QT is another language to learn,  but is there some other UI dev we can use instead of it?  It needs to crosscompile..  I guess thats why its QT,.   Are there some QT IDE?

marshmellow42, 2017-04-13:
I agree, but I left the old one for now to still compare and see if I missed something causing the hang issue...  Once that bug is fixed I will delete it

iceman1001, 2017-04-13:
I think you will need to get a way of getting the active qt-window-handles,  and kill them before pm3 client exits.  QT must have a way of getting refs to window handles.   Maybe we need to save them in an array, when created.  Which we can use to close each handle at the end, if they exists.

marshmellow42, 2017-04-13:
i think all qt windows should close when the `plotapp->quit();` is called

iceman1001, 2017-04-13:
well,  does it?  or does your trials show that it hangs?

marshmellow42, 2017-04-13:
that is the thing, it appears to work on the old branch but not my new one...

iceman1001, 2017-04-13:
I got hangs on Holimans old code too.  Dont use his unfinished code as baseline for yours.

pwpiwi, 2017-04-14:
> i think all qt windows should close when the `plotapp->quit();` is called

This is what I would expect as well. The issue could be related to issue #275.

pwpiwi, 2017-04-14:
> I got hangs on Holimans old coide too. Dont use his unfinished code as baseline for yours.

Why not? This is what we do since years. :smile:

marshmellow42, 2017-04-14:
i believe i have fixed the issues. ? maybe?   to be tested on other platforms...

marshmellow42, 2017-04-14:
i just need to figure out how to add uic ui/overlays.ui > ui/ui_overlays.h to the makefile and it will compile and work on kali/other qt versions as well.

marshmellow42, 2017-04-14:
still have a segmentation fault error after exit.  (but that existed before i think...)

marshmellow42, 2017-04-14:
i tried to add the ui_overlays.h uic to the make file and it compiles on windows, but doesn't on kali.  it appears it doesn't happen soon enough as it complains the file doesn't exist yet.  what did i miss?

pwpiwi, 2017-04-15:
Cannot confirm. Compiles fine on my Kali. Do you have Qt4 or Qt5 with Kali?

marshmellow42, 2017-04-15:
Qt 4.  Strange.   I'll do some more testing

pwpiwi, 2017-04-15:
Qt4 doesn't have a separate QtWidgets module. Use QtGui instead of QtWidgets in the #includes.

marshmellow42, 2017-04-15:
That is what uic function does automatically.  It re compiles the .h for the correct version.  Right now the ui_overlays.h isn't getting made in my Kali.

marshmellow42, 2017-04-15:
proxguiqt.h:21:28: fatal error: ui/ui_overlays.h: No such file or directory
 #include "ui/ui_overlays.h"
i get this on ubuntu 14 and kali. (both qt4.)  as it appears the .h file is getting cleaned as i intend but not re built before it is needed.

marshmellow42, 2017-04-15:
i think i need to do something resembling the below but am not sure how
"You need to list the targets that need to be built as prerequisites for the targets that need those files to exist for their own compilation/etc"

marshmellow42, 2017-04-15:
ok, so now i'm really confused...  in kali and ubuntu to test in the client directory i ran `make ui/ui_overlays.h` and it was successful . i tried make clean && make all (in client directory) and it failed. same error.
i ran the `make ui/ui_overlays.h` again and then ran a `make` and all was successful.
then i tried `make clean && make all` and now it works everytime....
??confused??

marshmellow42, 2017-04-15:
i can recause the error by reloading my git environment with:
`git reset`
`git checkout .`
`git clean -fdx`

pwpiwi, 2017-04-15:
Is it better if you add `#include "ui/ui_overlays.h"` to proxguiqt.cpp?

marshmellow42, 2017-04-15:
no change.

marshmellow42, 2017-04-15:
also tried including it in proxgui.cpp (where original error starts as proxgui.cpp includes proxguiqt.h)
but no change
only way i get get around it so far is the procedure i outlined above (make ui_overlays.h - make - make clean && make all)  but once i do that procedure it always works until i reset the code from git. 
i'm going to try loading up the qt4 version of the ui_overlays.h to git and see if that makes a difference...

marshmellow42, 2017-04-15:
so i give up on the uic make...
i adjusted the ui/ui_overlays.h to be compatible with supposedly qt4 and qt5.
it works fine on all my environments now.

holiman, 2017-04-15:
Hey all!

Nice to see the activity on proxmark! I haven't kept up to date, but check in from time to time. Regarding the graphing, I find qt designer to be quite easy to use, and used qt creator as a general purpose c IDE for coding proxmark stuff on. I may be able to help out with qt designer stuff. 

Regarding segfaults and hangs, you need to prevent QT from gc:ing your windows / components, which you do by maintaining references. Put them in a member or vector or something, but put them somewhere, otherwise bad things happen. 

The big problem with graphing, as I recall it, is that it's difficult to make a 'general' input dialog for various parameters that you may want to fiddle with. However, it's easier to make the console still being the input, and the graph is just for show. But then you miss the nice thing which is to turn a knob or drag a dial and see the output change immediately. So the best would be to generate inputs based on the number of parameters. 

Which branch contains the relevant code for graphing right now?

marshmellow42, 2017-04-15:
graphwork2 is the current working branch.  i think we are very close to being ready to merge it with the master.  have a look and fix any of my mistakes ;)

marshmellow42, 2017-04-15:
i was about to remove graphwork with your permission.

marshmellow42, 2017-04-15:
the biggest changes i made were:
any lf demod will apply a blue demod line over the graph where the demod data was obtained (along with the binary output in blue between the lines)

on the overlays form:
I removed askdemod tab and added askedge detection.
i adjusted the "Sticky" button to be "Restore" to undo any operation on the graph.

i adjusted some qt code to allow hiding and showing the graph and overlays form from the pm3 client data commands.

and i think i fixed the errors given when the client tried to exit when the overlays form was shown.

there is still a segment fault error when the client exits but that may be unrelated to the new code. (preexisting?)

marshmellow42, 2017-04-15:
i also found for qt version compatibility that the ui_overlays.h has to be either made by uic during the make  or has to be built on QT4 and manually adjusted to remove the `QtGui/` in front of the #includes and remove all occurrences of `, QApplication::UnicodeUTF8`. (QT4 only)

marshmellow42, 2017-04-15:
can you build the pm3 code from within Qt Creator?

marshmellow42, 2017-04-16:
found a couple grid bugs and patched...

was thinking we should add a button to toggle overlays on and off

holiman, 2017-04-16:
> i was about to remove graphwork with your permission.

I won't mind, I've been away too long to have objections :)

> i adjusted the "Sticky" button to be "Restore" to undo any operation on the graph.

Regarding 'sticky' versus 'restore', my idea was that you would fiddle knobs until it looks good, and then 'lock in' on that, and go to the next operation, and do the same thing again. So you could go from a noisy signal, to a digital signal, to a decoded signal. 

If you do the instant-wipe, you'll lose the original signal each time you fiddle the knob, and if you have a slider with 10 steps, that you slide up 1,2,3, then you will go through the graphs
`f(signal,1), f( f(signal,1),2) , f( f( f(signal,1),2),3) ` instead of `f(1,signal), f(2,signal), f(3,signal)`

But maybe I misread you. The 'revert'-version works well with commandline, but I think less well from a graphical UI perspective. 

> can you build the pm3 code from within Qt Creator?

I never did, no. I like QT creator, but not the build-system that keeps trying to overwrite the makefiles and whatnot.

marshmellow42, 2017-04-17:
Regarding the sticky, I'd agree with that model.  but we need more commands made and included to make that useful. 

For now I think being able to visualize what a commands setting will do then applying what looks good, and being able to run a demod cmd from the client (for now) and see the results on the graph and if it failed be able to go back and try again, is a fair starting point.

It would be nice if we had various filters we could apply to really clean up the wave.

marshmellow42, 2017-04-19:
you can learn a lot with pictures and the new graph could help many learn simple modulation demodulation.
psk1 trace demodulated with client commands and graphed
[image]
and ask with askedge (red) overlaid and client demod (em410x) overlaid:
[image]
(ask edge slider setting):
[image]

iceman1001, 2017-04-19:
This is an huge step forward for PM3 users! It will be great learning tool and I belive a start for something new. Great work @marshmellow42.

closed 2017-04-19 by pwpiwi