Proxmark/proxmark3 #65: hf list 14a  - shows always parity error for all bytes.

iceman1001, 2015-02-10:
I tried the latest sourcecode and the "hf list 14a" always gives me a "!"  (ie: parity error)  for every byte listed. I tried a couple of different tags, its the same for those tags aswell.

     Start |       End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |

-----------|-----------|-----|-----------------------------------------------------------------|-----|--------------------|

         0 |       992 | Rdr | 52!                                                             |     | WUPA
      2228 |      4596 | Tag | 44! 00!                                                         |     |
      7040 |      9504 | Rdr | 93! 20!                                                         |     | ANTICOLL
     10676 |     16500 | Tag | 88! 04! 8f! 48! 4b!                                             |     |
     18688 |     29216 | Rdr | 93! 70! 88! 04! 8f! 48! 4b! 5f! 49!                             |     | SELECT_UID
     30388 |     33908 | Tag | 04! da! 17!                                                     |     |
     35200 |     37664 | Rdr | 95! 20!                                                         |     | ANTICOLL-2
     38836 |     44724 | Tag | 9a! f8! 38! 80! da!                                             |     |
     46848 |     57312 | Rdr | 95! 70! 9a! f8! 38! 80! da! 68! d4!                             |     | ANTICOLL-2
     58548 |     62132 | Tag | 00! fe! 51!                                                     |     |
     65664 |     70432 | Rdr | 30! 00! 02! a8!                                                 |     | READBLOCK(0)
     71604 |     92404 | Tag | 04! 8f! 48! 4b! 9a! f8! 38! 80! da! 48! 00! 00! 00! 00! 00! 00! |     |
           |           |     | 28! c9!                                                         |     |
    105088 |    109856 | Rdr | 50! 00! 57! cd!                                                 |     | HALT

--------
pm3 --> hw ver
#db# Prox/RFID mark3 RFID instrument
#db# bootrom: /-suspect 2015-01-28 19:34:43
#db# os: /-suspect 2015-02-10 20:54:12
#db# HF FPGA image built on 2015/01/15 at 12:19:06

iceman1001, 2015-02-10:
I guess this exclamation mark is the reason. It should be removed.
https://github.com/Proxmark/proxmark3/blob/master/client/cmdhf.c#L478

holiman, 2015-02-10:
Ha!

holiman, 2015-02-10:
My bad, I'll fix

closed 2015-02-12 by holiman