Proxmark/proxmark3 #69: OS X - Unable to flash - Waiting for Proxmark to reappear on USB......

richiejenkins, 2015-02-19:
Hi All,

I have just been given a Proxmark Device.

I'm following the steps for OS X from the GitHub wiki page.

I'm currently stuck when I try update the boot loader

    richtea -> ./flasher -b ../../bootrom/obj/bootrom.elf
    Loading ELF file '../../bootrom/obj/bootrom.elf'...
    Loading usable ELF segments:
    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
    1: V 0x00200000 P 0x00100200 (0x00000b60->0x00000b60) [RWX] @0x298

    Waiting for Proxmark to appear on USB...
    Connected units:
    1. SN: ChangeMe [020/008]
     Found.
     Entering bootloader...
     (Press and release the button only to abort)
     Waiting for Proxmark to reappear on USB..........................................................................................................................
    
I have tried restarting after installing dummy driver.

I am holding the button down for 5 seconds when inserting the Proxmark.

My Proxmark is getting reported as:

ProxMark-3 RFID Instrument:

          Product ID: 0x4b8f
          Vendor ID: 0x9ac4
          Version: 0.01
          Serial Number: ChangeMe
          Speed: Up to 12 Mb/sec
          Manufacturer: J. Westhues
          Location ID: 0x14100000 / 19
          Current Available (mA): 500
          Current Required (mA): 500

Any help would be appreciated.

marshmellow42, 2015-02-19:
Try keeping the button held until it says it is done.

richiejenkins, 2015-02-19:
@marshmellow42 that doesn't work.

iceman1001, 2015-04-06:
Don't forget the blacklist part if you are running on Linux/Mac.  I think that part is still on the googlecode site. 

If your problem has been resolved, please close this issue.

richiejenkins, 2015-04-06:
@iceman1001 I installed the dummy kernel driver but it has no effect on Mac OS X (Yosemite).

iceman1001, 2015-04-06:
Did you:   https://code.google.com/p/proxmark3/wiki/Linux
Read section on "proxmark driver installation" part   regarding:  77-mm-usb-device-blacklist.rules

dmarafetti, 2016-01-19:
Holding the PM button try this way:
sudo ./flasher /dev/cu.usbmodemXXXX -b ../bootrom/obj/bootrom.elf

iceman1001, 2016-02-19:
time to close?

closed 2016-10-06 by iceman1001

masterix21, 2017-10-13:
I dears,

i have the same issue on OSX. This is the output of "hw version":

proxmark3> hw version
[[[ Cached information ]]]

Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-94-g77aecdd-suspect 2017-10-12 09:01:21
os: master/v1.1.0-974-gb62b5e2-suspect 2017-05-23 17:51:37
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2015/11/ 2 at  9: 8: 8

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 191326 bytes (36). Free: 332962 bytes (64).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

I have updated the bootrom, but after i can't flash fullimage. Any idea?