Proxmark/proxmark3 #70: ISO14443 - Can't emulate tags of greater than ~56 bytes.

peterfillmore, 2015-02-23:
Hello - I'm still having issues with emulating ISO14443a tags of lengths greater than 56 bytes.
I've tried increasing the default size of the ToSend buffer - which did help to extend the supported length - but am hitting an upper limit that i can't seem to locate why.

My current running code is forked in  https://github.com/peterfillmore/proxmark3 . This adds support for reading and emulating ISO14443a based NFC payment cards. This is 'experimental' code for now, but works for reading EMV cards at the moment.

I've checked the parity outputs for long tags, which indicates it is correctly generated.
However when testing i find that something is going wrong with longer tags.
Here is a test using an ACR-122U reader and the RFIDIOt tools

Failing tag - 64 bytes:
    `uint8_t selectPPSE[] = {`
    `0x6f,0x2f,0x84,0x0e,0x32,0x50,0x41,0x59,`
    `0x2e,0x53,0x59,0x53,0x2e,0x44,0x44,0x46,`
    `0x30,0x31,0xa5,0x1d,0xbf,0x0c,0x1a,0x61,`
    `0x18,0x4f,0x07,0xa0,0x00,0x00,0x00,0x04,`
    `0x18,0x4f,0x07,0xa0,0x00,0x00,0x00,0x04,`
    `0x18,0x4f,0x07,0xa0,0x00,0x00,0x00,0x04,`
    `0x10,0x10,0x50,0x0a,0x4d,0x61,0x73,0x74,`
    `0x65,0x72,0x43,0x61,0x72,0x64,0x90,0x00};`
`python ChAP.py -dv`
`>  00 A4 04 00 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 00`
`RESPONSE=`
`0`
`no PSE: 00 00`

Passing tag - 56 bytes:
      `uint8_t selectPPSE[] = {`
      `0x6f,0x2f,0x84,0x0e,0x32,0x50,0x41,0x59,`
      `0x2e,0x53,0x59,0x53,0x2e,0x44,0x44,0x46,`
      `0x30,0x31,0xa5,0x1d,0xbf,0x0c,0x1a,0x61,`
      `0x18,0x4f,0x07,0xa0,0x00,0x00,0x00,0x04,`
      `0x18,0x4f,0x07,0xa0,0x00,0x00,0x00,0x04,`
      `0x10,0x10,0x50,0x0a,0x4d,0x61,0x73,0x74,`
      `0x65,0x72,0x43,0x61,0x72,0x64,0x90,0x00};`
`>  00 A4 04 00 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 00`
`RESPONSE=`
`56`
`<  6F 2F 84 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 A5 1D BF 0C 1A 61 18 4F 07 A0 00 00 00 04 18 4F 07 A0 00 00 00 04 10 10 50 0A 4D 61 73 74 65 72 43 61 72 64 90 0 
PSE found!`

This indicates something weird is happening with the modulation - as once i got over ~56 bytes i encounter failing packets.
Any hints on why this is happening would greatly be appreciated (also, its not the reader - a normal EMV NFC card is read fine (i.e tags>64 bytes are common).
Regards,
Peter Fillmore

pwpiwi, 2015-02-23:
I am not familiar with RFIDIOt Tools. Can you please explain what `selectPPSE[]`, the output of `python ChAP.py -dv`and the `RESPONSE`are? Which comes from reader, which comes from tag? What do you see on the PM3 when doing these tests (hf list 14a)?

peterfillmore, 2015-02-23:
selectPPSE is a static response to the reader requesting the PPSE:
i.e <omitting PCB bytes and CRCs>
From Reader(RFIDIOt): 00 A4 04 00 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 00
From Tag(Proxmark):  6F 23 84 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 A5 11 BF 0C 0E 61 0C 4F 07 A0 00 00 00 04 10 10 87 01 01 90 00

Code in use is currently mainly altered from the “SimulateIso14443aTag” function in iso14443a.c

The issue is with generating tag responses of >60 or so bytes. Everything up to that point responds fine. 
I’ve tried increasing buffer sizes to no avail and can’t seem to locate what is causing them to fail.
No errors are generated - the packet just seems to be “lost”. unfortunately i can’t get a raw output from the ACR-122U to work out   what is happening.
Whether this is a timing issue or a buffer issue i’m not sure.

I’ll try pre-generating the modulation of the tag to eliminate any “timing” issue due to generating the response on the fly..

I’m having issues logging with my code presently; but will alter the SimulateIso14443aTag function to test and log tomorrow.

Regards,
-------------------------------------------------------
Peter Fillmore

merlokk, 2015-02-23:
Hi

Here
http://dexterous-programmer.blogspot.com/2012/04/emv-transaction-step-1-application.html

Its the first step of EMV transaction - select right application code.

pwpiwi, 2015-02-23:
Did you try to increase FWI in TB(1) of your ATS (currently FWI = 8) in order to convince the reader to wait longer?

pwpiwi, 2015-02-23:
i.e. change 
```uint8_t ratscmd[] = {0x0b, 0x78, 0x80, 0x81, 0x02, 0x4b, 0x4f, 0x4e, 0x41, 0x14, 0x11, 0x8a, 0x76};``` to .e.g. 
```uint8_t ratscmd[] = {0x0b, 0x78, 0x80, 0xa1, 0x02, 0x4b, 0x4f, 0x4e, 0x41, 0x14, 0x11, 0x8a, 0x76};```

peterfillmore, 2015-02-24:
Here's the test with logging (just a packet of "A"s with PCBs and CRCs added):

Good (64 byte packet):
`30123216 |  30149808 | Rdr | 02  00  a4  04  00  0e  32  50  41  59  2e  53  59  53  2e  44  |     |`          
`                 |                   |        | 44  46  30  31  00  e0  42                                                        |     |` `SELECT`          
`30158916 |  30167236 | Tag | 02  00  41  41  41  41  41  41  41  41  41  41  41  41  41  41  |     |`           
`                 |                   |        | 41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  |     |`           
`                 |                   |        | 41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  |     |`           
`                 |                   |        | 41  41  41  41  41  41  41  41  41  41  41  41  90  00  5f  75   |     |`          
        
`>  00 A4 04 00 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 00`
`RESPONSE=`
`61`
`<  00 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 `41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 41 90 0`  

BAD (65 byte packet):
`17636052 |  17662644 | Rdr | 02  00  a4  04  00  0e  32  50  41  59  2e  53  59  53  2e  44  |     |  `         
`                 |                   |        | 44  46  30  31  00  e0  42                                      |     | SELECT    `      
`17672008 |  17681480 | Tag | 02  00  41  41  41  41  41  41  41  41  41  41  41  41  41  41  |     | `       
`                 |                   |        | 41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  |     |  `        
`                 |                   |        | 41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  |     |  `         
`                 |                   |        | 41  41  41  41  41  41  41  41  41  41  41  41  90  90  00  5a  |     | `          
`                 |                   |        | b3                                                                                            |     |`           
`17778852 |  17782468 | Rdr | b2  67  c7                                                                                |     | ?`          
`17783640 |  17787160 | Tag | a3  6f  c6                                                                                 |     | `          
`17861162 |  17887754 | Rdr | 02  00  a4  04  00  0e  32  50  41  59  2e  53  59  53  2e  44  |     | `          
`                 |                   |        | 44  46  30  31  00  e0  42                                                        |  |SELECT`          
`17896926 |  17906398 | Tag | 02  00  41  41  41  41  41  41  41  41  41  41  41  41  41  41  |     |`           
`                 |           |       | 41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  41           |     |`           
`                 |           |       | 41  41  41  41  41  41  41  41  41  41  41  41  41  41  41  41           |     |`           
`                 |           |       | 41  41  41  41  41  41  41  41  41  41  41  41  90  90  00  5a           |     |`          
`                 |           |       | b3                                                                                                     |     |`
         
`>  00 A4 04 00 0E 32 50 41 59 2E 53 59 53 2E 44 44 46 30 31 00`
`RESPONSE=`
`0`
`no PSE: 00 00`

I've changed the FWT time in the RATS to 0xA1-0xF1; this breaks the communications (no response to ATR).
Tried pre-generating the response rather then dynamically doing it - same result as above.
Tried increasing the ISO14A_Timeout value as well - no change.

Thanks for your help

marshmellow42, 2015-02-24:
I know nothing of the code, but could it be related to
#define DYNAMIC_RESPONSE_BUFFER_SIZE 64
In the ISO14443a Sim  command?

merlokk, 2015-02-24:
Maybe its not for this topic but about your ATS, but:
https://smartcard-atr.appspot.com/parse?ATR=0b788081024b4f4e4114118a76

TA(1) = 0x80	Fi=RFU, Di=RFU, INVALID VALUE <-- needs to be fixed
TB(1) = 0x81	Programming Param P: 1 Volts, I: 4 milliamperes <-- according to EMV it must be 0x00
TC(1) = 0x02	Extra guard time: 2. <--- here "The Guard Time is the minimum delay between the leading edge of the previous character, and the leading edge of the next character sent." http://en.wikipedia.org/wiki/Answer_to_reset. It needs to put here something reasonable.... or just set it to 255.

pwpiwi, 2015-02-24:
> I've changed the FWT time in the RATS to 0xA1-0xF1; this breaks the communications (no response to ATR).

Now what? ATS or ATR?

I forgot: when changing the ATS, the last two (CRC) bytes in `ratscmd[]` need to be adjusted as well.

peterfillmore, 2015-02-24:
Apologies, setting the correct CRCs got it communicating however i’m still not able to send anything over 64 bytes.

I’ve played with the extending the FWT field in the ATS field (TA(1) high bits) with no success. (0x80, 0x90,0xA0,0xB0,0xC0,0xD0,0xE0)

I also dumped the raw tag response after modulation and they all match up (i.e are correctly encoded in the ToSend buffer prior to being sent to the FPGA, parity included).

I’m fairly sure its not a timing issue - as it appears to be related to some kind of 64 byte buffer limit (if it was timing its a heck of coincidence it occurs only after 64 bytes).
 (The buffer sizes have been sized to double the existing limits in the code i.e 64->128, 512->1024).

Does the FPGA have any hard coded buffer size limit? as i can’t locate anywhere otherwise that causes this limitation.

Regards,
-------------------------------------------------------
Peter Fillmore

pwpiwi, 2015-02-24:
The FPGA doesn't have a buffer except the FIFO delay queue (4 Bytes max).

Maybe we are out of memory? You have defined "big" arrays (e.g. of size MAX_FRAME_SIZE*8).

peterfillmore, 2015-02-25:
Dang it,
Worked it out.

The silly ACR122U only supports 64 byte packets, anything over that is chained. Naturally the ACR122U/Drivers don’t warn of overflows etc.

God I hate testing RFID stuff…
Buffer sizing was also an issue - i changed the dynamic buffer to be allocated in the BigBuf area which fixed it.

Might also be worth adding a function to spit out ATS info for readers and tags. And adding checks in the ATS/RATS commands to resize buffers according to reader/tag for functions.

Thanks all for your assistance. Feel free to close the issue.

-------------------------------------------------------
Peter Fillmore

pwpiwi, 2015-02-25:
> And adding checks in the ATS/RATS commands to resize buffers according to reader/tag for functions.

Yeah. Go ahead.

Does the ACR122U correctly provide the info about its maximum frame size in its RATS (FSDI field)?

peterfillmore, 2015-02-26:
Yes,
The ACR122u send the following ATS:
e0 50 bc a5
so reports 64  bytes.

Unfortunately doesn’t fix the differences in reader hardware, each one handles things a little different i’ve found. 
I’ve got my Nexus4 responding correctly to emulated tags, but the device i’m trying to test doesn’t fully work.

ps. nice work on the parity generation - works like a charm.
-------------------------------------------------------
Peter Fillmore

>

iceman1001, 2015-03-04:
close, someone?

closed 2015-03-04 by holiman