Proxmark/proxmark3 #76: missing commands. clone ioprox

elephunk21, 2015-03-12:
Hi i had bought proxmark from xpfga and i was given the indala cloner software for it. can kind soul enlighten how do i upload the ioprox patch as i am missing the io commands in my software. i tried attempting to do a fresh installation but i was unable to get even proxmark.exe running. would really love if i could get some detailed instructions on how to get these io commands. i really dont know where exactly to add the patch provided here. https://penturalabs.wordpress.com/2014/03/11/proxmark3-vs-kanotech-ioprox/

iceman1001, 2015-03-12:
Have you looked in the codebase here? If you have compiled / flashed the latest source then you should have everything,

elephunk21, 2015-03-12:
I have no idea how to compile/flash source or even understand what it means.

Would you be able to assist me? I am more than willing to pay for your time
to help me.

Priya
R008664B
HP: +65 9106 7234 / +65 9144 4078
Email: sgbestrealestate@gmail.com

condocard, 2015-03-12:
google show same info with this.

http://www.rfcopy.com

Priya
R008664B
HP: +65 9106 7234 / +65 9144 4078
Email: sgbestrealestate@gmail.com

iceman1001, 2015-03-12:
if you dont understand how to compile,   then I suggest you try Aspers's precompiled binaries.

http://www.proxmark.org/forum/viewtopic.php?id=1562

Read and try it.

holiman, 2015-03-24:
This is not really a bug, is it? I'll just close it, because I think the issue here is that the reporter does not use the latest code. Correct me if I'm wrong..

closed 2015-03-24 by holiman