Proxmark/proxmark3 #81: Command buffer overflow in hf tune?

lesbeben, 2015-03-20:
Hi everyone,

I was playing around with hf tune, trying a bunch of RFID cards to sse which were HF, when I encountered some kind of bug.

I usually just run hf tune and check for a drop in voltage when the card is close to my antenna. Today, however, after a little while (around a minute, maybe less) the client started displaying:

WARNING: Command buffer about to overwrite command! This needs to be fixed!

instead of the usual "#db# XXXXXmV". Just thought I'd let you know, since I'm in no way proficient enough to fix this myself (yet).

I'm running the latest linux client (git pull today on master) with firmware 2.0.0-rc2.

Thanks as usual!

TL;DR: Ran "hf tune" for around a minute, got "WARNING: Command buffer about to overwrite command! This needs to be fixed!" :kissing_heart:

holiman, 2015-03-24:
I think this is ok, IIRC, this basically says that commands coming in from the device faster than we're handling them. In 'normal' operation, this would be bad, thus the printout that it needs to be fixed. In this particular case, however, where the device just loops spitting out information about the voltage, it's not an issue. 

On the other hand, perhaps we should limit the voltage printouts a bit.

iceman1001, 2015-04-06:
yeah, there is alot of printouts with the "hf tune" command.  maybe slow it down to twice per second?

closed 2016-08-29 by iceman1001