Proxmark/proxmark3 #83: : obj/fullimage.elf section `.data' will not fit in region `osimage'

iceman1001, 2015-03-24:
Noooo,  but I guess it was only a matter of time.  When I add my desfire stuff into the armsrc, the fullimage image size gets too big and overflows into the osimage area.

Does someone know how this works? Is it in the makefiles (common?) or the bootrom?

-------------SNIPPET ERROR MESSAGE
arm-none-eabi-gcc -nostartfiles -nodefaultlibs -Wl,-gc-sections -n -Wl,-T,ldscript,-Map,obj/fullimage.map -o obj/fullimage.elf obj
/version.o obj/fpga_lf.o obj/fpga_hf.o obj/start.o obj/iso15693.o obj/iso15693tools.o obj/lfops.o obj/hitag2.o obj/lfsampling.o ob
j/appmain.o obj/printf.o obj/util.o obj/string.o obj/usb_cdc.o obj/cmd.o obj/fpgaloader.o obj/legicrf.o obj/lfdemod.o obj/epa.o ob
j/iso14443a.o obj/mifareutil.o obj/mifarecmd.o obj/mifaresniff.o obj/iso14443.o obj/crapto1.o obj/crypto1.o obj/des.o obj/aes.o ob
j/desfire_key.o obj/desfire_crypto.o obj/mifaredesfire.o obj/iso14443crc.o obj/crc.o obj/crc16.o obj/crc32.o obj/legic_prng.o obj/
iclass.o obj/BigBuf.o obj/optimized_cipher.o -lgcc
c:/projects/proxmark3/proxspace-130613/proxspace/devkitarm/bin/../lib/gcc/arm-none-eabi/4.7.1/../../../../arm-none-eabi/bin/ld.exe
: obj/fullimage.elf section `.data' will not fit in region `osimage'
c:/projects/proxmark3/proxspace-130613/proxspace/devkitarm/bin/../lib/gcc/arm-none-eabi/4.7.1/../../../../arm-none-eabi/bin/ld.exe
: region `osimage' overflowed by 936 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [obj/fullimage.elf] Error 1

iceman1001, 2015-03-24:
I could be wrong here, but the file:  /common/ldscript.common  seems to be the right place

iceman1001, 2015-03-24:
A suggestion on stackoverflow, to strip unneccesary sections, (ie making the .elf smaller)

-fdata-sections -ffunction-sections

I added it to  makfile.common,
CFLAGS =  -c $(INCLUDE) -Wall -Werror  -fdata-sections -ffunction-sections -pedantic -std=c99 -Os $(APP_CFLAGS)

The size of the fullimage.elf drop 10kb.  Gonna make some tests now, since flashing went well.

ref: http://stackoverflow.com/questions/6687630/c-c-gcc-ld-remove-unused-symbols

pwpiwi, 2015-03-26:
Good point and it will help to extend the available space for a while. We can also make the fpga section smaller (we know very well how big it needs to be - the fpga config file is always the same size). 

But in the end I think we should start to avoid to implement too much at the arm side. The usual command-response sequence is not time critical and can be done via USB.

iceman1001, 2015-03-26:
I agree, we should have some clear guidelines.   
@pwpiwi   You tested to compile my fork,  did it make a different in .elf size for you?

pwpiwi, 2015-03-28:
> @pwpiwi You tested to compile my fork, did it make a different in .elf size for you?

```
Flashing...
Writing segments for file: armsrc/obj/osimage.elf
 0x00116a00..0x001389df [0x21fe0 / 272 blocks].......................
```
This is commit 9552d5 from your repository plus the topaz commit 48ece4.

marshmellow42, 2015-03-28:
So about 136kb of the 160kb available.  Right?

iceman1001, 2015-03-28:
could be right, since the desfire adds alot...

marshmellow42, 2015-03-28:
I could move all the lfdemod stuff that has no current arm use implantation to another file just in the client and shrink the file for the arm.  Currently only hid, em410x and ioprox and its dependencies are used in the arm.  That might gain 5+ KB.  

Do you know about how much more you need and if it all has to be in the arm?

holiman, 2015-03-28:
If functions are not used, my feeling is that clever compiler options should be able to remove them...  it's a pretty common problem.

marshmellow42, 2015-03-28:
I'm not familiar with the compiler options, I could list the functions not used by the arm (but are used in the client) if it helps

marshmellow42, 2015-03-28:
Nrz and psk demods and detect clocks are an example that alone would likely gain 2kb

pwpiwi, 2015-03-29:
@iceman1001 pointed out above which compiler options to use. The required linker options (`-Wl,-gc-sections`) are already there. Together this results in elimination of unused functions/data. In addition we can
- shrink the fpga section
- use -O2 only for those object files which require speed.

iceman1001, 2015-03-29:
This raises the question of how should the device be used as?   Some want to use it in standalone mode, making it a pen-test device on the field.   Some, like me, sits with it connect to a commy testing tags at the desk,.   Two different kinds of builds,  since we don't have enough space on the device we can let the device/code be both for much longer. 

One build would be with simulation and different known/usuable attacks, for fieldtesting.
One build would be for analysing tags offline, or a controlled environment.

What do you say?

holiman, 2015-03-29:
Or, possibly, split into HF/LF ...

pwpiwi, 2015-03-29:
Or, possibly revert the FPGA LF/HF split...?

holiman, 2015-03-29:
What would be the gain from that? 
I have a feeling that it only postpones the inevitable... and it leaves less room for FGPA improvements...

marshmellow42, 2015-03-29:
Is it still an issue currently?  Sounds like you got it shrunk some.

holiman, 2015-03-30:
If it's not an issue right now, let's go with the current fixes (flags) and leave more dramatic solutions for later. I think that we might benefit from reducing the ARM-side code a bit via some refactorings which would both save space and make the code cleaner.

iceman1001, 2015-03-30:
Its a matter of time, could one year or more. It all depends on what kind of stuff which is in development.  The Desfire stuff is one,  and the EMV from @peterfilmore is another.

I don't know what you guys ( @holiman  @pwpiwi  @marshmellow42  ) are up to?

pwpiwi, 2015-03-30:
Agree with @holiman: for the time being let's implement those measures which have no disadvantage:
- shrink fpga-section
- compiler flags -ffunctions-section and -fdata-section
- revert O2 optimization (it hasn't been committed to master yet)

holiman, 2015-03-30:
@iceman1001 , I'm reading up on desfire atm, and UL-C, to get up to speed with what you're doing. I've also ordered some new tags to play with, and plan to implement aptus decoder (LF).

iceman1001, 2015-03-30:
There is nothing much new about UL-C.  I just added the crypto part on the deviceside, thinking that is need for speed.  But given the discusion about size, I'm not so sure the cryptopart needs to be on the deviceside for the UL-C part.    The next UL-Ev1  has a few more commands extra. Not much.

The Desfire part can be solved now since @pwpiwi  solved the parity problem with bigger frames.  Which was one of the reasons I failed,  looking back at my coding efforts last autumn for the PM3 community.  If we get the apdu:s together with the iso14443a frames,  we can save alot of space I think. However crypto and mac/cmac calculations is very much more default with the Desfire,  which I don't know how sensitive the timings are. DES/3DES/AES is the possible cryptos,  all needs to be supported if we want to be able to read any given Desfire tag (or application that relys upon Desfire tags) 

As a sidenote, I've been trying to talk to the Hashcat project people over at, http://hashcat.net/
to see if their expertis in cryptos could be to some use for our community.  Do we have a collection of given cryptos that are used,  and which of them are used with humans setting the password?  (ie computer generated masterkeys is out-of-reach,  bruteforce..)

pwpiwi, 2015-03-31:
See pull request #89. @marshmellow42: you had a few float operations in lfdemod.c. I removed them to avoid adding float libraries to the ARM os image (saves a few KBytes). Please check if this still works as expected.

iceman1001, 2015-03-31:
Great!

closed 2015-03-31 by iceman1001