Proxmark/proxmark3 #87: Experiencing many collissions / failed reads after latest changes to fpga-code.

iceman1001, 2015-03-25:
Since the latest fpga changes, my HF commands is hardly working anymore.  Collisions and failed reads 95% of the time.

Trying to pinpoint it, I have tested an old bit-file from january 2015.  With it, my HF commands works as expected.   

Does someone have some suggestions on what this might be?

https://github.com/Proxmark/proxmark3/blob/master/fpga/fpga_hf.bit

holiman, 2015-03-26:
What date in January?

iceman1001, 2015-03-26:
It's from: 2015-01-20

pwpiwi, 2015-03-26:
Pulled your repository, compiled it (needed to add two files to Makefile) and tested with three different versions of fpga_hf.bit. No issues. Neither with standard iso14443a:
```
pm3 --> hf 14a reader
ATQA : 00 04
 UID : ww xx yy zz
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1
proprietary non iso14443-4 card found, RATS not supported
Answers to chinese magic backdoor commands: NO
pm3 --> hf list 14a
Recorded Activity (TraceLen = 133 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

     Start |       End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
-----------|-----------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2228 |       4596 | Tag | 04  00                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL
      10676 |      16564 | Tag | ww  xx  yy  zz  b9                                              |     |
      18560 |      29024 | Rdr | 93  70  ww  xx  yy  zz  b9  ad  18                              |  ok | SELECT_UID
      30260 |      33780 | Tag | 08  b6  dd                                                      |     |
     482944 |     487712 | Rdr | e0  80  31  73                                                  |  ok | RATS
     488884 |     489524 | Tag | 04                                                              |     |
     917248 |     918240 | Rdr | 40                                                              |     | MAGIC WUPC1
    1053696 |    1055008 | Rdr | 43                                                              |     | MAGIC WUPC2
    1190784 |    1195552 | Rdr | 50  00  57  cd                                                  |  ok | HALT
pm3 -->
```

nor with new Topaz mode:
```
pm3 --> hf 14a raw -a -p -T 26
received 2 octets
00 0C
pm3 --> hf 14a raw -a -p -T -c 78 00 00 00 00 00 00
received 8 octets
12 4C 8D BD 64 00 92 DA
pm3 --> hf 14a raw -a -p -T -c 00 00 00 8d bd 64 00
received 124 octets
12 4C 8D BD 64 00 00 10 25 00 E1 10 3F 00 01 03 F2 30 33 02 03 F0 02 03 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0
0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55
 55 AA AA 12 4C 06 00 01 E0 00 00 00 00 00 00 D0 0D
pm3 --> hf list topaz
Recorded Activity (TraceLen = 238 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

     Start |       End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
-----------|-----------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |      57504 | Rdr | 00  00  00  8d  bd  64  00  99  5d                              |  ok | RALL
      58676 |      70580 | Tag | 12  4c  8d  bd  64  00  00  10  25  00  e1  10  3f  00  01  03  |     |
            |            |     | f2  30  33  02  03  f0  02  03  03  00  00  00  00  00  00  00  |     |
            |            |     | 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  |     |
            |            |     | 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  |     |
            |            |     | 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  |     |
            |            |     | 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  |     |
            |            |     | 00  00  00  00  00  00  00  00  00  00  55  55  aa  aa  12  4c  |     |
            |            |     | 06  00  01  e0  00  00  00  00  00  00  d0  0d                  |  ok |
pm3 --> hf topaz read
not yet implemented
pm3 --> 
```

I assume that you have the same problem with both your PM3s? In the forum you mentioned that it got slightly better with the fpga.bit from my topaz branch. Well, this differs only in a slightly increased threshold to avoid false detections when there is more noise than signal on the antenna. If this is the case it might also be worth to try the modified Mod_Miller_LUT from my topaz branch which expects more "ideal" signals instead of trying to ignore single bit errors (and thereby tending to accept noise as real signals). It then should fail more completely instead of produing false positives. Doesn't sound very promising? Right. In the end "a good antenna is the best amplifier". I suggest to go for it.

iceman1001, 2015-03-26:
I think I actually managed to brick one of them,  that would be my first time.  Always fun with doing something for the first time. 
One device, I'm using a pcb antenna from Gezhi Electronics, 14-16v  works great with the older bit file.  
The other is even lower in voltage,  and works great with the older bit file.

When I did try @pwpiwi 's topaz changes and fixes, I got it to work holding the antenna 1cm further away than normal. It started to become really unstable.  Lately I couldn't use any of the commands with a collision and/or failure to read.  

I think I will look into getting a better antenna.. But since I use a standard antenna from the Pm3 kits available, we can assume that more people with same hardware will start popping up in the forum asking for help.  So far, its been quiet,  lets see what happens with the v2 release.

As a sidenote, the antenna connection thingy sucks.  How hard was it to get mmcx connector?!?

closed 2015-06-23 by iceman1001

pwpiwi, 2015-06-24:
How did you solve that issue? What was the root cause?

iceman1001, 2015-06-24:
The root cause is changes to the fpga image. I solve it by changing hardware,  a stronger antenna.

The solution for those old prefab antenna's from xfpga.com etc I swap back to an old fpga-image from before these changes in january.