Proxmark/proxmark3 #88: reading KeyB in  hf mf nested/mifare ?

iceman1001, 2015-03-31:
It's not an issue, but an suggestion.  Someone can edit this issue and give it the right label.

According to Adam Laurie,   https://github.com/AdamLaurie/mfoc/commit/5d8bf95968138b880891103df082c5396bfbc56e

He suggested in the MFOC to use a found KeyA to try reading the KeyB from the sectortrailer.

ikarus23, 2015-04-04:
This is a great idea to speed up the process. If KeyB is readable there is no need to recover it. It's just user data. (An authentication with a readable KeyB can be performed, but other actions like read/write/incr/decr/rest can not.)

iceman1001, 2015-04-05:
When I saw Adam's commited code I wondered why we didn't come up with that one for Pm3.. 
@ikarus23  you should test it for your app.

iceman1001, 2016-01-16:
I implemented an version of this functionality in my fork, for the "hf mf nested" command.
If there is a foundkey[a] and not foundkey[b]  it tries to read the sectortrailer (ST) and see if the keyB data is not zero. 
The negative part is some annoying debugstatements from the device side when reading ST.

closed 2016-01-20 by iceman1001