Proxmark/proxmark3 #90: t55xx Biphase configuration

iceman1001, 2015-03-31:
At the moment @marshmellow42  and I have identified some minor problems with the biphase demodulation,  where the  lf t55xx read/info/trace will not manage to correctly decode a t55x7 tag configured for Biphase.   We are looking into it.   

I'm also noticing a strange behavior with t55x7 tags, where the Biphase configuration setting needs a password when setting it, regardless if the tag wasn't configured to be in password mode. Very strange indeed.  If someone wants to do some test it would be good. At least so I can rule out the possiblity that its my t55x7 tags that are strange.

marshmellow42, 2015-03-31:
I am not seeing the same issue.  but I am using code that is in my newly created pull request that did refactor the demods and fixed a few bugs.

marshmellow42, 2015-03-31:
i am also using an antenna with 45.65v

iceman1001, 2015-03-31:
Found it,

----
        case DEMOD_BIa:
            //sprintf(cmdStr,"0 %d %d 1", bitRate[config.bitrate], config.inverted );
            sprintf(cmdStr,"0 0 %d 1",  config.inverted );
            ans = ASKbiphaseDemod(cmdStr, FALSE);
-----

It's when I send in the config.bitrate into the ASKbiphaseDemod it becomes wrong for me.
When I send in 0, it autodetects the clock and works like it should

iceman1001, 2015-03-31:
All in all,  this leads into the ASKbiphaseDemod and deeper,  until the DetectASKClock call.

Where it suggests a good starting point if I don't call the function with a clock.
seen below as  42.
But when I call it with a clock,  it gives me the starting point of 0.
Understanding clock and starting offset is not the same thing.

---
pm3 ~/client$ proxmark3.exe com3
pm3 --> lf t55xx detect
[0 1 1] 3, 0, 1, 1, 48
askrawdemod:  suggested start: 42
Modulation : BIPHASE
Bit Rate   : 4 - RF/50
Inverted   : No
Offset     : 0
Block0     : 0x00110040

pm3 --> lf t55xx read 0
[50 0 1] 3, 50, 0, 1, 53
askrawdemod:  suggested start: 0
--

iceman1001, 2015-03-31:
But looking at @marshmellow42  latest PR,  I suspect that it fixes the beststart issue.
https://github.com/marshmellow42/proxmark3/commit/2eec55c8a4331daf5d523a1050e3381501b36b34

marshmellow42, 2015-03-31:
i think there was a bug around that that i fixed if i recall correctly...  (been working lots of stuff and have lost track...)

iceman1001, 2015-03-31:
I'll consider this issue closed with the latest PR from @marshmellow42

closed 2015-03-31 by iceman1001