Proxmark/proxmark3 #94: Suggestion:  identify "weak" mifare classic tags

iceman1001, 2015-04-12:
This is not an issue, its a enhancement suggestion (?) 
I started a thread on the forum about this idea. 
http://www.proxmark.org/forum/viewtopic.php?pid=15426#p15426

iceman1001, 2015-05-19:
If the memory serves me right,  the test should set all parity bits correct and databits wrong,  if it answers with a nack  then its a "weak" tag.  Right?
But this tag is end-of-line,  maybe we don't need a test.

iceman1001, 2015-05-25:
Does someone have a data sample on what triggers this NACK response?

iceman1001, 2015-06-25:
I'm gonna test this tonight.  Having the commands report this back to the user is good. 
I wonder how long it takes before the commands report back?

iceman1001, 2015-06-25:
Look at that!  

---
pm3 --> hf mf mi
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average  :-)
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
......

Card is not vulnerable to Darkside attack (its random number generator is not predictable).

pwpiwi, 2015-06-30:
@holiman: merged PR #125. You may now please change the lua script accordingly.

closed 2015-07-12 by holiman