Proxmark/proxmark3 #96: HF 14a READ identify Ultralight/Ultralight-C/Ultralight EV1

iceman1001, 2015-04-24:
I made a minor fix to the "hf 14a read" command to correctly identify between the different Ultralight tags.

---snipp--
pm3 --> hf 14a read
ATQA : 00 44
 UID : 04 b7 80 9a f8 38 80
 SAK : 00 [2]
MANUFACTURER : NXP Semiconductors Germany
TYPE : NXP MIFARE Ultralight EV1 48 bytes
proprietary non iso14443-4 card found, RATS not supported
Answers to chinese magic backdoor commands: NO

pm3 --> hf 14a read
ATQA : 00 44
 UID : 01 02 03 04 05 06 07
 SAK : 00 [2]
MANUFACTURER : Motorola UK
TYPE : NXP MIFARE Ultralight | Ultralight C
proprietary non iso14443-4 card found, RATS not supported
Answers to chinese magic backdoor commands: NO

iceman1001, 2015-04-24:
If someone with with less problems than I in creating PR can add:

https://github.com/iceman1001/proxmark3/commit/abcb166ffe26cab359cd80e0109e6131e31bef49

marshmellow42, 2015-04-28:
Ultralight C also seems to respond to the 0x60 command.  so the descriptions are incorrect.

iceman1001, 2015-04-28:
thats strange,  my UL-C tags doesn't react on 0x60...

marshmellow42, 2015-04-28:
MF0ICU2 ?

iceman1001, 2015-04-28:
Do you mean which datasheet?  MF0ICU2 full datasheet, rev3.1
My UL-C is one of those magic ones..  I'm not sure if it follows standards so well.

marshmellow42, 2015-04-28:
I've tried 5 different UL-C tags and the new code identifies it as UL-EV1.  it identifies original UL ok and UL-EV1 ok (I've tried them all).  my UL-C are genuine NXP

marshmellow42, 2015-04-28:
undocumented command for the genuine UL-C then?

iceman1001, 2015-04-28:
Maybe undocumented feature,   Do you have a later rev of the datasheet?

or your tags is UL-Ev1 but you didn't know it...  

Do you have a trace after you run "hf 14 read" ?  I'm curios of how it answers.

marshmellow42, 2015-04-28:
MF0ICU2 is the identification of the UL-C chip in my tags (tag-info app)

marshmellow42, 2015-04-28:
Tag-info identifies each chip correctly and i'm positive the are as I identified them.  i'll get the list result

iceman1001, 2015-04-28:
doesn't tag-info only look for ATQA and SAK?

marshmellow42, 2015-04-28:
no it does much more

marshmellow42, 2015-04-28:
Start |       End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
-----------|-----------|-----|-----------------------------------------------------------------|-----|--------------------|
         0 |       992 | Rdr | 52                                                              |     | WUPA
      2228 |      4596 | Tag | 44  00                                                          |     |
      7040 |      9504 | Rdr | 93  20                                                          |     | ANTICOLL
     10676 |     16564 | Tag | 88  04  09  a9  2c                                              |     |
     18688 |     29152 | Rdr | 93  70  88  04  09  a9  2c  92  74                              |     | SELECT_UID
     30388 |     33908 | Tag | 04  da  17                                                      |     |
     35200 |     37664 | Rdr | 95  20                                                          |     | ANTICOLL-2
     38836 |     44724 | Tag | 7a  e0  2b  80  31                                              |     |
     46848 |     57376 | Rdr | 95  70  7a  e0  2b  80  31  db  ea                              |     | ANTICOLL-2
     58548 |     62132 | Tag | 00  fe  51                                                      |     |
    748928 |    752544 | Rdr | 60  f8  32                                                      |     | AUTH-A(248)
    753716 |    754356 | Tag | 00!                                                             |     |
   1173120 |   1177888 | Rdr | e0  80  31  73                                                  |     | RATS
   1596032 |   1597024 | Rdr | 40                                                              |     | MAGIC WUPC1
   1732480 |   1733792 | Rdr | 43                                                              |     | MAGIC WUPC2
   1869568 |   1874336 | Rdr | 50  00  57  cd                                                  |     | HALT

iceman1001, 2015-04-28:
hm,  it answers with a 0x00..  which would be a UL-C  nack?

marshmellow42, 2015-04-28:
or invalid parameter?

marshmellow42, 2015-04-28:
it doesn't answer the magic or other commands tried, so 60 must be a valid command but not called correctly ?

iceman1001, 2015-04-28:
For Mifare Classic 0x60, would be authentice,..
For UL-EV1

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |

------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|

          0 |        992 | Rdr |52                                                               |     | WUPA
       2228 |       4596 | Tag |44  00                                                           |     |
       7040 |       9504 | Rdr |93  20                                                           |     | ANTICOLL
      10676 |      16500 | Tag |88  04  b7  80  bb                                               |     |
      18560 |      29024 | Rdr |93  70  88  04  b7  80  bb  d6  fa                               |  ok | SELECT_UID
      30260 |      33780 | Tag |04  da  17                                                       |     |
      35072 |      37536 | Rdr |95  20                                                           |     | ANTICOLL-2
      38708 |      44596 | Tag |9a  f8  38  80  da                                               |     |
      46720 |      57184 | Rdr |95  70  9a  f8  38  80  da  68  d4                               |  ok | ANTICOLL-2
      58420 |      62004 | Tag |00  fe  51                                                       |     |
     822912 |     826528 | Rdr |60  f8  32                                                       |  ok | EV1 VERSION
     827700 |     839348 | Tag |00  04  03  01  01  00  0b  03  fd  f7                           |  ok |
    1603840 |    1608608 | Rdr |e0  80  31  73                                                   |  ok | RATS
    2858112 |    2859104 | Rdr |40                                                               |     | MAGIC WUPC1
    2995840 |    2997152 | Rdr |43                                                               |     | MAGIC WUPC2
    3134208 |    3138976 | Rdr |50  00  57  cd                                                   |  ok | HALT
pm3 --> e

iceman1001, 2015-04-28:
Page 20 in datasheet,  states 0x00 answer is "any other error NAK"..

marshmellow42, 2015-04-28:
well I suppose we now know when the answer is 00 then it is a UL-C...  :)

marshmellow42, 2015-04-28:
60 must be an undocumented command of some sort or it wouldn't give an error... imo

marshmellow42, 2015-04-28:
I take it back.  upon first error it stops the communication - if I remove the 0x60 command it does the same thing on the rats command
     Start |       End | Src | Data (! denotes parity error)                                   | CRC | Annotation
-----------|-----------|-----|-----------------------------------------------------------------|-----|--------------------
         0 |       992 | Rdr | 52                                                              |     | WUPA
      2228 |      4596 | Tag | 44  00                                                          |     |
      7040 |      9504 | Rdr | 93  20                                                          |     | ANTICOLL
     10676 |     16500 | Tag | 88  04  35  b9  00                                              |     |
     18688 |     29216 | Rdr | 93  70  88  04  35  b9  00  60  29                              |     | SELECT_UID
     30388 |     33908 | Tag | 04  da  17                                                      |     |
     35200 |     37664 | Rdr | 95  20                                                          |     | ANTICOLL-2
     38836 |     44724 | Tag | 7a  e0  2b  80  31                                              |     |
     46848 |     57376 | Rdr | 95  70  7a  e0  2b  80  31  db  ea                              |     | ANTICOLL-2
     58548 |     62132 | Tag | 00  fe  51                                                      |     |
    750592 |    755360 | Rdr | e0  80  31  73                                                  |     | RATS
    756532 |    757172 | Tag | 00!                                                             |     |
   1172352 |   1173344 | Rdr | 40                                                              |     | MAGIC WUPC1
   1308800 |   1310112 | Rdr | 43                                                              |     | MAGIC WUPC2
   1445888 |   1450656 | Rdr | 50  00  57  cd                                                  |     | HALT

iceman1001, 2015-04-28:
Yes indeed!  Actually very good.   Now the "hf mfu" commands can be modified to deal with all three kinds easier.  :)

The datasheet states that all commands (besides the documented ones) sent to tag,  will be answered accordingly  0x0,0x1, 0xa..

0xa ACK
0x1 NACK (parity wrong)
0x0 NACK (any other error).

When it sends it,  the tag will reset.  So I guess it is a failsafe for fuzzying commands.

marshmellow42, 2015-04-28:
so really the first test if fails invalidates any further tests. (rats, magic1, magic2)...

iceman1001, 2015-04-28:
yes, until  WUPA/REQ?

marshmellow42, 2015-04-28:
looks like it may need
HALT - WUPA - SELECT_UID 
to continue

iceman1001, 2015-04-28:
I tried my UL,   it doesn't answer to the 0x60 command or magic..

the thing with the magic test, is that a magic tag can change its ATQA&SAK, so we test it no matter what kind of tag we got before.

marshmellow42, 2015-04-28:
taginfo app appears to try 0x60 f8 32 
tag errors
then they recover with HALT-WUPA-Select_UID-anticoll...
interesting too it appears it also attempts to log in - 1a - then af with key....
my tag errors on the key (af) command

iceman1001, 2015-04-28:
error on af means wrong key..
That is how I try to detect UL vs UL-C in the  "hf mfu dump" command among other places.
I try and see if the tag answers to a 0x1A command.   UL  doesnt   and Ul-EV1 doesnt ..

marshmellow42, 2015-04-28:
another interesting thing is it tries a second key and it works...  even though my cards are default config.

iceman1001, 2015-04-28:
You can test some default keys with the  "hf mfu cauth h"

iceman1001, 2015-04-28:
ok,  thanks @marshmellow42 

I think I cleaned it up now,  

https://github.com/iceman1001/proxmark3/commit/466bc4599c9aa7f3a925022322646f454f08d27f

marshmellow42, 2015-04-28:
why did you remove the manufacturer test for large UID's?

marshmellow42, 2015-04-28:
oops nvm
I found it duh....

iceman1001, 2015-04-28:
;)   sorry, but I just didn't like how the order of items in the output from "hf 14a read" looked like. So I changed it.    For the better I hope.

closed 2015-04-28 by iceman1001