Proxmark/proxmark3 #100: bufferoverflow when reading mfu (nintendo amiibo)

borjaburgos, 2015-05-02:
```proxmark3> hw version
#db# Prox/RFID mark3 RFID instrument
#db# bootrom: master/v2.0.0-45-g5023d4c-suspect 2015-05-02 17:17:31
#db# os: master/v2.0.0-45-g5023d4c-suspect 2015-05-02 21:05:24
#db# HF FPGA image built on 2015/03/09 at 08:41:42
uC: AT91SAM7S256 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory```

```proxmark3> hf 14a reader
 UID : 04 d2 57 7a e3 3e 80
ATQA : 00 44
 SAK : 00 [2]
*** buffer overflow detected ***: ./proxmark3 terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x73f1f)[0x7fa460dc2f1f]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x7fa460e5a82c]
/lib/x86_64-linux-gnu/libc.so.6(+0x10a6f0)[0x7fa460e596f0]
./proxmark3[0x425b3e]
./proxmark3[0x43c908]
./proxmark3[0x4269c2]
./proxmark3[0x43c908]
./proxmark3[0x425241]
./proxmark3[0x43c908]
./proxmark3[0x406873]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8182)[0x7fa462cd9182]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7fa460e4a00d]
======= Memory map: ========
00400000-00491000 r-xp 00000000 08:01 321899                             /home/borjabugos/proxmark3/client/proxmark3
00690000-00691000 r--p 00090000 08:01 321899                             /home/borjabugos/proxmark3/client/proxmark3
00691000-00694000 rw-p 00091000 08:01 321899                             /home/borjabugos/proxmark3/client/proxmark3
00694000-01b6a000 rw-p 00000000 00:00 0
02813000-02834000 rw-p 00000000 00:00 0                                  [heap]
7fa458000000-7fa458021000 rw-p 00000000 00:00 0
7fa458021000-7fa45c000000 ---p 00000000 00:00 0
7fa45c323000-7fa45ca05000 r--p 00000000 08:01 8234                       /usr/lib/locale/locale-archive
7fa45ca05000-7fa45ca06000 ---p 00000000 00:00 0
7fa45ca06000-7fa45d206000 rw-p 00000000 00:00 0                          [stack:3551]
7fa45d206000-7fa45d207000 ---p 00000000 00:00 0
7fa45d207000-7fa45da07000 rw-p 00000000 00:00 0                          [stack:3550]
7fa45da07000-7fa45da0c000 r-xp 00000000 08:01 13395                      /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fa45da0c000-7fa45dc0b000 ---p 00005000 08:01 13395                      /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fa45dc0b000-7fa45dc0c000 r--p 00004000 08:01 13395                      /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fa45dc0c000-7fa45dc0d000 rw-p 00005000 08:01 13395                      /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7fa45dc0d000-7fa45dc2a000 r-xp 00000000 08:01 14452                      /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fa45dc2a000-7fa45de2a000 ---p 0001d000 08:01 14452                      /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fa45de2a000-7fa45de2b000 r--p 0001d000 08:01 14452                      /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fa45de2b000-7fa45de2c000 rw-p 0001e000 08:01 14452                      /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0
7fa45de2c000-7fa45de30000 r-xp 00000000 08:01 137809                     /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fa45de30000-7fa45e02f000 ---p 00004000 08:01 137809                     /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fa45e02f000-7fa45e030000 r--p 00003000 08:01 137809                     /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fa45e030000-7fa45e031000 rw-p 00004000 08:01 137809                     /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fa45e031000-7fa45e038000 r-xp 00000000 08:01 13666                      /usr/lib/x86_64-linux-gnu/libffi.so.6.0.1
7fa45e038000-7fa45e237000 ---p 00007000 08:01 13666                      /usr/lib/x86_64-linux-gnu/libffi.so.6.0.1
7fa45e237000-7fa45e238000 r--p 00006000 08:01 13666                      /usr/lib/x86_64-linux-gnu/libffi.so.6.0.1
7fa45e238000-7fa45e239000 rw-p 00007000 08:01 13666                      /usr/lib/x86_64-linux-gnu/libffi.so.6.0.1
7fa45e239000-7fa45e276000 r-xp 00000000 08:01 137754                     /lib/x86_64-linux-gnu/libpcre.so.3.13.1
7fa45e276000-7fa45e475000 ---p 0003d000 08:01 137754                     /lib/x86_64-linux-gnu/libpcre.so.3.13.1
7fa45e475000-7fa45e476000 r--p 0003c000 08:01 137754                     /lib/x86_64-linux-gnu/libpcre.so.3.13.1
7fa45e476000-7fa45e477000 rw-p 0003d000 08:01 137754                     /lib/x86_64-linux-gnu/libpcre.so.3.13.1
7fa45e477000-7fa45e479000 r-xp 00000000 08:01 13384                      /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fa45e479000-7fa45e679000 ---p 00002000 08:01 13384                      /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fa45e679000-7fa45e67a000 r--p 00002000 08:01 13384                      /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fa45e67a000-7fa45e67b000 rw-p 00003000 08:01 13384                      /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7fa45e67b000-7fa45e6da000 r-xp 00000000 08:01 13423                      /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7fa45e6da000-7fa45e8da000 ---p 0005f000 08:01 13423                      /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
[... 97 lines cut ...]
7fa462ad0000-7fa462ccf000 ---p 00003000 08:01 137668                     /lib/x86_64-linux-gnu/libdl-2.19.so
7fa462ccf000-7fa462cd0000 r--p 00002000 08:01 137668                     /lib/x86_64-linux-gnu/libdl-2.19.so
7fa462cd0000-7fa462cd1000 rw-p 00003000 08:01 137668                     /lib/x86_64-linux-gnu/libdl-2.19.so
7fa462cd1000-7fa462cea000 r-xp 00000000 08:01 137771                     /lib/x86_64-linux-gnu/libpthread-2.19.so
7fa462cea000-7fa462ee9000 ---p 00019000 08:01 137771                     /lib/x86_64-linux-gnu/libpthread-2.19.so
7fa462ee9000-7fa462eea000 r--p 00018000 08:01 137771                     /lib/x86_64-linux-gnu/libpthread-2.19.so
7fa462eea000-7fa462eeb000 rw-p 00019000 08:01 137771                     /lib/x86_64-linux-gnu/libpthread-2.19.so
7fa462eeb000-7fa462eef000 rw-p 00000000 00:00 0
7fa462eef000-7fa462f2c000 r-xp 00000000 08:01 137776                     /lib/x86_64-linux-gnu/libreadline.so.6.3
7fa462f2c000-7fa46312c000 ---p 0003d000 08:01 137776                     /lib/x86_64-linux-gnu/libreadline.so.6.3
7fa46312c000-7fa46312e000 r--p 0003d000 08:01 137776                     /lib/x86_64-linux-gnu/libreadline.so.6.3
7fa46312e000-7fa463134000 rw-p 0003f000 08:01 137776                     /lib/x86_64-linux-gnu/libreadline.so.6.3
7fa463134000-7fa463135000 rw-p 00000000 00:00 0
7fa463135000-7fa463158000 r-xp 00000000 08:01 137626                     /lib/x86_64-linux-gnu/ld-2.19.so
7fa46332b000-7fa46333c000 rw-p 00000000 00:00 0
7fa46334b000-7fa46334e000 rw-p 00000000 00:00 0
7fa46334e000-7fa463355000 r--s 00000000 08:01 266429                     /usr/lib/x86_64-linux-gnu/gconv/gconv-modules.cache
7fa463355000-7fa463357000 rw-p 00000000 00:00 0
7fa463357000-7fa463358000 r--p 00022000 08:01 137626                     /lib/x86_64-linux-gnu/ld-2.19.so
7fa463358000-7fa463359000 rw-p 00023000 08:01 137626                     /lib/x86_64-linux-gnu/ld-2.19.so
7fa463359000-7fa46335a000 rw-p 00000000 00:00 0
7fff18cfe000-7fff18d1f000 rw-p 00000000 00:00 0                          [stack]
7fff18dc1000-7fff18dc3000 r-xp 00000000 00:00 0                          [vdso]
7fff18dc3000-7fff18dc5000 r--p 00000000 00:00 0                          [vvar]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]```

iceman1001, 2015-05-03:
Yes,  
The version-array is too small, it should be 10 instead of 8.
https://github.com/Proxmark/proxmark3/blob/master/client/cmdhf14a.c#L180

borjaburgos, 2015-05-03:
That change seems to work, thanks @iceman1001 Should we do a PR and fix in master?

holiman, 2015-05-03:
That does not look right to me... : 

			uint8_t version[8] = {0,0,0,0,0,0,0,0};
			memcpy(&version, resp.d.asBytes, resp.arg[0]);
			uint8_t len = resp.arg[0] & 0xff;

version is already an array, i.e, a pointer. The operation `&version` returns the *address* of the pointer. I believe it should be like this:

			uint8_t version[10] = {0};
			memcpy(version, resp.d.asBytes, resp.arg[0] < sizeof(version) ? resp.arg[0] : sizeof(version));

holiman, 2015-05-03:
Same here:	

	    memcpy(&card.ats, resp.d.asBytes, resp.arg[0]);

Since card-struct looks like this:

	typedef struct {
		byte_t uid[10];
		byte_t uidlen;
		byte_t atqa[2];
		byte_t sak;
		byte_t ats_len;
		byte_t ats[256];
	} __attribute__((__packed__)) iso14a_card_select_t;

=> `ats` is already a pointer. Normally I'd just commit this, but @borjaburgos indicated the fix from @iceman1001  worked, so I'd rather someone confirmed that I'm not just confused here... ??

holiman, 2015-05-03:
On second thought, I think I'm right... Let me know if I can close this..

borjaburgos, 2015-05-03:
@holiman LGTM, just in case, I'll test tonight and provide feedback.

marshmellow42, 2015-05-03:
I agree with holiman however I believe the compiler fixes it in this case.  As a second item do we have good evidence that declaring an array = {0} fills the array with zeros as I thought most compilers didn't.

iceman1001, 2015-05-04:
What is the current problem?
Array written overbounds with a 10byte string into an 8byte.  (which will overwrite the eip or whatever that pointers name is)  
or that the memcpy was called with a pointer of a pointer,  overwritting potentioal code places.

The solution for overboundery writing is to make sure it fits,  which @holiman's fixes, but I don't agree with silently cutting of inputs is a better way of dealing with it.  If the message doesn't match up precisly then it should fail. 

The memcpy calling solution, 
When I search the PM3 code, we can find some more  "memcpy(&"   possible issues. We would need to verify that they don't introduce the same issue as mine here and not trust the compiler to fix it for us.

@marshmellow42, I don't see how this would prove that declaring an array with zeros works.

marshmellow42, 2015-05-04:
@iceman1001  I was asking if we had evidence not pointing to this as evidence, as I think all the ={0} s can be misleading if it does not fill the array with zeros like I suspect it doesn't.  And this really is not part of the original problem.  I Just keep seeing it used, and wanted to know why.

iceman1001, 2015-05-04:
Indeed,  I was mislead to think you can get the array filled with any value, which was a mistake, 
there is potential issue with not clearing array in the code before usuage. It could give us false feedbacks. However using it as now  will put  index[0] == 0x00 at least when  zero-termination is used thats a good thing.  but yes, confusing.

pointer problem:   lets take this one.

I find this version of code in places,   a call to memcpy  with a specific index of a array.   Is this an example of the pointer problem too? 

https://github.com/Proxmark/proxmark3/blob/master/client/mifarehost.c#L423

iceman1001, 2015-05-04:
I found this one, on the internet.  Looks like the pointer issue is dealt with the compiler.

--snippet... 
Incidentally, when dealing with arrays, passing a is synonymous with passing &a which is synonymous with passing &a[0]. The following are all compiled the same.
memcpy(a, b, sizeof(int) * 4);
memcpy(&a, b, sizeof(int) * 4);
memcpy(&a[0], b, sizeof(int) * 4);

marshmellow42, 2015-05-04:
btw, I think trimming the data if it exceeds 10 bytes is wise as some tags (not UL) may respond with a bunch of data and crash the client :(.  so Holiman's solution is good.  I see Iceman's point about it being silent though.  maybe we should just exit: 
    if (resp.arg[0] > sizeof(version)) return 1;

holiman, 2015-05-05:
>>The solution for overboundery writing is to make sure it fits, which @holiman's fixes, but I don't agree with silently cutting of inputs is a better way of dealing with it. If the message doesn't match up precisly then it should fail.

No, silently cutting inputs is the correct fix to the buffer overflow. The fix is independant of the context, and logical. Works the same as snprintf or any other 'safe' method: try to copy too much means some gets discarded. 

It changes it from a crash-bug into a usability quirk. Someone more familiar with the context can fix it some more, handle the error a bit cleaner, or log it or whatever. 

It feels weird that `&a` is the same as `&a[0]` or `a` (both these last ones I agree with). But I also guess that's the case, otherwise the original fix wouldn't have worked for @borjaburgos, and the compiler would probably have warned about it.

closed 2015-05-05 by holiman