Proxmark/proxmark3 #103: ISO14443B is not working anymore

buggii, 2015-05-24:
The actual code seems to have problems with ISO14443B; when sending commands [ex. hf 14b raw -c -p 06 00] to tags the pm3 hangs (solid red led) and resets itself after some seconds (10-15 seconds).

The code was surely working when 14b raw was introduced: https://github.com/Proxmark/proxmark3/commit/7cf3ef203cceb0d5caab8c01b87324080f0d8fd9

marshmellow42, 2015-05-31:
it appears @iceman1001 was correct.  this was broken during the bigbuf adjustment.  
it appears the code now gets stuck in an infinite loop at the for(;;) loop at ~line 689 in iso14443.c .
  the `while(((lastRxCounter-AT91C_BASE_PDC_SSC->PDC_RCR) & (DMA_BUFFER_SIZE-1)) > 2)` 
is never true.
it appears instead of DMA_BUFFER_SIZE (which = 128) this used to be DEMOD_DMA_BUFFER_SIZE ( which = 1024).

it appears the buffers were changed but some of the logic wasn't.

marshmellow42, 2015-05-31:
but it appears there are more issues dating back further...  have to dig back farther... :(

marshmellow42, 2015-05-31:
actually looks like it started crashing the pm3 at the tracing merge, @7d5ebac NOT the bigbuf.
(or somewhere inbetween...)

marshmellow42, 2015-05-31:
confirmed...  the issues start at 7d5ebac

holiman, 2015-05-31:
Ok, it's my commit, but @pwpiwi, do you have any input here? I'm a  bit confused as to the DMA_BUFFER_SIZE versus DEMOD_DMA_BUFFER_SIZE (I think the latter was removed).

pwpiwi, 2015-06-01:
I dont' have an ISO14443B tag to test. If someone can try my PR ... its an educated guess.

iceman1001, 2015-06-01:
hm,  strange commands 14b.. 
I tried hooking up one PM3 with "hf 14b sim"
and another one with "hf 14b read"

The exiting result of.. Nothing...
and no help texts either.. What is it that Im missing?

marshmellow42, 2015-06-01:
it appears the crashing is gone, so that is an improvement.  but I can't seem to get a tag to respond either.  but that may be my antenna.  i'll verify with an old code version.

marshmellow42, 2015-06-01:
btw the hf 14b read appears to be "Obsolete" ???  (at least what is says in the code.)  i'm attempting the raw command hf 14b raw -c -p 06 00    and hf 14b raw -c -p 0B

marshmellow42, 2015-06-01:
oy..  the 14b code is a mess.  seems like every version has some issue with one or more of the commands..

I found at commit 0318894 the snoop worked (mostly...) and was able to snoop another reader send the 05 00 08 39 73 and get a response from my tag.  however, that same version the hf 14b read command does not work on my setup.  (my antenna isn't very strong but have never had an issue before) I also tried the raw command at that version and again no response.

marshmellow42, 2015-06-01:
Update...  it appears the last time 14b raw, 14b snoop and 14b list were all working (not 14b read...) 
was 16b75f2.  just before bigbuf.  after that the commands didn't crash but didn't get anything back.  

at the same point the snoop began throwing a circular buffer error:
```
proxmark3> hf 14b snoop
#db# Snooping buffers initialized:
#db#   Trace: 34880 bytes
#db#   Reader -> tag: 2048 bytes
#db#   tag -> Reader: 2048 bytes
#db#   DMA: 1024 bytes
#db# blew circular buffer! behindBy=0x39a
#db# Snoop statistics:
#db#   Max behind by: 922
#db#   Uart State: 0
#db#   Uart ByteCnt: 0
#db#   Uart ByteCntMax: 100
#db#   Trace length: 0
```

at  7d5ebac the raw commands then crashed the whole client... (which has been fixed... I guess)

marshmellow42, 2015-06-01:
That said, in my testing from the beginning of the raw command (when it worked) with my antenna it always took sending the raw command at least 4 times on a cold start to get it to function. after that it seemed to function fairly well.

pwpiwi, 2015-06-02:
Found another one... and this time it was indeed me who introduced it with the Bigbuf rework. I didn't notice that the FPGA code for iso14443b indeed sends signed samples.

marshmellow42, 2015-06-02:
Good catch.  that gets the raw working!!! 

the snoop doesn't immediately blow the circular buffer, but it doesn't work.  (and it did blow the circular buffer a couple times on me...)
i'm getting "Tag | <empty trace - possible error>" in the hf list 14b after the snoop

marshmellow42, 2015-06-02:
but it does get the reader data sometimes...

marshmellow42, 2015-06-02:
and if anyone else is testing, ignore the hf 14b read command as it is coded only to call the raw function and never output anything...  an easy fix later.

iceman1001, 2015-06-02:
ok,  lets see..
I need to send what commands from which pm3?!?

pm3.nr1   hf 14b raw 05 00 08 39 73 
pm3.nr2    hf 14b raw -c -p 0B

marshmellow42, 2015-06-02:
Just the hf 14b raw 05 00 08 39 73  will get you the atqb response with the uid (or ruid)

iceman1001, 2015-06-02:
and the other device should do a sim?

marshmellow42, 2015-06-02:
sure.  I haven't tried the sim yet..  it would be good to check it as well
(i'm using a 14b reader to snoop test and an actual tag to read)

iceman1001, 2015-06-02:
I don't have that.  I use two pm3.. one for sim, one for read?!?

marshmellow42, 2015-06-02:
that should be a valid test, but if it fails you won't know which failed.  (except I guess we know the raw read now works [most of the time]...)

pwpiwi, 2015-06-02:
Indeed. That would be a final test. But I believe that we are not yet there. Having seen this:
```
	// This row make the error blew circular buffer in hf 14b snoop
	//if (Uart.state == STATE_ERROR_WAIT) LED_A_OFF(); // Error
```
If this small piece of code blew the circular buffer then there must be a serious timing issue - the statement itself cannot cause an issue.

pwpiwi, 2015-06-02:
All in all this piece of code is not very well maintained. Which isn't a surprise because ISO14443B tags seem to be very rare? I couldn't find a source except China?

buggii, 2015-06-02:
I can send you a spare one.
Unfortunately not so much time for testing those weeks (and previous) 

Inviata dal mio dispositivo Android.

https://github.com/Proxmark/proxmark3/issues/103#issuecomment-108063770

pwpiwi, 2015-06-02:
More fixes. Started debugging at the very low level (i.e. the FPGA).

marshmellow42, 2015-06-02:
Looks like the code was a little thrown together.  If you can get the low level stuff worked out I don't mind working on some of the interface stuff.  (General Info Cmd, maybe generic read...  Etc.)

marshmellow42, 2015-06-03:
the raw command still works but the snoop blows the buffer right away:
```
proxmark3> hf 14b snoop
#db# Snooping buffers initialized:
#db#   Trace: 39360 bytes
#db#   Reader -> tag: 256 bytes
#db#   tag -> Reader: 256 bytes
#db#   DMA: 128 bytes
#db# blew circular buffer! behindBy=0x74
#db# Snoop statistics:
#db#   Max behind by: 116
#db#   Uart State: 0
#db#   Uart ByteCnt: 0
#db#   Uart ByteCntMax: 256
#db#   Trace length: 0
proxmark3>
```

iceman1001, 2015-06-03:
I started a thread on the forum,  to discuss changes to 14b.
ref:  http://www.proxmark.org/forum/viewtopic.php?id=2496

pwpiwi, 2015-06-03:
Rats! I could have tested this myself - don't even need a card for that.

Next commit brings a slight speedup and doesn't blow the circular buffer anymore when idling:
```
proxmark3> hf 14b snoop
#db# Snooping buffers initialized:
#db#   Trace: 39360 bytes
#db#   Reader -> tag: 256 bytes
#db#   tag -> Reader: 256 bytes
#db#   DMA: 128 bytes
#db# cancelled
#db# Snoop statistics:
#db#   Max behind by: 2
#db#   Uart State: 0
#db#   Uart ByteCnt: 0
#db#   Uart ByteCntMax: 256
#db#   Trace length: 0
proxmark3>
```

marshmellow42, 2015-06-03:
snoop no longer blows the buffer, but I also don't get any info.  I noticed the one LED is on when snooping until I hit read on 3rd pty reader then LED goes off, but list has nothing in it.

```
proxmark3> hf 14b snoop
#db# Snooping buffers initialized:
#db#   Trace: 39360 bytes
#db#   Reader -> tag: 256 bytes
#db#   tag -> Reader: 256 bytes
#db#   DMA: 128 bytes
#db# cancelled
#db# Snoop statistics:
#db#   Max behind by: 12
#db#   Uart State: 0
#db#   Uart ByteCnt: 0
#db#   Uart ByteCntMax: 256
#db#   Trace length: 0
proxmark3> hf list 14b
Recorded Activity (TraceLen = 0 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

     Start |       End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
-----------|-----------|-----|-----------------------------------------------------------------|-----|--------------------|
proxmark3> hf 14b snoop
```

marshmellow42, 2015-06-03:
the raw command does work.

pwpiwi, 2015-06-03:
Which LED is on?

pwpiwi, 2015-06-03:
BTW: I will be offline for a few days...

marshmellow42, 2015-06-03:
looking at the pm3 it is the center led.  (sorry i'm color blind and i'm not positive which in the code it is at the moment)

marshmellow42, 2015-06-03:
it comes on when you tell it to snoop and off if you hit the button, or if it a 3rd pty reader sends a command to the tag.

marshmellow42, 2015-06-03:
`BTW: I will be offline for a few days...`
I understand, thanks for your hard work on this.  :)

marshmellow42, 2015-06-03:
the center LED, LED_A? goes on full with snoop command sent.  goes off on first "read" from 3rd pty reader, then flashes for every "read" from 3rd pty reader from then on.

running further tests if I hold the antenna a little away from the reader I can get some of the readers Commands:
(this was clicking "identify" on the 3rd pty reader about 30 times while snooping)
```
proxmark3> hf 14b snoop
#db# Snooping buffers initialized:
#db#   Trace: 39360 bytes
#db#   Reader -> tag: 256 bytes
#db#   tag -> Reader: 256 bytes
#db#   DMA: 128 bytes
#db# cancelled
#db# Snoop statistics:
#db#   Max behind by: 106
#db#   Uart State: 0
#db#   Uart ByteCnt: 0
#db#   Uart ByteCntMax: 256
#db#   Trace length: 280
proxmark3> hf list 14b
Recorded Activity (TraceLen = 280 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

     Start |       End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
-----------|-----------|-----|-----------------------------------------------------------------|-----|--------------------|
         0 |         0 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
   1432328 |   1432328 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
   7064164 |   7064164 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
   8971104 |   8971104 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
  14244444 |  14244444 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
  14721172 |  14721172 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
  16150000 |  16150000 | Rdr | 05  00  08 [39  73]                                             |  ok | REQB
  74400438 |  74400438 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
  77265086 |  77265086 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
  79875484 |  79875484 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
  81307802 |  81307802 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
  83445702 |  83445702 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
  85770782 |  85770782 | Rdr | 05  00  08 [39  73]                                             |  ok | REQB
  85772166 |  85772166 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
proxmark3>
```

marshmellow42, 2015-06-03:
the first command the reader should send is the 05 00 08 (reqb).. then after the tag responds it will follow up with the 1D [UID] [Param]... (attrib)

marshmellow42, 2015-06-12:
So we currently stand with the raw commands fixed, sim command not tested (will work on) and with the snoop command I can pick up some reader commands intermittently, but never any tag commands.

pwpiwi, 2015-06-12:
I am still on it. @buggii did send me some tags for testing but  they did not yet arrive...

marshmellow42, 2015-06-12:
Great!  I wasn't trying to rush things, I just wanted to summarize before I
forgot where we left off.

iceman1001, 2015-06-13:
@marshmellow42  we could always look into the 15 commandes instead...

pwpiwi, 2015-06-15:
@buggi's tag arrived. Unfortunately I cannot get anything out of it. Not with the current or an ancient version of PM3 and not with an Omnikey Card Reader. Trying to build a smaller antenna...

iceman1001, 2015-06-15:
try the fpgahi image from before your changes in january ;) maybe it works better...

marshmellow42, 2015-06-15:
If it is a st tag it won't respond to a 05 00 08 command.  But a 06 00 instead.

marshmellow42, 2015-06-15:
I have noticed positioning and distance are sometimes finicky to get a good read.    but I assumed it was my antenna.  btw, is there anyway to "see" the wave the HF side is demoding?  (possibly see why different reads with the same positioning will sometimes fail?)

buggii, 2015-06-15:
Unfortunately the tag is really small and piwi need a smaller antenna just like the one jonor suggested in a post.

Inviata dal mio dispositivo Android.

https://github.com/Proxmark/proxmark3/issues/103#issuecomment-112153255

marshmellow42, 2015-06-16:
could this have anything to do with our issues:
http://www.proxmark.org/forum/viewtopic.php?pid=16548#p16548 :
"I see in revision 655 the change of optimization parameter in common/Makefile.common from -O2 to -Os.
I tried to set -O2 instead of -Os and now all is working with ISO1444B also sri and srix4k commands.
I tried to understand the problem without success, so -O2 for me is good solution."

piwi mentioned there are timing issues, could changing optimization for speed for this code help that?

iceman1001, 2015-06-16:
@marshmellow42, did you test the -02 change with your branch?

marshmellow42, 2015-06-16:
i haven't gotten the time yet.  but i plan to.

marshmellow42, 2015-06-16:
and that probably is just a band-aide anyway

pwpiwi, 2015-06-17:
Not necessarily. It speeds up the code and therefore avoids circular buffer overflow. But the speedup can be achieved by other means as well.

I now have a small antenna, fixed some bugs, and can read the SRIX4K tag very reliabel (no positioning issue - works to a distance of up to 1cm). Let me clean up the code a bit - maybe I can commit today evening.

Unfortunately I cannot snoop because my card reader doesn't talk to SRIX4K tags (it emits 05 00 08 instead of 06 00). But even if the tag would understand the reader, I think that there would be the same problem with antenna size (this time reader vs tag).

buggii, 2015-06-17:
Android should support iso14443b but in tests i did no one (Android OSes) seems to read srix. A good srix reader is SL500F but i don't think it worth the cost just to study that kind of tag.

Inviata dal mio dispositivo Android.

https://github.com/Proxmark/proxmark3/issues/103#issuecomment-112664023

marshmellow42, 2015-06-17:
i know how much easier it is to have something right in front of you, but if i can help in any way let me know, i can test with a 3rd pty reader, a 14b std tag, 2 pm3's.  we might be able to work out a time to work on it at the same time so you don't have to wait for me, just let me know when and i'll see if i can squeeze it in.

pwpiwi, 2015-06-17:
Pushed another bunch of changes. We are slowly getting there. hf 14b raw, hf 14b srix4kread now works for me. Code is still not fast enough for reliable snoops though.

Next step: I plan to remove the parity calculation to speed up the snoop loop. ISO14443B doesn't use parity bits - its only calculated to fit in the uniform tracing function and avoid exclamation marks when displaying with hf list.

marshmellow42, 2015-06-18:
hf 14b raw got a LOT more stable/reliable.  you are definitely getting it done!  :)

pwpiwi, 2015-06-18:
Abstaining from calculating unused parity bits should have solved the buffer overflow in snoop. If someone could test, please. (due to the lack of a matching reader-tag pair, I can only snoop the reader side of the communication - which was successful).

I have deleted all non-working commands which are obviously based on an ancient concept (reading into buffer, then use graphing functions - just like in LF - but HF is too fast for this approach).

There is still a lot to do (e.g. hf 14b reader) but I think the groundworks are there now to do this on the client side. The hf 14b write command may need to be renamed (its for ST tags only), can be moved to the client side. hf 14b write currently also relies on a previous hf 14b srix4k command which leaves the field on and the tag active - this could be eliminated by adding an anticollision phase to the hf 14b write command.

marshmellow42, 2015-06-18:
I'll test in an hour or two...  I do have a reader command built and ready in my fork.

marshmellow42, 2015-06-18:
looks like we are still not getting the tag data, but we are getting all the reader data:
```
proxmark3> hf 14b snoop
#db# Snooping buffers initialized:
#db#   Trace: 39360 bytes
#db#   Reader -> tag: 256 bytes
#db#   tag -> Reader: 256 bytes
#db#   DMA: 128 bytes
#db# cancelled
#db# Snoop statistics:
#db#   Max behind by: 47
#db#   Uart State: 0
#db#   Uart ByteCnt: 0
#db#   Uart ByteCntMax: 256
#db#   Trace length: 140
proxmark3> hf list 14b
Recorded Activity (TraceLen = 140 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

     Start |       End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
-----------|-----------|-----|-----------------------------------------------------------------|-----|--------------------|
         0 |         0 | Rdr | 05  00  08 [39  73]                                             |  ok | REQB
      1388 |      1388 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
    476690 |    476690 | Rdr | 05  00  08 [39  73]                                             |  ok | REQB
    478074 |    478074 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
    991514 |    991514 | Rdr | 05  00  08 [39  73]                                             |  ok | REQB
    992898 |    992898 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
   1472442 |   1472442 | Rdr | 05  00  08 [39  73]                                             |  ok | REQB
   1473828 |   1473828 | Rdr | 1d  70  b4  17  a2  00  03  01  00 [68  a0]                     |  ok | ATTRIB
```

[61 more replies]

closed 2015-06-23 by pwpiwi