Proxmark/proxmark3 #104: Huge slowdown after issuing some commands

jmichelp, 2015-05-24:
I was trying the new commands on mfu and discovered some strange behavior.

Here are the steps I followed:
1. plug the PM3
2. launch the client
3. hf mfu info
-> takes a couple of seconds at most
4. hf mfu
-> takes a minute to print the help
5. hf mfu info
-> takes a minute (instead of seing the red light blinking, it goes on and off at a very slow pace)!

After that, pretty much all commands on pm3 are really slow to execute.
The only way I've found so far to return to normal state is to restart the pm3 by unplugging it.

I also managed to get the same behavior after issuing "hf mfu info" 16 times in row. The last one is slowed down. So it seems to be related to some buffer that help commands can fill faster...

marshmellow42, 2015-05-24:
Check the memory usage of the client on your PC.  If it expands beyond the available memory it could be paging to your HD... ?

iceman1001, 2015-05-24:
try exit your client and see if your proxmark3.log is big?  it is big, rename or delete it.
then try running the client.

In my setup, the client works with out slowing down.
however if you run against a locked down tag the gui starts printing but it stops at the "tag Counters" for a moment.  This is normal, since the tag resets and the timeouts is long.

The new "hf mfu info" works with collection serveral differnt sorts of information, read from the tag. so its not a "fast" command to start with.

jmichelp, 2015-05-24:
Proxmark.log is ~500kB so I don't think this qualifies as "big". Regarding swapping, I have 8GB free RAM out of 16GB.

And I forgot to mention that I ran the command with always the same tag in place.
Once slowed down, it's slow even for displaying help about commands.

```
proxmark3> hw version
#db# Prox/RFID mark3 RFID instrument
#db# bootrom: master/v2.0.0-19-g2c772e6-suspect 2015-04-07 20:46:16
#db# os: master/v2.0.0-94-gf0eaac9-suspect 2015-05-24 12:11:11
#db# HF FPGA image built on 2015/03/09 at 08:41:42
uC: AT91SAM7S256 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
```

I will try under Linux, to see if this could be an issue with the FTDI driver.

buggii, 2015-05-24:
500kb of text data... it is indeed big!

Inviata dal mio dispositivo Android.

https://github.com/Proxmark/proxmark3/issues/104#issuecomment-105028978

holiman, 2015-05-24:
No, it's tiny. And not in memory anyway...

holiman, 2015-05-24:
Sounds like arm-side. Could you check usb-comms with wireshark? Works under linux

iceman1001, 2015-05-24:
do you have the output of the commmand?  Which type?

the "hf mfu info" command uses the  "hf 14a raw" command in the bottom and to call the device-side.

If it would be device-side, then the "hf 14a raw" could has issues, which I don't think.

Just to make sure,  you did a  make clean && make all & flashed osimage first.

marshmellow42, 2015-05-25:
it would help to know the type of tag to narrow it down.  as iceman said *most* of the mfu info calls use the unchanged hf 14a raw command.  UL_C uses some new device side auth procedures in some circumstances.  also, have you noticed this type of issue with any other command (if you never run the new mfu commands)?  I haven't been able to duplicate this issue yet, but I haven't had time to run a full battery of tests (like 16 mfu info cmds in a row).

iceman1001, 2015-05-25:
I have run the "hf mfu info"  together with "hf mfu dump"  and "hf 14a rea" on 9 different tags after eachother.   It never slows down,

However, the new tag detection is a bit slow,  you can notice it when you run "hf 14a read.
It starts printing, then it detects type via the new detection (only the new tags we added) and there is a timeout,  looks like the commnad pauses,  and then prints the rest.

That is all I found,  so I can't replicate your issue.

jmichelp, 2015-05-25:
The issue seemed to be related to the compiled image under Mac OS X.

I compiled and flashed the image under linux and the issue seemed to have vanished.

closed 2015-05-25 by jmichelp

iceman1001, 2015-05-25:
@buggii did just confirm the slowdown.

Using the GUI, there is a slowdown.  And I only tested it with the client cli and there is no slowdown. @buggii  confirms that too.

marshmellow42, 2015-05-25:
That cannot be the same issue as @jmichelp as he wasn't using the GUI.  ?

iceman1001, 2015-05-25:
Then we had two different issues that relates to client becomes slow.

One on Mac OS,  
One on PC for GUI.

marshmellow42, 2015-05-25:
@jmichelp, did you try the client on Mac os x with the firmware flashed from Linux, and did it still have an issue?

buggii, 2015-05-25:
In gui it got resolved pressing the "clear" button inside the gui itself.

Inviata dal mio dispositivo Android.

https://github.com/Proxmark/proxmark3/issues/104#issuecomment-105229322

jmichelp, 2015-05-25:
I've never used the GUI, only the CLI. And yes, before closing the issue, I compiled/flashed under Linux and then tried the CLI under Mac OS X to ensure the problem was solved.

iceman1001, 2015-05-25:
good that it solved itself,  would be interesting to know what caused it.

cynix, 2016-09-10:
I'm having the same issue on OS X, and compiling/flashing firmware under Linux doesn't help. If I use the CLI on OS X, all commands will be very slow after a while. Using the CLI on Linux doesn't seem to have this problem.

qaisjp, 2018-01-08:
Getting massive slowdowns on macOS as well :(

Quitting and reopening the client whenever a command doesn't run fast enough allows me to re-enter the command and get a fast response. No idea what to do.

@cynix what makefile modifications did you make?

qaisjp, 2018-01-08:
Building my own version worked well. Just comment out these lines: https://github.com/Proxmark/proxmark3/blob/d03fb29/client/Makefile#L40-L68

pwpiwi, 2018-01-09:
... which results in a client without graphing functions. Most of the `lf` commands are pretty much useless then.