Proxmark/proxmark3 #105: `hf mf sim` does not work while `hf 14a sim` does

p-l-, 2015-05-24:
Not sure where the problem is.

When I run `hf mf sim`, nothing seems to happen (I have tried several readers), but `hf 14a sim` works with all my readers.

```
proxmark3> hw version
#db# Prox/RFID mark3 RFID instrument                 
#db# bootrom: master/v2.0.0-46-g4b36037-suspect 2015-05-20 20:51:58                 
#db# os: master/v2.0.0-46-g4b36037-suspect 2015-05-20 20:52:00                 
#db# HF FPGA image built on 2015/03/09 at 08:41:42                 
uC: AT91SAM7S256 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          

proxmark3> hf mf eload empty
................................................................
Loaded 64 blocks from file: empty.eml          

proxmark3> hf mf sim i
 uid:N/A, numreads:0, flags:1 (0x01)           
Press pm3-button to abort simulation          
#db# 4B UID: 01234567                 
[nothing happens: no LED on, the reader does not "see" the simulated card, I press the button]
#db# Emulator stopped. Tracing: 1  trace length: 0                  

proxmark3> hf 14a sim 1
Emulating ISO/IEC 14443 type A tag with 4 byte UID (00000000)          
[yellow LED on, a lot of "#db# Auth attempt {nr}{ar}:" lines when I try to read the simulated card, and the reader "sees" a Mifare Classic 1K with NUID 00000000]
```

Any ideas about what could happen here? Thanks!

iceman1001, 2015-05-24:
You might want to move your antenna off the reader before running the "hf mf sim" command.
And then put the antenna on to the read, with some distance

p-l-, 2015-05-24:
Yeah I have tried this several times... I'll try again. Is it normal during `hf mf sim` that nod LED is on?

iceman1001, 2015-05-24:
if I remember it,  my leds are off until it detects some field changes.

p-l-, 2015-05-24:
If that may help, here is the trace I get with `hf 14a sim` (working):
```
         0 |      1056 | Rdr | 26                                                              |     | REQA          
      2228 |      4596 | Tag | 04  00                                                          |     |           
     11426 |     13890 | Rdr | 93  20                                                          |     | ANTICOLL          
     15062 |     20886 | Tag | 00  00  00  00  00                                              |     |           
     27684 |     38212 | Rdr | 93  70  00  00  00  00  00  9c  d9                              |     | SELECT_UID          
     39384 |     42904 | Tag | 08  b6  dd                                                      |     |           
```

And here is what I get with `hf mf sim i` (same reader)

```
        0 |      1056 | Rdr | 26                                                              |     | REQA          
      2484 |      4852 | Tag | 04  00                                                          |     |           
  38402346 |  38403338 | Rdr | 69                                                              |     | ?          
  89468522 |  89469514 | Rdr | 69                                                              |     | ?          
  96696868 |  96697924 | Rdr | 26                                                              |     | REQA          
  96699480 |  96701848 | Tag | 04  00                                                          |     |           
 104557534 | 104558590 | Rdr | 26                                                              |     | REQA          
 104560018 | 104562386 | Tag | 04  00                                                          |     |           
 157243806 | 157244862 | Rdr | 26                                                              |     | REQA          
 157246226 | 157248594 | Tag | 04  00                                                          |     |           
```

It seems to me that the reader does not get the `04 00` response to its `26` request with `hf mf sim`, while it does get it with `hf 14a sim`. What do you think?

iceman1001, 2015-05-24:
It could be a antenna position issue..

p-l-, 2015-05-24:
Then why would `hf 14a sim` work perfectly in the same conditions? Do both command use different signal levels?

iceman1001, 2015-05-24:
they use different kind of field-detection.. edge..  but whatever, if you tested it with different pos and its not work, then it doesnt work for you.
how is your "hw tune" output?

p-l-, 2015-05-24:
```
# HF antenna: 12,16 V @    13.56 MHz          
```

iceman1001, 2015-05-25:
I tried some "hf mf sim" yesterday, and I don't have an issue with it.
eload, sim,   take the antenna over the reader,  and it flashes lights when the reader tries to read,   and press  button,  then "hf list 14a" to download trace.   All works well.

p-l-, 2015-05-25:
OK. Does `12,16 V` seems like a reasonable value to you?

iceman1001, 2015-05-25:
is that with tag on antenna or off ;)

Yes, its not the best but good.  Its when you only have 6-8v where the problems arises

p-l-, 2015-05-25:
Without tag on the antenna, I forgot to mention.

iceman1001, 2015-05-25:
And you tested different distance between antenna and reader.
And you have the latest source, and flashed arm..
hm,  what type of reader do you have?

And do you have two pm3?  where one could sim, and the other be a reader or sniff.

p-l-, 2015-05-25:
Unfortunately I only own one PM3. I'll try to find someone else to test.

BTW, can you tell me with which exact firmware & client versions you have run your test? I'd like to try with the same versions.

iceman1001, 2015-05-25:
I run my fork..  
If you can't get it to work,  try @marshmellow42  fork instead, its more aligned with PM3 master..

p-l-, 2015-05-25:
OK I've just tried with your fork, same result. I'll try to find other readers.

I suppose there is no (easy) way to use the field detection from `hf 14a sim` in `hf mf sim`?

iceman1001, 2015-05-25:
This trace is from my setup, (fork and antenna  16.91v)  right now.
It works fine

pm3 --> hf li 14a
Recorded Activity (TraceLen = 3936 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |

------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|

          0 |       1056 | Rdr |26                                                               |     | REQA
       2420 |       4788 | Tag |44  00                                                           |     |
      57086 |      59550 | Rdr |93  20                                                           |     | ANTICOLL
      61170 |      66994 | Tag |88  04  e9  36  53                                               |     |
     131492 |     141956 | Rdr |93  70  88  04  e9  36  53  d6  6c                               |  ok | SELECT_UID
     143640 |     147160 | Tag |04  da  17                                                       |     |
     196542 |     199006 | Rdr |95  20                                                           |     | ANTICOLL-2
     200690 |     206578 | Tag |62  8f  33  80  5e                                               |     |
     270910 |     281438 | Rdr |95  70  62  8f  33  80  5e  00  f7                               |  ok | ANTICOLL-2
     282994 |     286514 | Tag |08  b6  dd                                                       |     |
     334432 |     339200 | Rdr |50  00  57  cd                                                   |  ok | HALT
     549952 |     551008 | Rdr |26                                                               |     | REQA
     771456 |     771808 | Rdr |02                                                               |     | ?

iceman1001, 2015-05-25:
The antenna is about 1cm from reader.

iceman1001, 2015-05-25:
however,  lets check something.
can u email me,  iceman at iuse dot se

Gator96100, 2016-03-29:
I'm having the same problem, did you manage to fix it?

iceman1001, 2016-03-29:
I got what I wanted out from the  "hf 14a sim" so I never bothered to dig more into the "hf mf sim" and why it doesn't work so well.

iceman1001, 2016-04-12:
When I found the bug for "hf mf mifare",   it seems to have fixed this one aswell..

pm3 --> hf mf sim i
 uid:N/A, numreads:0, flags:1 (0x01)
Press pm3-button or send another cmd to abort simulation
#db# 4B UID: 00000000
#db# Emulator stopped. Tracing: 1  trace length: 133
pm3 --> hf list 14a
Recorded Activity (TraceLen = 133 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |

------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|

          0 |        992 | Rdr |52                                                               |     | WUPA
       2676 |       5044 | Tag |04  00                                                           |     |
       7040 |       9504 | Rdr |93  20                                                           |     | ANTICOLL
      11252 |      17076 | Tag |00  00  00  00  00                                               |     |
      19584 |      30112 | Rdr |93  70  00  00  00  00  00  9c  d9                               |  ok | SELECT_UID
      31604 |      35124 | Tag |08  b6  dd                                                       |     |
    1479870 |    1484638 | Rdr |e0  80  31  73                                                   |  ok | RATS
    1486194 |    1486834 | Tag |04                                                               |     |
    2766296 |    2767288 | Rdr |40                                                               |     | MAGIC WUPC1
    2904280 |    2905592 | Rdr |43                                                               |     | MAGIC WUPC2
    3042648 |    3047416 | Rdr |50  00  57  cd                                                   |  ok | HALT
pm3 -->

marshmellow42, 2016-04-12:
What exactly was the bug and fix in the code?

iceman1001, 2016-04-12:
Good question.  Lately there has been a bunch of users  both sides not wanting to share and complaining about hardnested etc that I feel reluctant to share anymore my stuff.

Gator96100, 2016-04-13:
Please @iceman1001 share your findings, I need to have a working mifare emulation in order to finish my school project. Be not as selfish as other users and share your findings in order to enable a better proxmark3 experience for everybody.

iceman1001, 2016-04-14:
The "hf 14a sim" works, so your school project is save.
And a question,  what have you brought to the proxmark project? I fail to notice your username in the contributors list.

Gator96100, 2016-04-14:
The "hf 14a sim" command unfortunately isn't as powerful and would require modifications in order to work for my school project.
The reason I hadn't had a chance to contribute to this project is that I got my proxmark 3 weeks ago.

holiman, 2016-04-14:
Everyone doesn't have to contribute, that's all right. I'm using 100s of softwares I'm not contributing to. I don't understand this whole discussion.

iceman1001, 2016-04-14:
Thank you for your understanding.

merlokk, 2017-10-20:
#412 ?

pwpiwi, 2017-10-21:
Yes. Same issue is taken up in #412 again.

Closing this one as duplicate because #412 is based on much more recent software.

closed 2017-10-21 by pwpiwi