Proxmark/proxmark3 #110: Compiling error on OSX. Missing readline

zhovner, 2015-05-27:
My environment:
Mac OS 10.10
readline 6.3.8 installed from `brew` located at `/usr/local/Cellar/readline/6.3.8`

```
gcc -std=c99 -I. -I../include -I../common -I/opt/local/include -I../liblua -I/usr/local/Cellar/readline/6.3.6/include/ -Wall  -g -O4 -DHAVE_GUI -c -o obj/ui.o ui.c
clang: warning: -O4 is equivalent to -O3
ui.c:48:19: error: use of undeclared identifier 'rl_readline_state'
        int need_hack = (rl_readline_state & RL_STATE_READCMD) > 0;
                         ^
ui.c:48:39: error: use of undeclared identifier 'RL_STATE_READCMD'
        int need_hack = (rl_readline_state & RL_STATE_READCMD) > 0;
                                             ^
ui.c:52:16: warning: implicit declaration of function 'rl_copy_text' is invalid in C99 [-Wimplicit-function-declaration]
                saved_line = rl_copy_text(0, rl_end);
                             ^
ui.c:52:14: warning: incompatible integer to pointer conversion assigning to 'char *' from 'int' [-Wint-conversion]
                saved_line = rl_copy_text(0, rl_end);
                           ^ ~~~~~~~~~~~~~~~~~~~~~~~
ui.c:53:3: warning: implicit declaration of function 'rl_save_prompt' is invalid in C99 [-Wimplicit-function-declaration]
                rl_save_prompt();
                ^
ui.c:54:3: warning: implicit declaration of function 'rl_replace_line' is invalid in C99 [-Wimplicit-function-declaration]
                rl_replace_line("", 0);
                ^
ui.c:66:3: warning: implicit declaration of function 'rl_restore_prompt' is invalid in C99 [-Wimplicit-function-declaration]
                rl_restore_prompt();
                ^
5 warnings and 2 errors generated.
make[1]: *** [obj/ui.o] Error 1
make: *** [client/all] Error 2
```

I fix this error by adding readline path to `proxmark3/client/Makefile`

```
LDLIBS = -L/usr/local/Cellar/readline/6.3.8/lib/ -L/opt/local/lib -L/usr/local/lib ../liblua/liblua.a -lreadline -lpthread -lm
CFLAGS = -std=c99 -I. -I../include -I../common -I/opt/local/include -I../liblua -I/usr/local/Cellar/readline/6.3.8/include/ -Wall $(COMMON_FLAGS) -g -O4
```

VERTCraig, 2015-07-14:
Using MacPorts for readline I don't have to modify any makefiles.  I have readline @6.3.003_0 (active).

Perhaps try installing realine via ports and see if you have better luck?

closed 2015-07-16 by holiman

daveio, 2016-01-12:
I ran into this bug just now. The issue with Homebrew readline is that it's keg-only so that it doesn't interfere with OS X stock readline.

Before building the proxmark code, do

`brew link --force readline`

Then after building reverse the above with

`brew unlink readline`

Tested and verified.

commit 82e8eee6 (2017-06-30): Merge pull request #333 from micolous/osx-libedit

OSX: Disable GNU readline-specific hack when libedit is used (fixes #110)
commit ed50f7f3 (2017-06-30): OSX: Disable GNU readline-specific hack when libedit is used (fixes #110)