Proxmark/proxmark3 #116: Compile error on Archlinux

rooti, 2015-06-13:
Hi
Im getting following compile error. Tested on fully updated archlinux. Any suggestions on how to fix this? thx

------- snip --------
```
arm-none-eabi-gcc -c -I../include -I../common -Wall -Werror -pedantic -std=c99 -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG  -DWITH_CRC -DON_DEVICE -fno-strict-aliasing -ffunction-sections -fdata-sections -I. -Os -mthumb-interwork -o obj/optimized_cipher.o optimized_cipher.c 
In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                 from /usr/arm-none-eabi/include/stdio.h:48,
                 from optimized_cipher.c:64:
/usr/arm-none-eabi/include/sys/_stdint.h:31:19: error: conflicting types for 'int32_t'
 typedef __int32_t int32_t ;
                   ^
In file included from optimized_cipher.h:3:0,
                 from optimized_cipher.c:63:
./stdint.h:16:18: note: previous declaration of 'int32_t' was here
 typedef int      int32_t;
                  ^
In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                 from /usr/arm-none-eabi/include/stdio.h:48,
                 from optimized_cipher.c:64:
/usr/arm-none-eabi/include/sys/_stdint.h:32:20: error: conflicting types for 'uint32_t'
 typedef __uint32_t uint32_t ;
                    ^
In file included from optimized_cipher.h:3:0,
                 from optimized_cipher.c:63:
./stdint.h:21:24: note: previous declaration of 'uint32_t' was here
 typedef unsigned int   uint32_t;
                        ^
../common/Makefile.common:83: recipe for target 'obj/optimized_cipher.o' failed
make[1]: *** [obj/optimized_cipher.o] Error 1
make[1]: Leaving directory '/tmp/proxmark3/armsrc'
Makefile:11: recipe for target 'armsrc/all' failed
make: *** [armsrc/all] Error 2
```
------- snip --------

mpouyann, 2015-06-14:
Hello rooti,

Are you launching make from promark root directory or from client subdirectory ?

rooti, 2015-06-14:
im launching make from the promark root directory

mpouyann, 2015-06-14:
Did you try a make directly from /client subfolder ?

rooti, 2015-06-15:
make works in the /client subdirectory, but not in /

mpouyann, 2015-06-15:
Same on kali... I was assuming it as normal behavior... but is it needed to compile from root if client compilation leads to a working ./proxmark tool ?

pwpiwi, 2015-06-18:
There should be no `#include <stdio.h>` in ARM sources (how would you handle files there?). I have removed them. Let me know if this fixes the issue.

rooti, 2015-06-25:
Your fix didnt resolve the issue. Still the same compile error.

pwpiwi, 2015-06-25:
Please check or better repost your compile error. It can't be this same

```
In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                 from /usr/arm-none-eabi/include/stdio.h:48,
                 from optimized_cipher.c:64:
```

because there is no longer an `#include <stdio.h>` in line 64 of optimized_cipher.c

rooti, 2015-06-25:
ah yes. sorry you are right. here is the output:

```
arm-none-eabi-gcc -c -I../include -I../common -Wall -Werror -pedantic -std=c99 -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG  -DWITH_CRC -DON_DEVICE -fno-strict-aliasing -ffunction-sections -fdata-sections -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -I. -Os -mthumb-interwork -o obj/optimized_cipher.o optimized_cipher.c 
In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                 from /usr/arm-none-eabi/include/time.h:27,
                 from optimized_cipher.c:68:
/usr/arm-none-eabi/include/sys/_stdint.h:31:19: error: conflicting types for 'int32_t'
 typedef __int32_t int32_t ;
                   ^
In file included from optimized_cipher.h:3:0,
                 from optimized_cipher.c:63:
./stdint.h:16:18: note: previous declaration of 'int32_t' was here
 typedef int      int32_t;
                  ^
In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                 from /usr/arm-none-eabi/include/time.h:27,
                 from optimized_cipher.c:68:
/usr/arm-none-eabi/include/sys/_stdint.h:32:20: error: conflicting types for 'uint32_t'
 typedef __uint32_t uint32_t ;
                    ^
In file included from optimized_cipher.h:3:0,
                 from optimized_cipher.c:63:
./stdint.h:21:24: note: previous declaration of 'uint32_t' was here
 typedef unsigned int   uint32_t;
                        ^
../common/Makefile.common:84: die Regel für Ziel „obj/optimized_cipher.o“ scheiterte
make[1]: *** [obj/optimized_cipher.o] Fehler 1
make[1]: Verzeichnis „/tmp/proxmark3/armsrc“ wird verlassen
Makefile:10: die Regel für Ziel „armsrc/all“ scheiterte
make: *** [armsrc/all] Fehler 2
```

pwpiwi, 2015-06-25:
Removed the `#include <time.h` as well. Please try again.

rooti, 2015-06-25:
works without compile errors. thanks

closed 2015-06-25 by rooti

t3zla, 2015-12-11:
I am now getting the same error on arch. here it is:

    In file included from ./string.h:15:0,
                     from ../common/protocols.c:2:
    ./stdint.h:16:18: error: conflicting types for 'int32_t'
     typedef int      int32_t;
                      ^
    In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                     from /usr/arm-none-eabi/include/strings.h:13,
                     from ../common/protocols.c:1:
    /usr/arm-none-eabi/include/sys/_stdint.h:31:19: note: previous declaration of 'int32_t' was here
     typedef __int32_t int32_t ;
                       ^
    In file included from ./string.h:15:0,
                     from ../common/protocols.c:2:
    ./stdint.h:21:24: error: conflicting types for 'uint32_t'
     typedef unsigned int   uint32_t;
                        ^
    In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                     from /usr/arm-none-eabi/include/strings.h:13,
                     from ../common/protocols.c:1:
    /usr/arm-none-eabi/include/sys/_stdint.h:32:20: note: previous declaration of 'uint32_t' was here
     typedef __uint32_t uint32_t ;
                    ^
    ../common/Makefile.common:81: recipe for target 'obj/protocols.o' failed
    make[1]: *** [obj/protocols.o] Error 1
    make[1]: Leaving directory '/home/t3zla/Documents/Projects/AUR/proxmark/proxmark3/armsrc' 
    Makefile:10: recipe for target 'armsrc/all' failed
    make: *** [armsrc/all] Error 2

t3zla, 2015-12-16:
I currently get this error when trying to install on arch
    In file included from ./string.h:15:0,
                     from ../common/protocols.c:2:
    ./stdint.h:16:18: error: conflicting types for 'int32_t'
     typedef int      int32_t;
                      ^
    In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                     from /usr/arm-none-eabi/include/strings.h:13,
                     from ../common/protocols.c:1:
    /usr/arm-none-eabi/include/sys/_stdint.h:31:19: note: previous declaration of 'int32_t' was here
     typedef __int32_t int32_t ;
                       ^
    In file included from ./string.h:15:0,
                     from ../common/protocols.c:2:
    ./stdint.h:21:24: error: conflicting types for 'uint32_t'
     typedef unsigned int   uint32_t;
                        ^
    In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                     from /usr/arm-none-eabi/include/strings.h:13,
                     from ../common/protocols.c:1:
    /usr/arm-none-eabi/include/sys/_stdint.h:32:20: note: previous declaration of 'uint32_t' was here
     typedef __uint32_t uint32_t ;
                    ^
    ../common/Makefile.common:81: recipe for target 'obj/protocols.o' failed
    make[1]: *** [obj/protocols.o] Error 1
    make[1]: Leaving directory '/home/t3zla/Documents/Projects/AUR/proxmark/proxmark3/armsrc' 
    Makefile:10: recipe for target 'armsrc/all' failed
    make: *** [armsrc/all] Error 2