Proxmark/proxmark3 #126: "hf 15 xxx" issues

ikarus23, 2015-06-30:
Hi,
I have trouble reading a Type V tag (SL2 ICS2001, ICODE SLI) with version 377d692be1ee14b2377bc7fe3ad7f2486447c778 (`hf 15 reader`). Only in some rare cases I'm able to recover the UID. However, using version 9cefee6f42196a117cee2ad917ec4bedd88fdab3 everything works fine (at least way better than 377d692be1ee14b2377bc7fe3ad7f2486447c778).

And by the way: Does anyone know how to use the emulator (`hf 15 sim`)? Is it possible to emulate a full tag (UID + Memory)?

iceman1001, 2015-06-30:
When it comes to ( hf 15 sim )  see: #20 
Its not in the best shape yet.  I suspect that the hf 15 commands suffer from the BigBuffer modifications done. It will be dealt with after the "hf 14b" is fixed...

ikarus23, 2015-06-30:
Oh, I completely missed https://github.com/Proxmark/proxmark3/issues/20.
Thanks for pointing that out.

closed 2016-10-06 by iceman1001