Proxmark/proxmark3 #135: Card Simulation Ends Prematurely

volkjohn, 2015-08-03:
When trying to simulate a card, whether stored in buffer or from tag id, the simulation only last a split second. The orange led just flashes then the command line returns. I expect the simulation to go until the on board button is pressed. I have tried with the latest master commit and v2.2.0.

```
proxmark3> lf sim
Sending [39999 bytes]...............................................................................
Starting to simulate          
#db# Stopped                 
proxmark3> 
```

holiman, 2015-08-03:
This is likely the same issue that was discussed in #131, @VERTCraig reported:
> The original problem for me was that usb_poll() would return true and the simulation was aborted before it began. 

I made a patch for that, it was never confirmed if the patch solved the issue or not. Could you verify the 'hw version', which should be displayed immediately at startup of the client (and therefore it should be in your proxmark-log also)? I want to be sure that you tested it _with_ the patch from PR #131. 

Additionally, what OS (and/or virtualization) are you using?

volkjohn, 2015-08-03:
My apologizes @holiman, the latest master commit **does** work. I was running v2.0 firmware.
```
#db# Prox/RFID mark3 RFID instrument                 
#db# bootrom: master/v2.0.0-133-g9f9b6b7-suspect 2015-06-19 13:32:39                 
#db# os: master/v2.0.0-133-g9f9b6b7-suspect 2015-06-19 13:32:40                 
#db# HF FPGA image built on 2015/03/09 at 08:41:42                 
Prox/RFID mark3 RFID instrument          
          
uC: AT91SAM7S256 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 0 bytes ( 0). Free: 262144 bytes (100).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 256K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> 
```
After reflashing the latest firmware, it worked correctly.
```
Prox/RFID mark3 RFID instrument          
bootrom: master/v2.2.0-52-g5d01f50-suspect 2015-08-03 20:41:32
os: master/v2.2.0-52-g5d01f50-suspect 2015-08-03 20:41:34
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2015/06/22 at 21:47:54
          
uC: AT91SAM7S256 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 165670 bytes (63). Free: 96474 bytes (37).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 256K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3>
```
And just to make sure, I reverted to v2.2.0 firmware and client software, and it did not work. So your patch did in fact solve this problem for me (unless there is something else that changed it between v2.2.0 and HEAD). Thanks!
```
proxmark3> hw version
Prox/RFID mark3 RFID instrument          
bootrom: HEAD/v2.2.0-suspect 2015-08-03 20:56:43
os: HEAD/v2.2.0-suspect 2015-08-03 20:56:45
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2015/06/22 at 21:47:54
          
uC: AT91SAM7S256 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 160878 bytes (61). Free: 101266 bytes (39).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 256K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> 
```

closed 2015-08-03 by volkjohn

holiman, 2015-08-03:
Yay, it was good to get that confirmed, so no need to apologise :)