Proxmark/proxmark3 #136: lf t55xx read command serious bug

tbox1911, 2015-08-13:
Hi,
if by accident you issue this command on a blank/no passwd tag:

lf t55 read 0 51243648
(which IMHO, should not modify the configuration)

that change something (the block0 ?),  make the card  unreadable   !!

look the scenario:
from a never used card

proxmark3> lf t55 detect
clk 255         
Modulation : ASK         
Bit Rate   : 3 - RF/40         
Inverted   : No         
Offset     : 1         
Block0     : 0x000C8040         
         
proxmark3> lf t55 dump
[0] 0x000C8040  00000000000011001000000001000000         
[1] 0x12345678  00010010001101000101011001111000         
[2] 0x00000000  00000000000000000000000000000000         
...
proxmark3> lf t55 read 0 51243648
proxmark3> lf t55 detect
Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'

At this point the card is unreadable/unusable.

Workaroud: 
Iceman and me,  make our tag reappear with a all zeros pwd.
lf t55 write 0 000c8040 0

see: http://www.proxmark.org/forum/viewtopic.php?pid=17589#p17589

Regards

marshmellow42, 2015-10-15:
this appears to not exactly be a bug but a transponder design issue.  the read with password command is identical in it's makeup and structure as the std write command when the chip is in Fixed-bit-length protocol mode. (which is the standard mode)  

one work around i've seen with another reader is if the modulation and clock are known it attempts to read block 0 without a password first and compares the read config to the expected config and if it matches it will stop the read w/password command as the tag did not have the password bit set.
the warning the other reader gives is: "Password bit in config word is not set!  Reading with password is stopped.  The process could damage the transponder!"

so at the very least i suggest we put a large warning in the help for this command...

iceman1001, 2015-10-15:
I agree with @marshmellow42  here,   we have been chatting about this one sometime.

I did a change to the client,  when called with a pwd,
it will try load configblock,  decode it,   and find PWD bit.
if it is set,  it will allow reading with pwd.
if it is not set,  it will print a message and continue executing the command without the pwd.

---snippet

pm3 --> lf t55x7 read 0 00000000
Modulation : ASK
Bit Rate   : 5 - RF/64
Inverted   : No
Offset     : 31
Block0     : 0x00148040

PWD bit is NOT set in config block. Reading without password...
[0] 0x00148040  00000000000101001000000001000000

tbox1911, 2015-10-16:
Hi ,

thanks a lots for these informations,
It much clear for me now !

thanks also to iceman for the nice fix .

Best regards.

iceman1001, 2015-10-16:
Don't forget to be grateful to @marshmellow42  His solution is based on mine, but is better.

marshmellow42, 2015-10-16:
I'll have a push ready shortly to get some adjustments into the main
trunk.  After a little more testing.  :). Then we can probably close this
issue.

iceman1001, 2015-11-02:
Time to close this one.

closed 2015-11-02 by tbox1911

osysltd, 2016-05-11:
Just come across this bug, sorry for late feedback. Still is it feasible to have the ability set the tag unreadable by purpose (ex. lf t55xx kill)?

iceman1001, 2016-05-11:
Why do you want to kill a tag?   You can set the AOR, PWD bits in block0, and set a random pwd in block7. This way your tag will be very hard to read

osysltd, 2016-05-11:
For educational purposes :) Thanks for the tip.