Proxmark/proxmark3 #145: Flasher reference to old wikipage @ Google Code

dnet, 2015-11-09:
[Line 88 of client/flasher.c](https://github.com/Proxmark/proxmark3/blob/master/client/flasher.c#L88) refers to the `Linux` page back on the wiki at Google Code, but with GC being deprecated, this link should be updated. I shuffled through the wiki of this GitHub project, but couldn't find an appropriate page, there's no `Linux` page, and the [Gentoo Linux](https://github.com/Proxmark/proxmark3/wiki/Gentoo%20Linux) and [Ubuntu Linux](https://github.com/Proxmark/proxmark3/wiki/Ubuntu%20Linux) pages doesn't mention the modem-manager problem the error message refers to.

Is this modem-manager thing still an issue? If so, can it be removed? If yes, I'm happy to remove it in a PR. If no, what would be an appropriate replacement URL?

dnet, 2015-12-18:
Any thoughts on this one?

iceman1001, 2016-01-21:
Who knows if it still is a problem, the Linux ppl hasn't responded.
I've updated the flasher.c help text to point towards the new Wiki url's in my fork. 

The modem manager solution should be added to the GitHub Wiki.

closed 2016-08-29 by iceman1001