Proxmark/proxmark3 #150: Proxmark will not show up unless button is held down.

TopazA, 2015-12-10:
Hi there,
Using a proxmark 3.

If i hold down the switch while connecting the Proxmark to my OS-X machine (and continue holding it), I can see the device under /dev/cu.usbmodem1412/ and can run 

    ./proxmark /dev/cu.usbmodem1412

Without any errors being thrown.

While the button is being held down, the red and orange LED's are lit up and do not change.

I can also see the following output from 

    >> system_profiler SPUSBDataType

        Communication Device:

          Product ID: 0x504d
          Vendor ID: 0x2d2d
          Version: 0.01
          Speed: Up to 12 Mb/sec
          Manufacturer: proxmark.org
          Location ID: 0x14200000 / 8
          Current Available (mA): 500
          Current Required (mA): 0

I can't see any details if i run 'hw version' from within the proxmark app

If i let go of the button, the green LED briefly lights up and disappears. Every 18-20 seconds the relay also clicks audibly.  
In addition, i can't see any output from system_profiler and the device does not appear in /dev/cu* 

I've inherited this device so i'm not sure whats happened. I think the previous user may have attempted to flash the firmware/bootloader.

I successfully (i think) upgraded the image while running the following command (with the button held down)

    sh-3.2# ./flasher /dev/cu.usbmodem1421 ../armsrc/obj/fullimage.elf
    Loading ELF file '../armsrc/obj/fullimage.elf'...
    Loading usable ELF segments:
    0: V 0x00102000 P 0x00102000 (0x00025960->0x00025960) [R X] @0x94
    1: V 0x00200000 P 0x00127960 (0x00001a44->0x00001a44) [RW ] @0x259f4
    Note: Extending previous segment from 0x25960 to 0x273a4 bytes

    Waiting for Proxmark to appear on /dev/cu.usbmodem1421. Found.

    Flashing...
    Writing segments for file: ../armsrc/obj/fullimage.elf
     0x00102000..0x001293a3 [0x273a4 / 314      blocks]........................................................................................................................................................    ..................................................................................................................................................................     OK

    Resetting hardware...
    All done.
 
    Have a nice day!

If i do the same thing but attempt to flash the bootrom.elf file it doesn't matter if i have the button held or not, the proxmark will not 'Appear on USB'

The below output is produced:

    sh-3.2# ./flasher -b ../../bootrom/obj/bootrom.elf
    Loading ELF file '../../bootrom/obj/bootrom.elf'...
    Loading usable ELF segments:
    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
    1: V 0x00200000 P 0x00100200 (0x00000b68->0x00000b68) [RWX] @0x298

    Waiting for Proxmark to appear on USB...........................................................................^C

marshmellow42, 2015-12-10:
looks to me like you have a non matching bootrom to os image.  bootrom has to be flashed.
it also looks like you did not specify the usb address when you attempted to flash the bootrom, but you did when flashing the os/fullimage

lastly, i believe this is something best discussed on the main forum and not github as there is no clear bug.  at least not yet.

iceman1001, 2015-12-10:
use old hid-flasher to upgrade bootrom,  then use new flasher to flash fullimage.
disconnect and reconnect pm3.   If you do read on the forum, there is a lot of threads about this kinds of experiences as @marshmellow42  mentions.

TopazA, 2015-12-11:
Hey @marshmellow42 thanks for that, i can move to the forums if that's better.

@iceman1001 How do i use an old HID-flasher? Is there one included in the current software hosted here? Is there a specific version i should be looking for? 

Thanks!

marshmellow42, 2017-03-06:
the old HID-flasher is included in the current software.  but it is unlikely your firmware is that old.. ?  

i would like to assume you've already fixed this issue though as it is now over a year later....  time to close this issue?

closed 2017-03-06 by iceman1001

ronnyandre, 2020-03-10:
I just found this thread. Not sure where the old HID flashes is found. If anyone has a clue, please let me kno.