Proxmark/proxmark3 #151: Error when compiling on archlinux

t3zla, 2015-12-11:
I get this error when installing on arch

    In file included from ./string.h:15:0,
                     from ../common/protocols.c:2:
    ./stdint.h:16:18: error: conflicting types for 'int32_t'
     typedef int      int32_t;
                      ^
    In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                     from /usr/arm-none-eabi/include/strings.h:13,
                     from ../common/protocols.c:1:
    /usr/arm-none-eabi/include/sys/_stdint.h:31:19: note: previous declaration of 'int32_t' was here
     typedef __int32_t int32_t ;
                       ^
    In file included from ./string.h:15:0,
                     from ../common/protocols.c:2:
    ./stdint.h:21:24: error: conflicting types for 'uint32_t'
     typedef unsigned int   uint32_t;
                        ^
    In file included from /usr/arm-none-eabi/include/sys/types.h:62:0,
                     from /usr/arm-none-eabi/include/strings.h:13,
                     from ../common/protocols.c:1:
    /usr/arm-none-eabi/include/sys/_stdint.h:32:20: note: previous declaration of 'uint32_t' was here
     typedef __uint32_t uint32_t ;
                    ^
    ../common/Makefile.common:81: recipe for target 'obj/protocols.o' failed
    make[1]: *** [obj/protocols.o] Error 1
    make[1]: Leaving directory '/home/t3zla/Documents/Projects/AUR/proxmark/proxmark3/armsrc' 
    Makefile:10: recipe for target 'armsrc/all' failed
    make: *** [armsrc/all] Error 2

t3zla, 2015-12-17:
It works when I compile in the client directory

iceman1001, 2015-12-17:
thats because you only compiled the client....

The error states your armsrc (firmware) didn't get compiled.

Try comment the offensive #include in protocols.c

ikarus23, 2015-12-23:
As an archlinux user I've got this issue too. Removing the line `#include <strings.h>` from `common/protocols.c` did the trick.

Are there any downsides? Should we remove this include from the `master`?

iceman1001, 2015-12-23:
it compiles on mingw / win  enviornment without this include.

ikarus23, 2015-12-23:
Thanks for the reply, iceman. I will wait for some more days and remove the include if nobody complains.

iceman1001, 2015-12-23:
I also just tried it on Ubuntu 1404,  it compiles fine.

ikarus23, 2015-12-23:
Thanks for testing. I think it's fairly save to remove this include it it works for those two platforms. At least most of the proxmark3 people are using Windows or Ubuntu, I think.

commit 7ede3c71 (2015-12-23): Remove include statement that causes trouble (fix #151).

See: https://github.com/Proxmark/proxmark3/issues/151
closed 2015-12-23 by ikarus23