Proxmark/proxmark3 #155: RF Power ON between REQA. HF tag

pablito12, 2015-12-21:
Hi
I want to get the information between the RF power ON and the first RQA send by a reader. I am using a Mifare Classic  tag and a i wonder if Proxmark3 allow to get this information.

iceman1001, 2015-12-21:
I think questions like this is more suitable on the forum.
And read the wiki for answers to your question

pablito12, 2015-12-21:
well I allready read the wiki and that's the reason why I am asking it. I am not sure if Proxmark3 give this kind of information (time between RF Power On and REQA message sen by a reader)

iceman1001, 2015-12-21:
let me guess, you don't have pm3 device?

W8M2Hg9lLmWqXSGC, 2015-12-21:
Use the forum. It's there for a reason.
http://www.proxmark.org/forum/index.php

closed 2015-12-21 by W8M2Hg9lLmWqXSGC

pablito12, 2015-12-21:
Actually I do have a Proxmark3 device. Probably Iceman do not understand me