Proxmark/proxmark3 #159: issue compiling proxmark3 on kali ARM

thesle3p, 2016-02-08:
Hi so when I try and compile the proxmark3 client on a Arm version of the Kali OS I get the following error when it tries to compile  liblua: 
```
make
make -C client all
make[1]: Entering directory '/root/proxmark3/client'
Compiling liblua, using platform linux
cd ../liblua && make linux
make[2]: Entering directory '/root/proxmark3/liblua'
make all SYSCFLAGS="-DLUA_USE_LINUX" SYSLIBS="-Wl,-E -ldl -lreadline"
make[3]: Entering directory '/root/proxmark3/liblua'
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_LINUX    -c -o lapi.o lapi.c
cc1: internal compiler error: Segmentation fault
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
<builtin>: recipe for target 'lapi.o' failed
make[3]: *** [lapi.o] Error 1
make[3]: Leaving directory '/root/proxmark3/liblua'
Makefile:106: recipe for target 'linux' failed
make[2]: *** [linux] Error 2
make[2]: Leaving directory '/root/proxmark3/liblua'
Makefile:161: recipe for target 'lua_build' failed
make[1]: *** [lua_build] Error 2
make[1]: Leaving directory '/root/proxmark3/client'
Makefile:12: recipe for target 'client/all' failed
make: *** [client/all] Error 2
```

iceman1001, 2016-02-18:
http://www.proxmark.org/forum/viewtopic.php?pid=16495#p16495

"For compiling on Kali Linux (Nethunter) I had to add -ncurses as an argument for gcc and install the ncurses library, the readline library is insufficient for successful compilation. Similar to Mac OS X, I had to make changes to the QT library path locations to enable the source to compile on Kali Linux (Nethunter) as well."

thesle3p, 2016-02-18:
Can you perhaps post your changes to the Makefile and the names of the packages you had to install?

iceman1001, 2016-02-18:
You must missunderstand, I don't run on a Kali dev env.  I linked to a forum post where another user has solved his issues with compiling on Kali.  I thought you would at least read it and understand.

thesle3p, 2016-02-18:
Ok, rude remarks aside thank you.

iceman1001, 2016-02-18:
Indeed,  you are on your own.

thesle3p, 2016-08-02:
For anyone who wants to compile the proxmark3 client for kali arm (Nethunter, etc) run the following commands

```
git clone https://github.com/Proxmark/proxmark3.git
apt-get install libreadline6-dev
apt-get install gcc-arm-none-eabi
cd proxmark3
make
```

From there it is a standard linux client.

closed 2016-08-02 by thesle3p