Proxmark/proxmark3 #163: Bus Errors HID - DATA RAW FS - Elechouse pm3v2

Apt-Get1, 2016-02-15:
Elechouse PM3 v2

Prox/RFID mark3 RFID instrument          
bootrom: master/v2.2.0-142-g7ede3c7-dirty-suspect 2016-02-03 06:16:11
os: master/v2.2.0-153-gd2736c4-suspect 2016-02-15 02:11:43
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2015/11/ 2 at  9: 8: 8
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 168954 bytes (32). Free: 355334 bytes (68).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          

Checking for known tags:
          
HID Prox TAG ID: 2004901ec2 (3937) - Format Len: 26bit - FC: 72 - Card: 3937          

Valid HID Prox ID Found!          
proxmark3> data raw fs
/Users/macbook/Desktop/PM3.command: line 1: 10042 Bus error: 10           ./proxmark3/client/proxmark3 /dev/cu.usbmodem1451
logout

[Process completed]

marshmellow42, 2016-02-15:
does it give you the same error no matter what the second command is? 
what system os?
did you get any build warnings/errors?
are you using the master GitHub branch?

Apt-Get1, 2016-02-15:
OSX yosemite. Using Current Master.

gcc -std=c99 -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/nonce2key/crapto1.o nonce2key/crapto1.c
clang: warning: -O4 is equivalent to -O3
nonce2key/crapto1.c:102:1: warning: unused function 'binsearch' [-Wunused-function]
binsearch(uint32_t *start, uint32_t *stop)

gcc -std=c99 -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/cmdhfepa.o cmdhfepa.c
clang: warning: -O4 is equivalent to -O3
cmdhfepa.c:85:20: warning: duplicate 'const' declaration specifier [-Wduplicate-decl-specifier]
        static const char const *usage_msg =
cc -std=c99 -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/cmdhftopaz.o cmdhftopaz.c
clang: warning: -O4 is equivalent to -O3
cmdhftopaz.c:340:58: warning: implicit conversion from 'int' to 'uint8_t' (aka 'unsigned char') changes value from 256 to 0 [-Wconstant-conversion]
                        uint8_t size_in_bytes = TLV_value[1] ? TLV_value[1] : 256;
                                ~~~~~~~~~~~~~                                 ^~~

gcc -std=c99 -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/cmdlfviking.o cmdlfviking.c
clang: warning: -O4 is equivalent to -O3
cmdlfviking.c:69:18: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
        if (strlen(Cmd) < 0 || cmdp == 'h' || cmdp == 'H') return usage_lf_viking_clone();
            ~~~~~~~~~~~ ^ ~
cmdlfviking.c:92:18: warning: comparison of unsigned expression < 0 is always false [-Wtautological-compare]
        if (strlen(Cmd) < 0 || cmdp == 'h' || cmdp == 'H') return usage_lf_viking_sim();
            ~~~~~~~~~~~ ^ ~

gcc -std=c99 -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/fpga_compress.o fpga_compress.c
clang: warning: -O4 is equivalent to -O3
fpga_compress.c:90:112: warning: format specifies type 'unsigned long' but the argument has type 'int' [-Wformat]
                        fprintf(stderr, "Input files too big (total > %lu bytes). These are probably not PM3 FPGA config files.\n", num_infiles*FPGA_CONFIG_SIZE);
                                                                      ~~~                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                      %d
fpga_compress.c:141:70: warning: format specifies type 'unsigned long' but the argument has type 'uint32_t' (aka 'unsigned int') [-Wformat]
        fprintf(stderr, "compressed %lu input bytes to %lu output bytes\n", i, compressed_fpga_stream.total_out);
                                    ~~~                                     ^
                                    %u

iceman1001, 2016-02-15:
Most warnings is fixed in my coverity fixes,  which @marshmellow42  will soon merge.

That still doesn't explain "data raw fs" crasch.  Doubt it has something to do with the device per se.

Apt-Get1, 2016-02-15:
I did just update xcode.. so maybe its that or macports? Im currently reinstalling both.. the command is running fine in windows7 on the same machine in vmFusion.

Apt-Get1, 2016-02-15:
iceman do you have a current compiled osx version i can try?

iceman1001, 2016-02-15:
macports, or xcode sounds like the issue.
I don't have a mac dev environment up, sorry,  I have a Windows/Mingw  and Linux Ubuntu working.

if you want to compile my fork,  try the  "make all UBUNTU_1404_QT4=1"  option.  Usually something about libs also.

Apt-Get1, 2016-02-15:
reinstalled macports and xcode to currrent versions.. same issue.. 
i also get bus errors running lf t55 det  on tags that are cloned to HID.

iceman1001, 2016-02-15:
side notes,  you can change -04 in the makefiles,  to -03  and you'll get rid more warnings.
the double const is in cmdhfepa.c  nothing t55xx.
fpga_compress is still nothing t55xx,
binsearch is not used,   so none of the messages has something to do with bus errors..

marshmellow42, 2016-02-15:
bad port?  have you tried a different one?

Apt-Get1, 2016-02-15:
ports are good. Most commands are running fine. tried other port also. same issue.  the same usb port works fine in the windows vm. this is just a mac compiler issue with the new version of xcode or macports probably

Apt-Get1, 2016-02-15:
there are 2 less errors since your recent patch. Marshmellow.

marshmellow42, 2016-02-15:
I don't receive the warnings in windows.  I didn't see the issue with the 2 const(s) in iceman's commits (but I probably missed it since I was going through about 60 of them in an hour... :P )    The others are harmless warnings, but can ultimately be fixed...

marshmellow42, 2016-02-15:
plus a couple of the coverity fixes had to go into my other branch that isn't merged yet since i'll be editing the same file and I didn't want to run into merge conflicts...  (though I don't remember these specific ones...)  I do agree that these warnings have nothing to do with the bus issues

Apt-Get1, 2016-02-15:
this seems to be very recent with the updates in xcode.

marshmellow42, 2016-02-15:
you've had it working before recently then?

iceman1001, 2016-02-15:
If you compile on linux you'll get clang warnings,  but they disappear if you take my coverity fixes,.. like all 110 of them ;)

Apt-Get1, 2016-02-15:
yes it was either xcode 7.2.1 or macports upgrade that is different. i just did a complete reinstall of bothe and still the same issue.

Im going to try to downgrade both to an earlier release.

iceman1001, 2016-02-15:
I've did some upgrade withth QT4 -> QT5,,, never again,  what a mess to upgrade.  The Pm3 master doesn't like upgrading system stuff..  Same with the GCC4,8..  couldn't upgrade to GCC4.9 without errors. I gave up.

holiman, 2016-02-15:
This bug is about the bus errors, right - not the compiler warnings ? I'm having trouble following the conversation..

holiman, 2016-02-15:
Good question from @marshmellow42 :does it give you the same error no matter what the second command is?

iceman1001, 2016-02-15:
I think we have come the conclusion that it is not a compiling of the source bug..  More of upgrading the mac os related.  I'm curios if the problem exists if @Apt-Get1  reverts to older xcode & macports version

Apt-Get1, 2016-02-15:
@holiman ----------
Looks like all the data raw commands are throwing bus errors..
data save, fskhid, load, samples,buff,  all seem to work ok.

lf t55 det only throws a bus error when you run it on a cloned tag.
if you run it on a clean tag its fine.
----------------------------
@ Marshmellow.----------------------
installing 7.2.0 xcode now. that is what i was using before the upgrade to feb2 (7.2.1)
Ill keep you posted.
------------------------

holiman, 2016-02-15:
I don't believe it has to do with compiler, but who knows. 
Some googling: http://stackoverflow.com/questions/8716714/bus-error-10-error : 

> if it would have been an error due to memory allocation it would have been given segmentation fault at the run time .But this error is coming due to constant string modification

I wonder if we're trying to write into a constant string, somewhere. If you were on linux, I'd ask you to run gdb on it, don't know what to use on Mac. If you could obtain a stack trace, that'd be good.

holiman, 2016-02-15:
Or, if the flag `g_debugMode` was before the command was issued, I think `setdebugmode 2` ought to do it. So do that, then make it crash.

iceman1001, 2016-02-15:
we have the SwapEndian64,  which gives a coverity warning for linking out a buffer..

holiman, 2016-02-15:
Hm, I don't see it called from `data`...?

holiman, 2016-02-15:
That's just a resource leak, isn't it ?

Apt-Get1, 2016-02-15:
@Holiman 
Sorry im not sure where to set this g_debugMode. 

proxmark3> data set 2
proxmark3> lf se
Reading 30000 bytes from device memory
          
Data fetched          
Samples @ 8 bits/smpl, decimation 1:1           
NOTE: some demods output possible binary
  if it finds something that looks like a tag          
False Positives ARE possible
         
Checking for known tags:
          
DEBUG: Preamble not found          
DEBUG: Preamble not found          
HID Prox TAG ID: 3709cb6331 (45464) - Format Len: 36bit - FC: 0 - Card: 0          
DEBUG: idx: 84, Len: 96, Printing Demod Buffer:          
0001110101010101
0101101001101010
0101010110010110
1010010110011010
0110100101011010
0101101001010110
          

Valid HID Prox ID Found!          
proxmark3> data raw fs
Bus error: 10
MacBook-Pro:client macbook$

holiman, 2016-02-15:
What happens if you just start the client; `./proxmark foo` and then `data raw fs` ? I mean without all the communication with a tag, or even the pm3 ?

Apt-Get1, 2016-02-15:
Last login: Mon Feb 15 14:31:28 on ttys000
/Users/macbook/Desktop/PM3.command ; exit;
MacBook-Pro:~ macbook$ /Users/macbook/Desktop/PM3.command ; exit;
ERROR: invalid serial port
proxmark3> data load tagsaved.pm3
loaded 30000 samples          
proxmark3> data fskhi
HID Prox TAG ID: 3709cb6331 (45464) - Format Len: 36bit - FC: 0 - Card: 0          
proxmark3> data raw fs
/Users/Jamesmacbook/Desktop/PM3.command: line 1: 31044 Bus error: 10           ./proxmark3/client/proxmark3 /dev/cu.usbmodem1451
logout

[Process completed]

holiman, 2016-02-15:
Do you get the same on any of the traces in the `/traces/` folder ? I'm trying a few...

holiman, 2016-02-15:
For example, does this work for you: 

	proxmark3> data load /data/workspace/proxmark3-main/traces/HID-weak-fob-11647.pm3
	loaded 20000 samples          
	proxmark3> data fskhi
	HID Prox TAG ID: 211c1c5afe (11647) - Format Len: 32bit - FC: 0 - Card: 0          
	proxmark3> data raw fs

	Using Clock:16, invert:0, fchigh:10, fclow:8          
	FSK2 decoded bitstream:          
	1110000000110000
	1110001111110000
	1100001110000001
	1111100000001100
	0011111100000011
	1111000000011100
	0111000111000011
	0000111000111000

Apt-Get1, 2016-02-15:
Still OFFLINE..

proxmark3> data load HID-weak-fob-11647.pm3
loaded 20000 samples          
proxmark3> data fskhi
HID Prox TAG ID: 211c1c5afe (11647) - Format Len: 32bit - FC: 0 - Card: 0          
proxmark3> data raw fs
/Users/Jamesmacbook/Desktop/PM3.command: line 1: 54963 Bus error: 10           ./proxmark3/client/proxmark3 /dev/cu.usbmodem1451
logout

[Process completed]

holiman, 2016-02-15:
Thanks, one more thing, do it also with debug: 

	proxmark3> data load /data/workspace/proxmark3-main/traces/HID-weak-fob-11647.pm3
	loaded 20000 samples          
	proxmark3> data setdeb 2
	proxmark3> data raw fs
	DEBUG countfc: FC 1, Cnt 1, best fc: 1, best2 fc: 0          
	DEBUG countfc: FC 8, Cnt 640, best fc: 8, best2 fc: 1          
	DEBUG countfc: FC 10, Cnt 1196, best fc: 10, best2 fc: 8

Apt-Get1, 2016-02-15:
proxmark3> data load HID-weak-fob-11647.pm3
loaded 20000 samples          
proxmark3> data setdeb 2
proxmark3> data raw fs
/Users/macbook/Desktop/PM3.command: line 1: 56788 Bus error: 10           ./proxmark3/client/proxmark3 /dev/cu.usbmodem1451
logout

[Process completed]

also just installed the previous version of xcode 7.2 
compiled and flashed.
same bus error 10.

Apt-Get1, 2016-02-15:
Also went back to an older Jan18 version of the client and no bus errors.

marshmellow42, 2016-02-15:
jan18 this year?  not much has changed...

Apt-Get1, 2016-02-15:
Tracked down the commits to the last working version.
Downloaded and compiled commit 7ede3c7
No issues.
Something in the next commit 5490c2d6 is throwing the bus errors.

marshmellow42, 2016-02-15:
I bet it is the %hu in line 985 of the cmddata.c file...  change back to a %d and see if it works

Apt-Get1, 2016-02-15:
in the client folder? i dont see that at line 985

marshmellow42, 2016-02-15:
well it was that line number back then...  in the function: int FSKrawDemod

change: 		PrintAndLog("\nUsing Clock:%hu, invert:%hu, fchigh:%hu, fclow:%hu", rfLen, invert, fchigh, fclow);
and either remove the 'h' from each of the %hu 
or switch it to '%d'  (my current preference).   the 'u' alone sometimes does weird things on the clock output.

marshmellow42, 2016-02-15:
maybe line 998 now.

Apt-Get1, 2016-02-15:
Currently commit 5490c2d has the following
```
			PrintAndLog("\nUsing Clock:%u, invert:%u, fchigh:%u, fclow:%u", rfLen, invert, fchigh, fclow);
			PrintAndLog("%s decoded bitstream:",GetFSKType(fchigh,fclow,invert));
			printDemodBuff();

```
Changed them to %d
Bug Fixed!! 
Thank you Marshmellow Id buy you a beer if i could.

marshmellow42, 2016-02-15:
it appears %hu is broken for printf:
http://www.microchip.com/forums/m581592.aspx
i'll adjust it back to %d immediately.

marshmellow42, 2016-02-15:
it appeared to work fine on windows and Linux... :(  so I went with it...

Apt-Get1, 2016-02-15:
made the changes in the current master and still bus error..:(  there must be something else added?

marshmellow42, 2016-02-15:
I think you mean commit 6ca1477  ?  which was implemented at the same time into the master as 5490c2d

Apt-Get1, 2016-02-15:
no i downloaded jan 5 (5490c2d)
 (6c1477) is jan 7

marshmellow42, 2016-02-15:
try 			
PrintAndLog("\nUsing Clock:%u, invert:%u, fchigh:%u, fclow:%u", (unsigned int)rfLen, (unsigned int)invert, (unsigned int)fchigh, (unsigned int)fclow);

marshmellow42, 2016-02-15:
interesting even though that didn't hit the master until 2/9/2015 it retains my commit dates from my fork.  learn something new every day...

Apt-Get1, 2016-02-15:
still bus error in current master with your changes.

marshmellow42, 2016-02-15:
does : data detectclock fs work?

Apt-Get1, 2016-02-15:
Current master. No modifications to client files.

data detectclock fs = working

marshmellow42, 2016-02-16:
ultimately the issue appears somewhere in #158 - or so it seems.  any of the commits prior to and including 54c90c2d would be different code than the commit you reference as last working.  I believe it to have been 6ca1477.  but I cannot be sure as I can't reproduce the issue.

Apt-Get1, 2016-02-16:
When i follow the commits here
https://github.com/Proxmark/proxmark3/commits/master

7ede3c7 is the last working version for osx with the data raw commands.

marshmellow42, 2016-02-16:
correct, however if you downloaded commit 5490c2d, that isn't a commit directly to the master branch it is a commit in my fork that got pulled in via pull request #158 which had other commits directly behind it and probably didn't even have 7ede3c7 in it.  

if you notice at the link you provided 5490c2d isn't listed...

marshmellow42, 2016-02-16:
so we need to look at the entire #158 up to 5490c2d if you confirmed 7ede3c7 doesn't have the bug

Apt-Get1, 2016-02-16:
Yes 7ede3c7 is confirmed working in OSX

marshmellow42, 2016-02-16:
do you have issues with other commands besides data rawdemod fs, (and besides lf search u as it uses it and besides t55xx commands as they use it and besides lf fskhiddemod as it uses the same code)? 
data rawdemod am  ?
or lf read - data samples - data plot
or other commands?

Apt-Get1, 2016-02-16:
Using 7ede3c7
when I run "data raw am" on an HID trace it will dump but still bus error 10
when i run "data raw fs" it works fine.
lf read is working.
samples is working
save, load etc, all working.
fskpyr working.

[9 more replies]

closed 2016-02-16 by Apt-Get1