Proxmark/proxmark3 #169: Compile error on Windows 10 (x64) @ /tools/mkversion.pl

VinnieV, 2016-03-24:
Hi!

Note: This error doesn't occur on a fresh installed Windows 10.

When compiling the proxmark3 firmware the compiler gives an error at the perlscript mkversion.pl.
The number before [main] increments until I get a BSOD.

If you want more information feel free to contact me.
Meanwhile I am searching for a solution. (updates will be posted)

**Compile Error:**
```
make -C bootrom all
make[1]: Entering directory `/pm3/bootrom'
perl ../tools/mkversion.pl .. > version.c || cp ../common/default_version.c version.c
      2 [main] perl 17168 child_copy: linked dll data write copy failed, 0xC2000..0xC2370, done 0, windows pid 17168, Win32 error 87
5096694 [main] perl 17168 child_copy: linked dll data write copy failed, 0xC2000..0xC2370, done 0, windows pid 17168, Win32 error 87
10103219 [main] perl 17168 child_copy: linked dll data write copy failed, 0xC2000..0xC2370, done 0, windows pid 17168, Win32 error 87
```

[BSOD dump file](http://s000.tinyupload.com/?file_id=34383980811182412077 )
(Github didn't allow my zip for some reason..)

iceman1001, 2016-03-24:
The win32 error 87 could be a hint. Quick search gives "the parameter is wrong" as suggestion.

Do you have git installed?

VinnieV, 2016-03-24:
Yes, git is installed.

Error 487 is also given when compiling:
`1 [main] perl 16176 child_copy: linked dll data write copy failed, 0x652000..0x652370, done 0, windows pid 16176, Win32 error 487`

iceman1001, 2016-03-24:
git fetch --all
git describe --dirty
git rev-parse --abbrev-ref HEAD

Can you run these three git commands?  They are the basis in mkversion.pl

VinnieV, 2016-03-24:
Jep those commands work fine.

[image]

iceman1001, 2016-03-24:
leans towards something with perl and win10-64b...  
Which version of perl?

VinnieV, 2016-03-24:
v5.22.1 (newest version)

[image]

marshmellow42, 2016-03-24:
http://stackoverflow.com/questions/18502999/git-extensions-win32-error-487-couldnt-reserve-space-for-cygwins-heap-win32

iceman1001, 2016-03-25:
if its the 'msys-1.0.dll' then a upgrade of git and mingw is needed.

VinnieV, 2016-03-25:
Apparently adding /C/ProxSpace/msys/bin to the front of my path variable helped the perl error.
It compiles perfectly now. 
Thanks to Marshmellow and Iceman!

closed 2016-03-25 by VinnieV