Proxmark/proxmark3 #174: Windows 10 Compiling Problem

AlienDennis, 2016-04-22:
When in make all. I have this.
     
 3 [main] perl 8564 child_copy: linked dll data write copy failed, 0xD2000..0xD2370, done 0, windows pid 8564, Win32 error 87
5171205 [main] perl 8564 child_copy: linked dll data write copy failed, 0xD2000..0xD2370, done 0, windows pid 8564, Win32 error 87
10391689 [main] perl 8564 child_copy: linked dll data write copy failed, 0xD2000..0xD2370, done 0, windows pid 8564, Win32 error 87

When in make flash-all

2 [main] perl 8992 child_copy: linked dll data write copy failed, 0x192000..0x192370, done 0, windows pid 8992, Win32 error 87
5143995 [main] perl 8992 child_copy: linked dll data write copy failed, 0x192000..0x192370, done 0, windows pid 8992, Win32 error 487
10247430 [main] perl 8992 child_copy: linked dll data write copy failed, 0x192000..0x192370, done 0, windows pid 8992, Win32 error 87
15369415 [main] perl 8992 child_copy: linked dll data write copy failed, 0x192000..0x192370, done 0, windows pid 8992, Win32 error 487
20464707 [main] perl 8992 child_copy: linked dll data write copy failed, 0x192000..0x192370, done 0, windows pid 8992, Win32 error 87
25692924 [main] perl 8992 child_copy: linked dll data write copy failed, 0x192000..0x192370, done 0, windows pid 8992, Win32 error 487
30790023 [main] perl 8992 child_copy: linked dll data write copy failed, 0x192000..0x192370, done 0, windows pid 8992, Win32 error 487
arm-none-eabi-gcc -c -I../include -I../common -Wall -Werror -pedantic -std=c99 -Os -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG  -DWITH_CRC -DON_DEVICE -DWITH_HFSNOOP -fno-strict-aliasing -ffunction-sections -fdata-sections -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -I. -mthumb -mthumb-interwork -o obj/version.o version.c
arm-none-eabi-gcc -nostartfiles -nodefaultlibs -Wl,-gc-sections -n -Wl,-T,ldscript,-Map,obj/fullimage.stage1.map -o obj/fullimage.stage1.elf obj/version.o obj/fpga_all.o obj/start.o obj/iso15693.o obj/iso15693tools.o obj/lfops.o obj/hitag2.o obj/hitagS.o obj/lfsampling.o obj/pcf7931.o obj/lfdemod.o obj/protocols.o obj/inflate.o obj/inffast.o obj/inftrees.o obj/adler32.o obj/zutil.o obj/appmain.o obj/printf.o obj/util.o obj/string.o obj/usb_cdc.o obj/cmd.o obj/fpgaloader.o obj/legicrf.o obj/epa.o obj/iso14443a.o obj/mifareutil.o obj/mifarecmd.o obj/mifaresniff.o obj/iso14443b.o obj/crapto1.o obj/crypto1.o obj/des.o obj/aes.o obj/iso14443crc.o obj/crc.o obj/crc16.o obj/crc32.o obj/legic_prng.o obj/iclass.o obj/BigBuf.o obj/optimized_cipher.o obj/hfsnoop.o -lgcc
arm-none-eabi-objcopy -O binary -I elf32-littlearm --remove-section .data obj/fullimage.stage1.elf obj/fullimage.nodata.bin
arm-none-eabi-objcopy -O elf32-littlearm -I binary -B arm --rename-section .data=stage1_image obj/fullimage.nodata.bin obj/fullimage.nodata.o
arm-none-eabi-objcopy -O binary -I elf32-littlearm --only-section .data obj/fullimage.stage1.elf obj/fullimage.data.bin
../client/fpga_compress obj/fullimage.data.bin obj/fullimage.data.bin.z
compressed 11328 input bytes to 6557 output bytes
arm-none-eabi-objcopy -O elf32-littlearm -I binary -B arm --rename-section .data=compressed_data obj/fullimage.data.bin.z obj/fullimage.data.o
arm-none-eabi-gcc -nostartfiles -nodefaultlibs -Wl,-gc-sections -n -Wl,-T,ldscript,-Map,obj/fullimage.map -o obj/fullimage.elf obj/fullimage.nodata.o obj/fullimage.data.o
c:/projects/proxmark/devkitarm/bin/../lib/gcc/arm-none-eabi/4.7.1/../../../../arm-none-eabi/bin/ld.exe: warning: cannot find entry symbol Vector; defaulting to 00102000
make[1]: Leaving directory `/pm3/armsrc'
make -C client flasher
make[1]: Entering directory `/pm3/client'
make[1]: `flasher' is up to date.
make[1]: Leaving directory `/pm3/client'
client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000b88->0x00000b88) [RWX] @0x298

Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00029fe0->0x00029fe0) [R X] @0x94
1: V 0x00200000 P 0x0012bfe0 (0x000019a0->0x000019a0) [RW ] @0x2a074
Note: Extending previous segment from 0x29fe0 to 0x2b980 bytes

Waiting for Proxmark to appear on C:/Projects/Proxmark/msys/dev/ttyACM0.......................................

What went wrong ? I have been confused for days over the compilation.

iceman1001, 2016-04-23:
Two problems with the Makfiles. 
/common/Makefile   -  points to flash.ex as seen here.  https://github.com/Proxmark/proxmark3/blob/master/common/Makefile.common#L59   That doesnt exist.  Needs to be changed to flasher.exe
/Makefile  -  the flash port variable points to /dev/ttyACM0 which is a linux port.  Windows uses comX , where X is  a number starting with 3 or more.    https://github.com/Proxmark/proxmark3/blob/master/Makefile#L3

So lets test using the direct commands for flashing and see if you can flash.

AlienDennis, 2016-04-24:
Flash completed.

proxmark3> hw ver
[[[ Cached information ]]]

Prox/RFID mark3 RFID instrument
bootrom: /-suspect 2016-04-24 16:43:30
os: /-suspect 2016-04-24 16:45:54
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2015/11/ 2 at  9: 8: 8

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 186749 bytes (36%). Free: 337539 bytes (64%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

I guess i have done it right ?

iceman1001, 2016-04-24:
Looks all right to me.

AlienDennis, 2016-04-24:
Thanks iceman!

closed 2016-04-24 by AlienDennis