Proxmark/proxmark3 #177: Proxmark RDV2 sometimes unresponsive

ColdHeat, 2016-05-05:
I posted this on the Proxmark forums and iceman suggested I open a Github issue. 
http://www.proxmark.org/forum/viewtopic.php?pid=21336#p21336

Basically my Proxmark 3 RDV2 from Elechouse becomes unresponsive after every few commands. It will then either stop listening to commands or the client will stop sending them I am not sure. 

I have had the PM3 enter standalone mode where it will still appear to operate properly and read an Ultralight so I feel like the client is perhaps sending in some kind of different messaging rate or something. 

I have flashed the 2.5.0 firmware from: http://proxmark.org/forum/viewtopic.php?id=1562

Also I have noticed that the the GUI app available in the post above works much more consistently. Where the CLI would start to be unresponsive after perhaps 3 commands, the GUI app worked consistently in my testing.

I've been working from a Windows VM in VMWare Fusion., I'm not sure if that would cause an issue.

zhovner, 2016-05-05:
Sometimes virtual machines may cause error in USB forwarding. I see this issue with PN532 reader http://forums.nfc-tools.org/post/5277/#p5277 

Try to use native machine without VM for clear test.

ColdHeat, 2016-05-05:
Do you think there might be some sort of GUI code that could be ported to the CLI that might make it a little more reliable under a VM? 

I'll definitely try it on a host but it just seems a little odd to me that the GUI would work where the CLI doesn't...

iceman1001, 2016-05-05:
the GUI (by gaucho)  only sends / calls the client cli.  So the GUI can never be "more realiable" than the CLI...

ColdHeat, 2016-05-06:
Copying over my discussion from the proxmark forums:

> I got an example of me entering commands into the 2.5.0 client and only intermittedly getting responses.
> http://i.imgur.com/rx9Lc5k.png
> The setup is a 3" USB cable connected the PM3 and just the LF antennae plugged in.

iceman1001, 2016-05-06:
the pm3 works on win10.  You need to change the boot "to allow unsigned drivers to load"  then you are good to go.  

BUT,  big but here,  your virtual environment can't talk to the PM3 properly if the PM3 is not correctly installed on your host OS..   I guess that is the core of your problem here.

ColdHeat, 2016-05-06:
It's pretty unlikely that I'll install a driver like that on my personal computer outside of a virtual machine so I will have to get back to you in some other means to test the device. 

I also find it unlikely that the PM3 can't communicate properly due to the lack of a driver on the host OS. I can understand a timing issue due to the USB passthrough to the VM but the lack of a driver on the host? Seems unlikely as the device wouldn't be connected to the host OS. 

Not to mention I've successfully operated the card using the GUI application earlier today on an OSX laptop without drivers using a Windows VM with drivers. 

Is there documentation or history indicating that the PM3 cannot be used in a virtual machine? Even though I have successfully used the original PM3 and PM3 RDV2 (albeit with a GUI) in a VM?

iceman1001, 2016-05-06:
If you successfully used it with prio installtion on the host os,  then thats great. 
Not sure about your doubt about the driver.  
I've running my pm3 devices on virtual machines since two years back,  the usb to serial speed is real issue on guest os with windows on.  Drops down to 14kbs,   on ubuntu guest os I get decent speeds.  This is vmware images.  Havn't measured on Oracle's virtualbox yet. 

I did install my pm3 on my host os, to minimize identification problems when the device is hooked up on usb.   But then, I'm not the one with problems and I don't think I'll be able to help you further.

ColdHeat, 2016-05-07:
The likely issue is that the serial driver requires precise timing whereas the libusb driver likely had more checks in place as the USB protocol has error handling and I believe timing requirements. 

I was able to run the proxmark successfully on a Windows host OS as shown below:

[image]

This had nothing to do with the driver not being installed on the host OS or not using firmware from master.

closed 2016-05-07 by ColdHeat