Proxmark/proxmark3 #180: Error compiling on OSX 10.11.5 -

TefaTBob, 2016-05-24:
Hey all, n00b here so be gentle. :)

Been trying for a couple days to sort this out but can't seem to compile on OSX.
Pretty sure I've followed the instructions on the OSX page here as well as in the "compiling.txt", but no luck.

...
Compiling liblua, using platform macosx
cd ../liblua && make macosx
/Applications/Xcode.app/Contents/Developer/usr/bin/make all SYSCFLAGS="-DLUA_USE_MACOSX" SYSLIBS="-lreadline"
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_MACOSX    -c -o lapi.o lapi.c
error: unable to open output file 'lapi.o': 'Permission denied'
1 error generated.
make[3]: *** [lapi.o] Error 1
make[2]: *** [macosx] Error 2
make[1]: *** [lua_build] Error 2
make: *** [client/all] Error 2

zhovner, 2016-05-24:
Why you try compile lua from sources? Use [brew](http://brew.sh/)

TefaTBob, 2016-05-24:
Hey zhovner, I wasn't specifically trying to compile lua. That seems to be part of the process of compiling proxmark 3 when I cd in to the proxmark3 folder and run "make clean; make". The full text is below:

REllis:proxmark3 rellis$ sudo make clean; make
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C client clean
rm -f cli cli.exe flasher flasher.exe proxmark3 proxmark3.exe fpga_compress fpga_compress.exe snooper snooper.exe obj/nonce2key/crapto1.o obj/nonce2key/crypto1.o obj/nonce2key/nonce2key.o obj/loclass/cipher.o obj/loclass/cipherutils.o obj/loclass/des.o obj/loclass/ikeys.o obj/loclass/elite_crack.o obj/loclass/fileutils.o obj/mifarehost.o obj/crc.o obj/crc16.o obj/crc64.o obj/iso14443crc.o obj/iso15693tools.o obj/data.o obj/graph.o obj/ui.o obj/cmddata.o obj/lfdemod.o obj/cmdhf.o obj/cmdhf14a.o obj/cmdhf14b.o obj/cmdhf15.o obj/cmdhfepa.o obj/cmdhflegic.o obj/cmdhficlass.o obj/cmdhfmf.o obj/cmdhfmfu.o obj/cmdhftopaz.o obj/cmdhw.o obj/cmdlf.o obj/cmdlfio.o obj/cmdlfhid.o obj/cmdlfawid.o obj/cmdlfem4x.o obj/cmdlfhitag.o obj/cmdlfti.o obj/cmdparser.o obj/cmdmain.o obj/cmdlft55xx.o obj/cmdlfpcf7931.o obj/cmdlfviking.o obj/pm3_binlib.o obj/scripting.o obj/cmdscript.o obj/pm3_bitlib.o obj/aes.o obj/protocols.o obj/sha1.o obj/*.o *.o *.moc.cpp
cd ../liblua && make clean
rm -f liblua.a lua luac lapi.o lcode.o lctype.o ldebug.o ldo.o ldump.o lfunc.o lgc.o llex.o lmem.o lobject.o lopcodes.o lparser.o lstate.o lstring.o ltable.o ltm.o lundump.o lvm.o lzio.o lauxlib.o lbaselib.o lbitlib.o lcorolib.o ldblib.o liolib.o lmathlib.o loslib.o lstrlib.o ltablib.o loadlib.o linit.o  lua.o luac.o
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C bootrom clean
rm -rf obj/*.o
rm -rf obj/*.elf
rm -rf obj/*.s19
rm -rf obj/*.map
rm -rf obj/*.d
rm -rf version.c
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C armsrc clean
rm -rf obj/*.o
rm -rf obj/*.elf
rm -rf obj/*.s19
rm -rf obj/*.map
rm -rf obj/*.d
rm -rf obj/*.z
rm -rf obj/*.bin
rm -rf version.c
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C recovery clean
rm -f bootrom.bin fullimage.bin proxmark3_recovery.bin
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C client all
Compiling liblua, using platform macosx
cd ../liblua && make macosx
/Applications/Xcode.app/Contents/Developer/usr/bin/make all SYSCFLAGS="-DLUA_USE_MACOSX" SYSLIBS="-lreadline"
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_MACOSX    -c -o lapi.o lapi.c
error: unable to open output file 'lapi.o': 'Permission denied'
1 error generated.
make[3]: *** [lapi.o] Error 1
make[2]: *** [macosx] Error 2
make[1]: *** [lua_build] Error 2
make: *** [client/all] Error 2

TefaTBob, 2016-05-24:
Figured it out. Permissions issues in place by CorpIT :P

closed 2016-05-24 by TefaTBob