Proxmark/proxmark3 #182: ./update_svn.sh error

AlienDennis, 2016-06-16:
svn: E160013: Unable to connect to a repository at URL'http//proxmark3.googlecode.com/svn/trunk'
svn: E160013: 'http//proxmark3.googlecode.com/svn/trunk' path not found

Error occured while updating.

I patched 2 computers to the environment. This is my third one.

I keep having this error appearing. 

Need some help with this.

iceman1001, 2016-06-16:
It was years since we switch from googlecode/SVN to Github/GIT. 

The next question, is the shellscript "update_svn.sh" in the PM3 sourcecode?  if so, we need to remove it.

AlienDennis, 2016-06-16:
Yes. It is still in the sourcecode. Thanks iceman!

iceman1001, 2016-06-16:
Where do you find it?  I can't find it in the PM3 master?

AlienDennis, 2016-06-17:
Proxspace. 

https://code.google.com/archive/p/proxmark3/downloads

closed 2016-06-17 by AlienDennis

milkaty, 2017-12-16:
how do I make the change as below:\

****@iceman1001  

It was years since we switch from googlecode/SVN to Github/GIT.

The next question, is the shellscript "update_svn.sh" in the PM3 sourcecode? if so, we need to remove it.

iceman1001, 2017-12-16:
you dont.   You just download / clone the proxmark3 repo here on GitHub.

milkaty, 2017-12-20:
I really need you help to do the installation for the proxmark3 rdv2 i am new user experience 

If any posibality to give support by access my PC  

Whatsapp +971507094040
Facebook messenger : m.me/mostafalaban

joack81, 2020-07-06:
> Pasaron años desde que cambiamos de googlecode / SVN a Github / GIT.
> 
> La siguiente pregunta, ¿es el shellscript "update_svn.sh" en el código fuente de PM3? Si es así, necesitamos eliminarlo.

joack81, 2020-07-06:
buenas iceman!! yo soy nuevo en proxmark3 y no hay manera de ponerlo a andar, tengo el mismo problema arriba mencionado y no se que firmware utilizar, gracias..