Proxmark/proxmark3 #183: Compiling in OS X El capital 10.11.4 using brew

koalazak, 2016-07-19:
Hi guys, 
following the wiki instructions to build proxmark3 in Mac Os X, but using `brew` instead of `port` I have some problems that I found a solution:

1- `arm-none-eabi-gcc` formula is missing after update brew. Solution:
```bash
brew tap nitsky/stm32
brew install arm-none-eabi-gcc
```
2- Installing `readline` and `qt5` formulas `brew` alert:
```
OS X provides the BSD libedit library, which shadows libreadline.
In order to prevent conflicts when programs look for libreadline we are
defaulting this GNU Readline installation to keg-only.
Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

LDFLAGS:  -L/usr/local/opt/readline/lib
CPPFLAGS: -I/usr/local/opt/readline/include
```
So to avoid a dangerous `brew link --force readline` I add this paths to proxmark3 Makefiles:
```diff
# client/Makefile
-LDLIBS = -L/opt/local/lib -L/usr/local/lib -lreadline -lpthread -lm
+LDLIBS = -L/usr/local/lib -L/usr/local/opt/readline/lib -L/usr/local/opt/qt5/lib -lreadline -lpthread -lm

-CFLAGS = -std=c99 -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall $(COMMON_FLAGS) -g -O4
+CFLAGS = -std=c99 -I/usr/local/opt/qt5/include -I/usr/local/opt/readline/include -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall $(COMMON_FLAGS) -g -O4

# common/Makefile.common
-LDFLAGS = -nostartfiles -nodefaultlibs -Wl,-gc-sections -n
+LDFLAGS = -L/usr/local/opt/readline/lib -L/usr/local/opt/qt5/lib -nostartfiles -nodefaultlibs -Wl,-gc-sections -n
```

Then the compilation was complete without errors.

if you are in agreement, I can make a PR with the changes and update the wiki with brew instructions.

regards,

iceman1001, 2016-07-19:
Nice! you seem to have also added some references to qt5  but fail to mention it.

zhovner, 2016-07-19:
Can you please add this here https://github.com/Proxmark/proxmark3/blob/master/COMPILING.txt#L81

koalazak, 2016-07-20:
Yes, qt5 installation says the same alert. And I notice that my `LDFLAGS` modification in `common/Makefile.common` its not necessary.

iceman1001, 2016-07-20:
I belive you can edit the Wiki page by yourself here no?  
https://github.com/Proxmark/proxmark3/wiki/OSX

koalazak, 2016-07-20:
Yes, sure.

bigboyq, 2016-07-20:
I have tried modification on the client/Makefile, my OS X works correctly.
Deadline problem resolved

koalazak, 2016-07-21:
Wiki updated.
Why not change client/Makefile to add the -L params here mentioned? and make compilation clean for users instead of force users to edit its own client/Makefile?

Users compiling in other platforms (where those paths doesnt exist) only will see a warning messages.

what do you think?

iceman1001, 2016-07-21:
there is alread detection of OSX in the makefile,  you can add the extra -L params there,  only warnings will be for OSX which doesn't have brew installed etc. (or thats what I guess since I don't have a mac)

pwpiwi, 2017-03-10:
Can this be closed? Or does the issue still persist?

zhovner, 2017-03-10:
Our [homebrew tap](https://github.com/Proxmark/homebrew-proxmark3/blob/master/proxmark3.rb) formula still builds without GUI.

Universal fix needed as I see https://github.com/Proxmark/proxmark3/pull/191#issuecomment-244366539

pwpiwi, 2017-03-10:
#191 seems to be incomplete? Did you test it? Is it an improvement at least?

pwpiwi, 2017-03-12:
@zhovner or any other Mac users: please note that none of the active Proxmark maintainers is using OS X. If you want OS X issues fixed, your active feedback is required.

pwpiwi, 2017-03-31:
Makefile has been updated. You may test again.

However you have to fix your environment first. You cannot expect the proxmark application to correct incomplete Qt or pkg-config setups.

zhovner, 2017-03-31:
```
In file included from proxgui.cpp:12:
In file included from ./proxguiqt.h:11:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/QApplication:1:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/qapplication.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/qtwidgetsglobal.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtGui/qtguiglobal.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qglobal.h:97:
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qcompilerdetection.h:562:6: error: Qt requires a C++11 compiler and yours does not seem to be that.
#    error Qt requires a C++11 compiler and yours does not seem to be that.
     ^
In file included from proxgui.cpp:12:
In file included from ./proxguiqt.h:11:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/QApplication:1:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/qapplication.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/qtwidgetsglobal.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtGui/qtguiglobal.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qglobal.h:1129:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qatomic.h:46:
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:61:4: error: "Qt requires C++11 support"
#  error "Qt requires C++11 support"
   ^
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:93:13: error: unknown type name 'QAtomicOps'
    typedef QAtomicOps<T> Ops;
            ^
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:93:23: error: expected member name or ';' after declaration specifiers
    typedef QAtomicOps<T> Ops;
    ~~~~~~~~~~~~~~~~~~^
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:96:23: error: use of undeclared identifier 'QAtomicOpsSupport'
    Q_STATIC_ASSERT_X(QAtomicOpsSupport<sizeof(T)>::IsSupported, "template parameter is an integral of a size not supported on this platform");
                      ^
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:96:53: error: no member named 'IsSupported' in the global namespace
    Q_STATIC_ASSERT_X(QAtomicOpsSupport<sizeof(T)>::IsSupported, "template parameter is an integral of a size not supported on this platform");
                                                  ~~^
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qglobal.h:747:63: note: expanded from macro 'Q_STATIC_ASSERT_X'
#define Q_STATIC_ASSERT_X(Condition, Message) Q_STATIC_ASSERT(Condition)
                                                              ^~~~~~~~~
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qglobal.h:742:110: note: expanded from macro 'Q_STATIC_ASSERT'
    enum {Q_STATIC_ASSERT_PRIVATE_JOIN(q_static_assert_result, __COUNTER__) = sizeof(QStaticAssertFailure<!!(Condition)>)}
                                                                                                             ^~~~~~~~~
In file included from proxgui.cpp:12:
In file included from ./proxguiqt.h:11:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/QApplication:1:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/qapplication.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/qtwidgetsglobal.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtGui/qtguiglobal.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qglobal.h:1129:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qatomic.h:46:
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:96:5: error: type name requires a specifier or qualifier
    Q_STATIC_ASSERT_X(QAtomicOpsSupport<sizeof(T)>::IsSupported, "template parameter is an integral of a size not supported on this platform");
    ^
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qglobal.h:747:47: note: expanded from macro 'Q_STATIC_ASSERT_X'
#define Q_STATIC_ASSERT_X(Condition, Message) Q_STATIC_ASSERT(Condition)
                                              ^
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qglobal.h:742:121: note: expanded from macro 'Q_STATIC_ASSERT'
    enum {Q_STATIC_ASSERT_PRIVATE_JOIN(q_static_assert_result, __COUNTER__) = sizeof(QStaticAssertFailure<!!(Condition)>)}
                                                                                                                        ^
In file included from proxgui.cpp:12:
In file included from ./proxguiqt.h:11:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/QApplication:1:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/qapplication.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtWidgets/qtwidgetsglobal.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtGui/qtguiglobal.h:43:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qglobal.h:1129:
In file included from /usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qatomic.h:46:
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:98:14: error: use of undeclared identifier 'Ops'
    typename Ops::Type _q_value;
             ^
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:98:19: error: expected a qualified name after 'typename'
    typename Ops::Type _q_value;
[... 14 lines cut ...]
[... cut ...]
/usr/local/Cellar/qt5/5.8.0_1/include/QtCore/qbasicatomic.h:110:86: error: use of undeclared identifier 'Ops'
    static Q_DECL_CONSTEXPR bool isReferenceCountingNative() Q_DECL_NOTHROW { return Ops::isReferenceCountingNative(); }
                                                                                     ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[1]: *** [obj/proxgui.o] Error 1
make: *** [client/all] Error 2
/usr/local/Homebrew/Library/Homebrew/debrew.rb:11:in `raise'
BuildError: Failed executing: make all
```

pwpiwi, 2017-03-31:
Ok. Tries to build with gui finally. I assume the root cause for the new issue is at the very beginning:
> Qt requires a C++11 compiler and yours does not seem to be that

Proxmark sources and compiler options enforce the C99 standard. I recommend to downgrade Qt.

pwpiwi, 2017-04-01:
... Or you can try to add `-std=C++11` to the CXXFLAGS in client/Makefile. Let us know the results please.

closed 2017-04-11 by pwpiwi