Proxmark/proxmark3 #189: Compiling on OS X with GUI

yrjyrj123, 2016-08-29:
I tried to compiling proxmark3 client on OS X El capital 10.11.6 with the help of wiki(https://github.com/Proxmark/proxmark3/wiki/OSX)

But it says "No GUI in this build!" when I use some command such as "hw tune".

I use HomeBrew to install components essential.

```
brew tap nitsky/stm32
brew install readline libusb p7zip libusb-compat wget qt5 pkgconfig arm-none-eabi-gcc
```

I read the client/Makefile and find the problem. 'QT' has been replaced by 'QT5' and QApplication has been moved to QtWidgets, so I change the client/Makefile:

```
else ifeq ($(platform),Darwin)
    CXXFLAGS = $(shell pkg-config --cflags Qt5Core Qt5Gui Qt5Widgets 2>/dev/null) -Wall -O4
    QTLDLIBS = $(shell pkg-config --libs Qt5Core Qt5Gui Qt5Widgets 2>/dev/null)
    MOC = $(shell pkg-config --variable=moc_location Qt5Core 2>/dev/null)
    LUAPLATFORM = macosx
```

After this, I find "`pkg-config --variable=moc_location Qt5Core`" can't find moc_location variable,
I have to add one line "`moc_location=${prefix}/bin/moc`" in "`/usr/local/Cellar/qt5/5.6.1-1/lib/pkgconfig/QT5Core.pc`"

now, proxmark3 client can use GUI.

Could you fix the client/Makefile, so that everyone can use GUI easily.

Thx.

iceman1001, 2016-08-29:
Great,
Someone on the forum or here, wasn't too impressed with homebrew and started using something else instead. If I only could remember what it was.

How about you make a PullRequest with your changes, and we take it from there?

bigboyq, 2016-08-29:
For brew users, readline modification is required also
I am not sure about qt5, but if want to use qt, make file modification required also

bigboyq, 2016-08-29:
I have made a PR, #190 
This is my first PR, every suggestion or comments welcomed.

yrjyrj123, 2016-08-31:
I make a PR for QT5 compiling problem[#191](https://github.com/Proxmark/proxmark3/pull/191), every suggestion or comments welcomed.

pwpiwi, 2017-03-31:
Did you have the chance to test with the new proxmark brew tap?

closed 2017-04-11 by pwpiwi