Proxmark/proxmark3 #200: hf mf dump needs robust enhancement

bigboyq, 2016-10-13:
While using proxmark3 to dump cards, found the dump process require some lucky
Especially on some bad quality card.
During the dump process, once the command timeout, the whole process is interrupted, need to repeat again, and depends on the lucky again.
I think these enhancement could be done:
1. timeout configure (default 9ms)
2. retires configure (currently 0, I think 3 is suggested)
3. continue work? (if failed, continue from the last stop)

as the dumpdata.bin is written by the end of hf mf dump, probably method 1 and 2 could be easily implemented.

To implement method 3, more work might be done.

Anyone agree, or agree PR could be submitted?

:P

iceman1001, 2016-10-13:
1. Don't know about the timeouts,  they are there to because the read-block command somehow failed. ie the dump command failed.
2. Retry dump?  If you fail you run the command again or are you thinking using this in a automatic function?  
3. if the dump continues from a failed state, there is no garanties that the produced dump file is correct.  

I suggest writing your own dump-command in lua,  where you can add better reading block, and retry that block read until you are satisfied.  I did that for tnp-dump.lua script.

bigboyq, 2016-10-13:
> Don't know about the timeouts, they are there to because the read-block command somehow failed. ie the dump command failed.

Select card timeout, interrupt the dump process.

> Retry dump? If you fail you run the command again or are you thinking using this in a automatic function?

Hmm, I think what I write confused you, my original meaning is if select card timeout, retry the select function, if failed more than 3 times, interrupt the dump process

>if the dump continues from a failed state, there is no garanties that the produced dump file is correct.

Agree, small enhancement requires lots of work, may be not worthful.

iceman1001, 2016-10-13:
Do you get these "select card time outs" with different cards or is it certain types?

bigboyq, 2016-10-13:
Hmm, I must admit that the card I am using is bad quality, but these message might occurs more frequently with the card age.
What's more, 
if the fail rate is 1% per block read, 64 blocks dump might 52.55% success
if the fail rate is 2% per block read, 64 blocks dump might 27.44% success
if the fail rate is 0.1% per block read, 64 blocks dump might 93.7% success

for my experience, one of my card used 4 years, dump success around 40%, 2 years magic card only hits 20% success.

May be enlarge the timeout value might reduce the fail rate, not sure yet.

Under this situation, I think this enhancement is worthful.

iceman1001, 2016-10-13:
Have you tried with a stronger HF antenna if you still get those timeouts?

bigboyq, 2016-10-13:
Dont't have strong HF antenna currently, can not do the experiment

iceman1001, 2016-10-13:
there are some suggestions on the forum on how to build a HF antenna.  Can't hurt testing with a stronger antenna.  I don't have these time outs so I can't verify it.

iceman1001, 2016-12-20:
How did it go with a stronger antenna?

bigboyq, 2017-01-18:
Sorry for missing days, quite busy recently , no yet finished.
I have no plan to buy a stronger antenna.
I will try a code enhancement in 2017.

iceman1001, 2017-01-18:
No need to buy a stronger antenna,  its quite easy to build one yourself.  Check the antenna section of the forum for guides  (or the wiki)

bigboyq, 2017-01-18:
[image]
You could see my board looks like that:(
No ability to make modification

iceman1001, 2017-01-18:
Is that the Radiowars dual antenna?   It should connect to the pm3 board somehow?

bigboyq, 2017-01-18:
```

F:\pm3-bin-2.5.0\win32 (client+GUI)>proxmark3 com7
Prox/RFID mark3 RFID instrument
bootrom: /-suspect 2015-11-19 10:08:02
os: /-suspect 2015-11-19 10:08:09
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2015/11/ 2 at  9: 8: 8

uC: AT91SAM7S256 Rev C
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 169916 bytes (65%). Free: 92228 bytes (35%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> hw tune

Measuring antenna characteristics, please wait...#db# DownloadFPGA(len: 42096)
....#db# DownloadFPGA(len: 42096)
.
# LF antenna: 28.05 V @   125.00 kHz
# LF antenna: 33.69 V @   134.00 kHz
# LF optimal: 38.09 V @   130.43 kHz
# HF antenna: 15.31 V @    13.56 MHz
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

proxmark3>
```

bigboyq, 2017-01-18:
with LF/MF mixed Card

```
proxmark3> hw tune

Measuring antenna characteristics, please wait...#db# DownloadFPGA(len: 42096)
....#db# DownloadFPGA(len: 42096)
.
# LF antenna: 25.30 V @   125.00 kHz
# LF antenna: 29.98 V @   134.00 kHz
# LF optimal: 31.90 V @   130.43 kHz
# HF antenna: 10.56 V @    13.56 MHz
Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

```

iceman1001, 2017-01-18:
LF part looks great.
HF part is good,  15v is good.
Your issues you have might not be antenna related  (hardware wise)

With 15v HF,  the positioning / placment of tag over the antenna becomes more vital.
1cm-1.5cm distance.    I have this issues with my mifare tags,  but once the right distance the problems are gone.  I use a post-it block (or two) between antenna & tag.

iceman1001, 2017-01-18:
And I see you use @buggii 's pre-compiled distro, v2.5.0,  which is quite old. (one year or more)
I suggest compile & flash of the latest source from Github.

bigboyq, 2017-01-18:
> LF part looks great.
> HF part is good, 15v is good.
> Your issues you have might not be antenna related (hardware wise)

How to detect which voltage is the best choice, any instruction to determine the antenna quality?

> And I see you use @buggii 's pre-compiled distro, v2.5.0, which is quite old

I could compile it under mac os, with the latest firmware, and latest program, but I normally using proxmark3 in windows environment,(No experience with windows compile) if I am using the latest build, without satisfied win32 program, not sure what might happens:P

The best choice is offically push the build to the next step, depends you, iceman. haha

iceman1001, 2017-01-18:
There is many posts on the forum about antennas.  I leave it up to you to find and read about it.

You have to use the same firmware (flashed on device) as the client you are running from the same build.
otherwise you be asking for trouble. 

Use your mac instead for windows.   There is a homebrew tap,  there is a docker container,   you find something to use.  As long as it allows you to pull the latest source from github and compile,  you are fine.

marshmellow42, 2017-01-26:
this issue occasionally annoys me as well.  i might have a look at it when i find time.

iceman1001, 2017-01-26:
@marshmellow42  tired of LF again?  ;)
Both this and the other issue with "hf mf restore" annoys me. If I would compare, the restore issue annoys me the most.  With a magic gen1 tag, not so much since its easy to "wipe". But with these gen2 tags...

bigboyq, 2017-02-04:
Hmm, it sounds like everyone is annoyed by these two issues, but compare with coding work, normally we choose to adapt it.
:P

PR #220 (2017-03-01): small patches and small added features listed below:
Add a retry loop in `hf mf dump` to retry each read up to 3 times (helps fix #200 )
Fixed a small bug in STT detection for strong (or cleaned) signal.
Fixed a bug preventing the detection of EM410x XL format AKA Unique XL or 128 bit Unique ID
add `data setgraphmarkers` <orange> <blue> to allow setting markers on the graph
add `data mtrim` <start> <stop> to allow middle trimming (or trimming out samples between start and stop)
make the askmandemod mark the first STT on the graph if it is found
closed 2017-03-01 by pwpiwi