Proxmark/proxmark3 #201: hf mf restore enhancement

bigboyq, 2016-10-13:
Currently, the hf mf restore can only "Restore MIFARE classic binary file to BLANK tag"
This is useful for clone cards!
During the real life, two enhancement required:
1. restore check, I found that there is no check process after restore, if any data is missing, or any operation problem occurs, the cloned card is different from the original one, the fastest way I could found is "hf mf dump" for different cards, and diff from bash.
2. restore by dumpkey.bin. Currently , BLANK tag means format before restore, format needs key A/B reset first, no script or function could do, manually required. If restore could be done between same Key A/B cards, it could be convenient for users who clone card for backups. 

Further discuss welcomed

iceman1001, 2016-10-13:
if its a magic tag generation1, (backdoor commands)  there is a -wipe tag option for  **hf mf cset**"  it wipes the whole tag and fills it with zeros.

There is also a **remagic.lua** script I made long ago that resets a magic tag to clean state.

bigboyq, 2016-10-13:
In fact, I face the problem is sync data between two cards:) To ensure that I am not modified money data in the card, some time from magic card to nxp card

iceman1001, 2016-10-13:
**hf mf restore**  
uses a keyfile. So if you written a dump onto a tag,  you need that dumpkeys file next time you want to restore to that tag.

bigboyq, 2016-10-13:
> hf mf restore
> 
> uses a keyfile. So if you written a dump onto a tag, you need that dumpkeys file next time you want to restore to that tag.

Yup, That is exactly what I need, but in default , hf mf restore using FFFFFFFFFFFF as the restore keys, ignore what inside dumpkeys.bin, I think if any args could control using or not might be enhancements

iceman1001, 2016-10-13:
It shouldn't ignore the dumpkeys.bin file. if it fails reading it, you see a message about it.

bigboyq, 2016-10-13:
Hmm, question comes: if as you described, How I can control it reads or not? By delete the key file or keep it? No arg control?

iceman1001, 2016-10-13:
I see the bug,  the  array with read keys "fkeys"  is not used when writing.  it only uses the default 0xFF,.,0xFF key variable.

iceman1001, 2016-10-13:
it uses the keys file for writing the sector trailers. check.
and default key for writing to blank tag check.
Agreed, 
the command should have an option to "use keys file" and a option to give a key.

bigboyq, 2016-10-13:
Hmm, happy you noticed, waiting for PR now

iceman1001, 2016-10-13:
This one I'm not fixing.  I merge your PR when you push it.

bigboyq, 2016-10-13:
You are pushing me, let me think

pwpiwi, 2017-03-19:
Restoring to a card which already has keys and access conditions set is impossible in many cases:

- Keys  changed since last dump
- Increment/decrement on value blocks happened since last dump
- Access conditions changed to deny writing since last dump

Probably this is the reason that it isn't implemented and will not be implemented?

samturner3, 2017-03-28:
My idea of how it should work:

1. When you do a hf mf restore, it should use the dumpkeys file to ACCESS the card to be written to (I think it does?)

2. The card is restored EXACTLY as dumpdata (including keys in dump data).

Currently when I have dumped a cracked card using nested, and I have the entire dumpdata file (including keys ie. exact copy of original cloned card) and I use hd mf restore, it restores the data as on the original, however it does not put on the original keys. It inserts the default F keys...

[image]

This photo is comparing two dumps, the original on the right, and the clone on the left after using hf mf restore. As you can see hf mf restore has just inserted default F keys not the original keys as in the dumpfile.

marshmellow42, 2017-04-01:
@samturner3. That is how it works.  Looks like you may be not setting the key file correctly.

I agree with piwi.  The enhancement suggested here would not really help most situations and that is likely why no one has built it.

pwpiwi, 2017-04-02:
However, @samturner3 has a point when he assumes the keys to be stored in the dumpdata.bin file. There is no reason to have two separate files for a complete dump/restore.

pwpiwi, 2017-04-05:
>  I got the answer it was like this because it mimics the behavior of a Mifare Classic card.

I can't find the discussion, but as far as I remember it was more about being compatibel to mfok files?

iceman1001, 2017-04-05:
The discussion went towards the compatibility issue and how a new dump-format would look like.
Now also to take in concern is MCT app if we ever want to build a comp-dumpformat :)

I still suggest a json-based format,  where we can add other information for **mfu** dumps etc.

digitalentropy, 2017-05-03:
I agree that the mifare dump format issue can be quite a confusing one for new users.  Already the fact that a different format is used for working with Chinese cards and standard cards is difficult to grasp for most people.  The conversion scripts help, but they are a hacky workaround to a standards issue.

There is no good solution.  Trying to change the format to a standard is not an insignificant amount of work and I can appreciate why such an action item would have a lower priority than squashing other bugs.

bigboyq, 2017-06-13:
I have time now, I am backing
Just flashed my board to 3.0.1, quite good work for iceman and marshmellow.
For dumpkey.bin, there should be two solutions:
1. keep the current hf mf restore command, not changes made, but make a new command, such as hf mf resetkey, using dumpkey.bin to reset card key to FFFF... , then hf mf restore might work properly.

2. change the hf mf restore command, check whether dumpkey.bin exists, if so , using key in dumpkey.bin file, otherwise, FFFF...

Please vote which one I should try?
@iceman1001 @pwpiwi @marshmellow42

iceman1001, 2017-06-13:
For  1) you could make as a lua-script. 
For 2) you will need to fiddle with c code.   

I say, Go for both!    Start with the simple 1...

marshmellow42, 2017-06-13:
The logic for 2 would need to be more complex or user optioned. (I always have a dumpkey.bin, doesn't mean I want to restore to that card.)

bigboyq, 2017-06-13:
Hmm, I have other opinions... as for hf mf dump command, in default, it use dumpkey.bin as the key for dump, if dumpkey.bin exists, and card key is different, it might failed.
If the logic for 2 need to be changed, the hf mf dump might also changed
As I am a lazy guy, I guess some code(dumpkey process) in the hf mf dump, might be reused in hf mf restore process....
Not sure yet, need to look for it first....

iceman1001, 2017-06-13:
as @marshmellow42 notices,  the dumpkeys.bin file is almost always present in the client folder but that doesn't mean I want to use it when I'm running 'hf mf restore'..   Could be a nasty surprice to find the keys changed from the dumpdata.bin.

I'd still do a simple lua script first.

bigboyq, 2017-07-01:
File located
proxmark3/client/cmdhfmf.c
Quite thanks to @marshmellow42 , hf mf dump, already enhanced for commit 40c6a02

I think the better solution is read dumpkey.bin, before dump, check key first, try FFF or dumpkey.bin content, then judge which one for use....

qaisjp, 2018-04-03:
What I do currently to achieve this is:

- use the mifare app on my phone to factory reset the card
- proceed to use `hf mf restore` to restore a dump on my now factory reset card

@bigboyq Have you yet added functionality to factory reset a card using a `dumpkeys.bin`?

deed02392, 2018-12-15:
I fell foul of this today. I had assumed the `dumpkeys.bin` was only used where keys weren't explicitly provided. In my mind having the keys in the `dumpdata.bin` is me providing the keys to write, therefore those should be used, not the ones in `dumpkeys.bin`.

DidierA, 2019-04-17:
Hi, I also had trouble trying to use hf restore to clone  a card. From what I understand reading your comments and the source code of` int CmdHF14AMfRestore(const char *Cmd)`, restore always uses FFFFFFFFFFFF and key type 'A' to write blocks (assuming keyType=0 means A). 
During all my testing I  was not able to write to chinese card with hf restore, because I used hf mf cwipe, and this leaves all blocks  with permission to write only to key type 'B'.
If this assumptions are correct, the solution in my case could be a lua script such as formatMifare.lua to rewrite the trailer of each sector and allow key B to write, then launch the restore.

DidierA, 2019-04-18:
In the end, what is the intended behavior of hf mf restore? 
Should it be used to write to a regular mifare bank card, hence using the defautl FFFFFFFFFFFF key?
Should it overwrite the keys and access control data with the ones in dumpdata.bin?
Or should it be used only to restore data on a card from a previous dump, thus reading keys from dumpdata.bin and not attempting to write on the trailer blocks?

pwpiwi, 2019-04-18:
It is intended to restore a dump to a blank card. The data (including the access conditions) is taken from dumpdata.bin, the keys which should be written to the trailers is taken from dumpkeys.bin. The blank card is supposed to have the transport configuration, i.e. writable with KeyA = 0xffffffffffff.

DidierA, 2019-04-18:
And it is exactly how it currently works. Thank you for the clarification.

Ollibolli85, 2019-05-08:
Hello everybody,

is it possible now to restore the keys in "dumpdata.bin" to a second tag?

pwpiwi, 2019-05-08:
The keys in dumpkeys.bin are written to a second tag.

Ollibolli85, 2019-05-08:
Yes, i already know that!

But i want to restore the keys in "dumpdata.bin"!

Is this possible with my Proxmark3?

pwpiwi, 2019-05-08:
No. And I am not sure why this should be necessary, You need the dumpkeys.bin to run `hf mf dump` which creates dumpdata.bin. So both files are there.

But it would not be difficult to implement.

Ollibolli85, 2019-05-08:
Without that, my tag will not be recognized by the Mifare-reader of the machine!

pwpiwi, 2019-05-08:
If the keys are coming from dumpkeys.bin or dumpdata,bin would not make a difference. Did you check if your clone contains the same data? Dump both the original and the clone and compare.

Ollibolli85, 2019-05-09:
Okay!
But there seems to be differences, because the keys to write/read the clone(s) are still the same default-keys from the beginning and not the keys from the dumped tag!
In the Windows-tool "mfocGUI", there is an option to "write keys" and after that (and just after that), my cloned tags works, but i want to do this with my Proxmark3!:-)

pwpiwi, 2019-05-09:
Do you have all keys in dumpkeys.bin? Do you see any errors during 'hf mf dump' ?

This is no longer related to the code. Please continue the discussion in the forum.

Ollibolli85, 2019-05-09:
No, there are no errors at all!
What so you mean with all keys?
One dumpkeys.bin-file for Key A, and one for Key B, from/for the new tag i want to resore with the dumpfile.bin-file from the first main-tag!
I think, while restore the new tag, the Proxmark3 replace the real (and important) keys from dumpfile.bin with the default-keys from dumpkeys.bin (which i actual also need to restore this new Tag)!

pwpiwi, 2019-05-09:
1. You need **one** dumpkeys.bin with **all** keys.
2. 'hf mf dump' uses the keys from this file to access the original card and to create dumpdata.bin.
3. 'hf mf restore' writes the keys from dumpkeys.bin and the data from dumpdata.bin to a blank card.

No more answers here. Continue in the forum.

Ollibolli85, 2019-05-10:
Yes, this is what i mean!

Now i just have to find out, how to creat dumpkeys.bin for all keys after hardnested attack...

Many thanks for your great help!

See you in the forum!

pwpiwi, 2019-05-14:
I have added an option to create dumpkeys.bin from emulator memory. See PR #822.