Proxmark/proxmark3 #203: `lf hid fskdemod` continues to TX a CW

samyk, 2016-12-19:
Running d1057e7a2105d5e2696df185482176f22a5bd669

When ending `lf hid fskdemod` by pressing the button and confirm the light turns off and getting`#db# Stopped` in pm3 client, an oscilloscope reveals the antenna is still transmitting a carrier wave.

iceman1001, 2016-12-19:
You are correct and thanks!

There is no call to **FpgaWriteConfWord(FPGA_MAJOR_MODE_OFF);**
inside lfops.c  methods CmdHIDdemodFSK  nor CmdAWIDdemodFSK ..  I've added it to my fork. I'm guessing @marshmellow42  will add them and push a PR to Pm3 Master soon.

iceman1001, 2016-12-19:
There was also the same problem with 'lf io fskdemod' 
also fixed and pushed.

marshmellow42, 2016-12-19:
this is addressed for the master in #202 pull request.

iceman1001, 2016-12-19:
fixed, merged and now closed.

closed 2016-12-19 by iceman1001

samyk, 2016-12-19:
Sweet!