Proxmark/proxmark3 #204: `hf mfu gen` has fixed inputs and is actually for mfc

micolous, 2017-01-02:
Looking at `CmdHF14AMfuGenDiverseKeys`, this appears to be a key diversification function for MIFARE Classic, not MIFARE Ultralight.

Also, it appears to have fixed inputs, so it is not very useful in the present state.

iceman1001, 2017-01-02:
You are right, its not useful at the moment.  That keydiv could also be used for the UL-C keys hence mfu because thats were I was trying to use it.  I'm thinking it could be used in Desfire but never gotten to it.
The interest is low because no-one seems to use NXP's diversification in their system so the cmd never got adapted for allround purposes. 

Feel free to contribute with enhancements.

iceman1001, 2017-02-28:
Since no response in two months. I'm closing it.

closed 2017-02-28 by iceman1001