Proxmark/proxmark3 #205: Homebrew Tap for OS X users

chrisfu, 2017-01-10:
Hi guys,

I've made a Homebrew tap that will automatically compile and install Proxmark3, the flasher and the hid-flasher so that you don't need to do any manual steps to get up and running. The repo and instructions for this are below:

https://github.com/chrisfu/homebrew-tap

At the moment the latest official release (v2.3.0) does not compile due to a bug with that release (the same issue existed for several Linux distros too I believe), but you should be fine to compile and install from HEAD with no troubles.

Also included are the bootrom and fullimage ELF's for your firmware flashing needs.

Feel free to either use this or include it within the mainline repository. Either way, with this the old Macports and Homebrew instructions (whilst useful for folk interested in delving deeper) could well be considered secondary to this more user-friendly method of installation.

Cheers,

Chris

zhovner, 2017-01-10:
Cool! Thank you.

iceman1001, 2017-01-10:
Lovin' it,   I cloned yr repo last autumn.  Haven't been able to test it, but I hope the mac users gets it to work.  I've updated yr changes and applied them for my pm3 fork.   
One of these days a Mac user could test it out and see if it works.

zhovner, 2017-01-11:
Tested successfully on MacOS Sierra 10.12.2 with `--HEAD` flag.

How we can add this formula to main repository without creating new repo with `homebrew-` prefix?  For example in `github.com/Proxmark/proxmark3/tools/homebrew/formula.rb`

iceman1001, 2017-01-21:
Don't know.  With the homebrew copy I did for iceman fork, it fails compiling "hid-flasher".
The libusb dependency doesn't seem to work.  When I commented out those lines in the .rb file it worked.   Howver without the old HID-flasher.   

I think we need verification that the libusb gets properly installed?  or change of makefile for old HID-flasher project. 
Sidenote, that project doesnt compile on Mingw either.  It does compile on Ubuntu14.04.

iceman1001, 2017-01-24:
Some feedback, I've asked the administrators of Proxmark github account if we should add another repo.  So far of those who has responded seems ok with it.   I let you know how it develops.

zhovner, 2017-03-06:
Any updates here?
 We need to create new repo named `homebrew-proxmark3`.

pwpiwi, 2017-03-06:
@iceman1001: what did the admins tell you? Is anyone of them willing to add the new repo? I would vote for it.

iceman1001, 2017-03-06:
The screaming silence from the ancient gods, roars thru the wast space of internet. 
No feedback from @ikarus23 @holiman @iZsh @midnitesnake  @proxmarkorg  @W8M2Hg9lLmWqXSGC 
or @pwpiwi  the ones in the proxmark github team.   However this is kind of the way the communication usually goes.  I ask, and maybe after x amount of time something may happend.   The ancient gods are very busy.

pwpiwi, 2017-03-06:
😄 
I am not 100% sure, but afaik none of these except the owner @W8M2Hg9lLmWqXSGC could add a new repo.

iceman1001, 2017-03-06:
I'll ask @W8M2Hg9lLmWqXSGC  again...

W8M2Hg9lLmWqXSGC, 2017-03-06:
Seems I missed the first message @iceman1001. Repository created.

zhovner, 2017-03-06:
@W8M2Hg9lLmWqXSGC can you please add README.md file in repo so we can make pull requests into.

W8M2Hg9lLmWqXSGC, 2017-03-06:
No problem @zhovner. Wasn't sure if we were doing a push or init.

iceman1001, 2017-03-06:
Perfect!  I'll close this issue.  For further discussions regarding homebrew-proxmark,  do it in the new repo-issues.

closed 2017-03-06 by iceman1001