Proxmark/proxmark3 #206: libnfc says it's a mifare 1k and pm3 says it's a MF0ICU1; only ISO14443A sure

joker42, 2017-01-10:
Hello,

It has been a few days that I am trying to find out what is this keyfob. In order to do it I would like to dump it and analyse the data. But trying known techniques to discover the keys doesn't work.

Here is the output of few relevant proxmark3 commands:

```
proxmark3> hf search

 UID : 65 93 7f d1
ATQA : 00 04
 SAK : 00 [2]
Tag is not Ultralight | NTAG | MY-D  [ATQA: 00 04 SAK: 00]

TYPE : MIFARE Ultralight (MF0ICU1) <magic>
proprietary non iso14443-4 card found, RATS not supported
Answers to chinese magic backdoor commands: NO

Valid ISO14443A Tag Found - Quiting Search

```

```
proxmark3> hf mfu info
Tag is not Ultralight | NTAG | MY-D  [ATQA: 00 04 SAK: 00]

```

hf mf chk * ? //doesn't find any keys
hf mk chk * ? default_keys.dic //doesn't find any keys

hf mf mifare //never stop / never find anything (with debug on: #db# Mifare: Can't select card)
hf mf nested //not useful without at least 1 key.
script run mifare_autopwn //never stop (it use darkside attack so hf mf mifare, that's why)

Also this is the output of the latest compiled nfc-tools (libnfc):
```
./utils/nfc-list -v
NFC device: SCM Micro / SCL3711-NFC&RW opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
* UID size: single
* bit frame anticollision supported
       UID (NFCID1): 65  93  7f  d1
      SAK (SEL_RES): 00
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
* SmartMX with MIFARE 1K emulation
```

Others:

```
proxmark3> hw version

Prox/RFID mark3 RFID instrument
bootrom: master/v2.2.0-272-gbbecd69-suspect 2017-01-10 19:13:16
os: master/v2.2.0-272-gbbecd69-suspect 2017-01-10 19:13:17
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2015/11/ 2 at  9: 8: 8

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 186204 bytes (36). Free: 338084 bytes (64).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

```

```
proxmark3> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   BigBuf_hi  .............39296
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............143
#db# Fgpa
#db#   mode.............HF
#db# LF Sampling config:
#db#   [q] divisor:           95
#db#   [b] bps:               8
#db#   [d] decimation:        1
#db#   [a] averaging:         1
#db#   [t] trigger threshold: 0
#db# USB Speed:
#db#   Sending USB packets to client...
#db#   Time elapsed:      1500ms
#db#   Bytes transferred: 894464
#db#   USB Transfer Speed PM3 -> Client = 596309 Bytes/s
#db# Various
#db#   MF_DBGLEVEL......2
#db#   ToSendMax........39
#db#   ToSendBit........8
proxmark3>
```

```
proxmark3> hf 14a reader
 UID : 65 93 7f d1
ATQA : 00 04
 SAK : 00 [2]
Tag is not Ultralight | NTAG | MY-D  [ATQA: 00 04 SAK: 00]

TYPE : MIFARE Ultralight (MF0ICU1) <magic>
proprietary non iso14443-4 card found, RATS not supported
Answers to chinese magic backdoor commands: NO
proxmark3>
```

Weird, it says it's an ultralight here:
```
proxmark3> script run mfkeys.lua
--- Executing: ./scripts/mfkeys.lua, args''
This script implements check keys. It utilises a large list of default keys (currently 92 keys).
If you want to add more, just put them inside mf_default_keys.lua.
Found a NXP MIFARE Ultralight | Ultralight C tag
I don't know how many sectors there are on this type of card, defaulting to 16
Testing block 3, keytype 0, with 85 keys
Testing block 3, keytype 0, with 7 keys
Testing block 3, keytype 1, with 85 keys
... // doesn't find anything
```

This one never ends. Also with debug it's flooded with "#db# Mifare: Can't select card", make sens because it doesn't seems to be a mifare...:
```
proxmark3> script run mifare_autopwn.lua
--- Executing: ./scripts/mifare_autopwn.lua, args''
Card found, commencing crack	65937FD1
```

iceman1001, 2017-01-11:
it could be a missconfigured s50/s70 magic card.   The pm3 detection fails because its sending a SAK which indicates its UL,  but is not.   Try the "hf mf cgetsc 0" command to read sector 0.

You also need some space between antenna & tag. 1.5cm about.

joker42, 2017-01-11:
It's not a magic card. It's a keyfob that you can attach at your keychain where it's written on it "65937FD1-A" and "AZTEK". I can do a photo of it if you want to dig into it.

```
proxmark3> hf mf cgetsc 0
--sector number:0
#db# wupC1 error
Can't read block. 0 error=2
```

iceman1001, 2017-01-11:
```
hf 14a reader
hf list 14a
```

joker42, 2017-01-11:
Thank you for helping @iceman1001 

```
proxmark3> hf 14a reader
 UID : 65 93 7f d1
ATQA : 00 04
 SAK : 00 [2]
Tag is not Ultralight | NTAG | MY-D  [ATQA: 00 04 SAK: 00]

TYPE : MIFARE Ultralight (MF0ICU1) <magic>
proprietary non iso14443-4 card found, RATS not supported
Answers to chinese magic backdoor commands: NO
proxmark3> hf list 14a
Recorded Activity (TraceLen = 143 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2244 |       4612 | Tag | 04  00                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL
      10692 |      16580 | Tag | 65  93  7f  d1  58                                              |     |
      18816 |      29344 | Rdr | 93  70  65  93  7f  d1  58  48  c7                              |  ok | SELECT_UID
      30532 |      34116 | Tag | 00  fe  51                                                      |     |
     504448 |     509216 | Rdr | e0  80  31  73                                                  |  ok | RATS
     510404 |     511620 | Tag | 00                                                              |     |
     975104 |     976096 | Rdr | 40                                                              |     | MAGIC WUPC1
     979524 |     980740 | Tag | 00                                                              |     |
     982144 |     983456 | Rdr | 43                                                              |     | MAGIC WUPC2
    1119360 |    1124128 | Rdr | 50  00  57  cd                                                  |  ok | HALT
proxmark3>
```

iceman1001, 2017-01-11:
the thing is as mentioned before,  your card has a strange sak/atqa response when selected.
thats why pm3 can't identify it correct.   There is a bug in the output of "hf 14a reader", trying to print "type" when it failed earlier.    I have pushed a fix in my fork for it.

joker42, 2017-01-11:
What if I mod the code and force the SAK to be 08, is there a chance for it to work?
(it will think the tag is a mifare classic 1k and won't be blocked because it doesn't identify it)

iceman1001, 2017-01-11:
its worth a try at least.

joker42, 2017-01-11:
Ok. Can you guide me what part in the code should I mod?

Because currently there is more than 150 references to 0x08 in the code and I don't know where to start...

iceman1001, 2017-01-11:
looks like a good place to fixate the SAK.

joker42, 2017-01-11:
Well it didn't work as expected...

Btw I get the Dbprintf("Mifare: Couldn't receive tag nonce"); message sometimes when playing with the distance between of the hf antenna and the tag during a hf mf mifare...

I am trying to go this path a little bit more before coming back to modding/forcing the SAK.

iceman1001, 2017-01-11:
after you get that message, would you mind running "hf list 14a"  and post the tracelog here?

joker42, 2017-01-11:
I can't. I can never have the prompt back without restart/reset of the board

iceman1001, 2017-01-11:
there seems to be something wrong with yr tag,  with the faulty SAK response. Do you have access to another card from same system?

joker42, 2017-01-11:
No I don't. But I know for sure the keyfob work perfectly (for the use case it is made for).

joker42, 2017-01-12:
If someone would like to take over "as a challenge", I could send him the keyfob by mail. I am based in France btw.

For the moment my only left option is to try to bring the pm3 on site in order to capture/snoop between the reader and the keyfob.

iceman1001, 2017-01-13:
If you don't mind, I can have a look at it. Email me.

iceman1001, 2017-01-23:
That is one strange tag.  Answers to ISo14443a,  strange sak/atq but doens't answer to mifare commands (0x60, 0xa0 etc)   I suspect it answers to iso7816..  
tag answers with 0x00  to things like chinese backdoor,   and to some other bytes combo sent after card select.

iceman1001, 2017-01-23:
hypotesis,  since its UID is printed on tag,  and it only answers to iso14443a basic commands,  it could be the same dumb LF tags which sends its ID over and over,  but for HF?

joker42, 2017-01-23:
@iceman1001 thank you for investigating.

Well the tag must have data to store an amount (can be reloaded by coins) within it because it is also used to obtain goods from a machine (like the coffee machine) which is disconnected from any network.

The reader on the machine doesn't react from a hf 14a sim with the uid/atqa/sak of the card. I tried a few days ago. I had to be in standalone mode so I don't know if the reader sent something.

buggii, 2017-01-23:
Need to sniff a transaction between tag and reader.

buggii, 2017-01-23:
It can be a smart card.

iceman1001, 2017-01-23:
Either sniff ,   or sim but no in standalone mode.  So you can get the tracelog from the transaction.

joker42, 2017-01-25:
I will try before Friday.

Just to be prepared and because I will have a very short window of time to do it:

- do we agree that "hf mf sniff" will probably return an error as the pm3 doesn't see the tag as a mifare? So it's not worth trying this idea?
- "hf 14a snoop" is worth trying, but I never played with that command yet and I didn't found example of use case that deal with that command.

- Does "hf list 14a" is saved in memory of the pm3 or do I have to dump it right after the snoop command before shutting down the pm3?
- As it's possible to sim (equivalent to hf 14a sim) in standalone, wouldn't it possible to check "hf list 14a" later (linked to the previous question I agree).

iceman1001, 2017-01-26:
try the "Hf 14a snoop",  
Havent tested the standalone mode,  to see if you can connect it later to get the tracelog out from  the device. 
You can also try the "hf 14a sim x" to see if the reader tries to authenticate..

joker42, 2017-01-26:
Hi, below are the results, I am still analyzing it (honestly it doesn't make much sens for me yet).

To put things in context, I had to do it with a new keyfob (exactly the same, just another UID), so this is following a "hf 14a reader":
```
proxmark3> hf list 14a
Recorded Activity (TraceLen = 143 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2228 |       4596 | Tag | 04  00                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL
      10676 |      16564 | Tag | 65  6b  ca  d1  15                                              |     |
      18816 |      29344 | Rdr | 93  70  65  6b  ca  d1  15  a5  7f                              |  ok | SELECT_UID
      30516 |      34100 | Tag | 00  fe  51                                                      |     |
     522624 |     527392 | Rdr | e0  80  31  73                                                  |  ok | RATS
     528564 |     529780 | Tag | 00                                                              |     |
     993152 |     994144 | Rdr | 40                                                              |     | MAGIC WUPC1
     997556 |     998772 | Tag | 00                                                              |     |
    1000192 |    1001504 | Rdr | 43                                                              |     | MAGIC WUPC2
    1137408 |    1142176 | Rdr | 50  00  57  cd                                                  |  ok | HALT
```

Here is a simulation:
```
proxmark3> hf 14a sim 4 656bcad1
Emulating ISO/IEC 14443 type A tag with 4 byte UID (656bcad1)
#db# Received unknown command (len=5):
#db# 10 0b 00 29 c4
#db# Received unknown command (len=5):
#db# 10 0b 00 29 c4
#db# Received unknown command (len=5):
#db# 10 0b 00 29 c4
...
#db# Received unknown command (len=5):
#db# 10 0b 00 29 c4
#db# Button press
#db# 0 0 92
```

Here is after a snoop (just putting the hf antenna between the reader and the keyfob while in snoop mode):
```
Recorded Activity (TraceLen = 1938 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |       1056 | Rdr | 26                                                              |     | REQA
     217104 |     218160 | Rdr | 26                                                              |     | REQA
     434256 |     435312 | Rdr | 26                                                              |     | REQA
     651840 |     652896 | Rdr | 26                                                              |     | REQA
     868384 |     869440 | Rdr | 26                                                              |     | REQA
    1085392 |    1086448 | Rdr | 26                                                              |     | REQA
    1302528 |    1303584 | Rdr | 26                                                              |     | REQA
    1519632 |    1520688 | Rdr | 26                                                              |     | REQA
    1736784 |    1737840 | Rdr | 26                                                              |     | REQA
    1953744 |    1954800 | Rdr | 26                                                              |     | REQA
    2170880 |    2171936 | Rdr | 26                                                              |     | REQA
    2388000 |    2389056 | Rdr | 26                                                              |     | REQA
    2605008 |    2606064 | Rdr | 26                                                              |     | REQA
    2822720 |    2823776 | Rdr | 26                                                              |     | REQA
    3039136 |    3040192 | Rdr | 26                                                              |     | REQA
    3256352 |    3257408 | Rdr | 26                                                              |     | REQA
[... 330 lines cut ...]
   44727364 |   44739012 | Tag | df  4d  14  25  60  24  01  16  a2  79                          |  ok |
   44815168 |   44821088 | Rdr | 10  15  00  a8  cb                                              |  ok | ?
   44822276 |   44833860 | Tag | 98  a4  87  85  f0  ef  c6  2c  df  96                          |  ok |
   44910336 |   44916192 | Rdr | 10  16  00  c0  e1                                              |  ok | ?
   44917444 |   44929092 | Tag | da  49  ee  b8  4a  73  a8  e4  18  da                          |  ok |
   45005248 |   45011168 | Rdr | 10  17  00  18  f8                                              |  ok | ?
[... cut ...]
   45979344 |   45994480 | Rdr | b0  1b  00  da  49  ee  b8  4a  73  a8  e4  7f  19              |  ok | TRANSFER(27)
   46032628 |   46044276 | Tag | da  49  ee  b8  4a  73  a8  e4  18  da                          |  ok |
   46128496 |   46143632 | Rdr | b0  1a  00  a2  eb  30  1b  ce  d4  73  f1  3e  e9              |  ok | TRANSFER(26)
   46181780 |   46193364 | Tag | a2  eb  30  1b  ce  d4  73  f1  7e  06                          |  ok |
proxmark3>
```

Thank you for your help @buggii , @iceman1001 and others that are interested to help me figure out this keyfob.

iceman1001, 2017-01-26:
This is great!
I can get the tag you sent me to answer!  

i.e    hf 14a raw -s -c 10 0b 00

And we have lots of commands, unknown, but with data.

iceman1001, 2017-01-26:
You should start a thread on the forum for this.

buggii, 2017-01-26:
I have a theory:

Machine asks for UID from the tag -> UID receive

Machine starts to read tag:

10 XX 00 + 2bytes-CRC (where XX is the memory address and 2bytes-CRC is probably ISO14443A CRC, need to be verified); if this is true the machine is reading 11 blocks, and receive back 8 bytes+2bytes-CRC at each read request.

Memory blocks "scanned" by thge machine seems to be:

0B

0C

10

15

1A

1F

10

11

12

13

14

Iceman if you can, try to send for example:
10 0A 00 + 2bytes-CRC
or
10 09 00 + 2bytes-CRC
and see if the tag is answering 8 different bytes+2bytes CRC.

A: Proxmark/proxmark3
Cc: buggii; Mention

iceman1001, 2017-01-26:
Yes,  the crc is the ISO14443a version. Sending 14a raw works.
I can send blockno  from 0x00 - 0x20 right now, Im writing a lua script to iterate them all
Block No 00 is funny, since I can see my tag UID there.

buggii, 2017-01-26:
Ok, the write command is:

B0 XX 00 + 8bytes data + 2bytes-ISO14443A-CRC where XX is the address where to write the 8 bytes data

Tag answer the exact 8bytes data + 2bytes-ISO14443A-CRC

Just for general info:
10 is the "write" command used for Topaz while B0 is the "transfer" command used for Mifare but the syntax is different here so i think we have a brand new nfc chip :)

iceman1001, 2017-01-26:
Yes!
0x10 - read
0xB0 - write

--read--
hf 14a raw -s -c -p 10 03 00

---write--
hf 14a raw -s -c -p B0 03 00 11 22 33 44 55 66 77 88

joker42, 2017-01-27:
That's amazing!
A lot of information to fully understand and to play with this week-end :)

Thank you guys!

iceman1001, 2017-01-30:
I've added a dumpscript https://github.com/iceman1001/proxmark3/commit/1cec48cc0e608397288ebbdc68768b02b8dcacab

iceman1001, 2017-02-04:
Rounding up: 
Tag is identified as Aztek.  follows iso14443a anticollision and have its own command set.

Closing this one.

closed 2017-02-04 by iceman1001