Proxmark/proxmark3 #210: Proxmark.org hijacked?

pwpiwi, 2017-02-02:
Forwards to aviationmegastore.com.

iceman1001, 2017-02-02:
its been noted.

Xoib, 2017-02-03:
If the administrator needs help, I can offer some sysadmin skills.

iceman1001, 2017-02-03:
Thank you for the offer,  but at this moment it is more of getting hold of the Webhosting company.

iceman1001, 2017-02-03:
Seems like the site is actually up,  but the aviationmegastore redirect is caused from a local installed virus(?!)
Running on another browser,  I can reach the forum.
Its on my firefox I have this problem. 
Restarting my computer, I can get access also for ftp.

dylangerdaly, 2017-02-03:
I can get to the site from a Linux box:
```
[user@personal-web ~]$ dig @8.8.8.8 A www.proxmark.org

; <<>> DiG 9.10.4-P5-RedHat-9.10.4-3.P5.fc24 <<>> @8.8.8.8 A www.proxmark.org
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 43297
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;www.proxmark.org.		IN	A

;; ANSWER SECTION:
www.proxmark.org.	14399	IN	A	37.44.12.83

;; Query time: 895 msec
;; SERVER: 8.8.8.8#53(8.8.8.8)
;; WHEN: Sat Feb 04 00:17:54 AEDT 2017
;; MSG SIZE  rcvd: 61

[user@personal-web ~]$ curl http://www.proxmark.org/forum/viewforum.php?id=33 | grep \<title\>
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100 19252    0 19252    0     0   7360      0 --:--:--  0:00:02 --:--:--  7362
<title>iClass / Proxmark developers community</title>
```

iceman1001, 2017-02-03:
yes, its up.   But if you were hit by that local redirect,  you need to clean it out from web browser.

dylangerdaly, 2017-02-03:
Yikes :/ maybe put out a tweet?

zhovner, 2017-02-03:
I was get redirected too. So it's not a local virus.

pwpiwi, 2017-02-03:
Redirected on Windows, Linux and Android. And only proxmark.org is affected. Either it is the Mega-Virus of the centry or no local virus at all. I would bet on the latter.

iceman1001, 2017-02-03:
Must have been local,  since I use multiple computers from same IP,  and those who surfed site when mega-hit,  always got redirected. 
But those computers who didn't, can surf direct to site. 
Cleaning out webcache, sorted it out.

My guess is javascript-based crap and a bug in fluxBB ..  Im guessing the user "signature" bug.

closed 2017-02-07 by iceman1001