Proxmark/proxmark3 #217: Undefined identifier `PATH_MAX` in `client/whereami.c`

starrify, 2017-02-24:
When trying to compile the latest version (`4197a3f`), it stopped due to an undefined identifier. Part of the error messages:

```
whereami.c: In function ‘wai_getExecutablePath’:
whereami.c:179:15: error: ‘PATH_MAX’ undeclared (first use in this function)
   char buffer[PATH_MAX];
```

The issue is believed to have been introduced in `4197a3f`. When compiling the code at a previous commit `4c16ae8`, everything looks good to me.

pwpiwi, 2017-02-24:
What is your environment? OS and compiler version?

starrify, 2017-02-24:
Oh.. Sorry for not having included the info. I thought that was a issue of the code (I see no `PATH_MAX` defined anywhere).

I'm using Arch Linux (x86_64) with kernel 4.9.8 and 6.3.1

iceman1001, 2017-02-24:
Ubuntu 16.04,   http://pastebin.com/V4xxLLX8

iceman1001, 2017-02-24:
Mingw32, http://pastebin.com/aAcuLBMj

pwpiwi, 2017-02-24:
Please replace line 166 by

#include <linux/limits.h>

and try again.

starrify, 2017-02-24:
> Please replace line 166 by
> #include <linux/limits.h>
> and try again.
Resolves the issue to me. 👍

iceman1001, 2017-02-24:
It solves the problem on Ubuntu,
problem remains on Mingw.

pwpiwi, 2017-02-24:
This becomes interesting. It works for me on mingw and gcc 6.2
Anyone else having issues compiling on mingw?

pwpiwi, 2017-02-24:
In fact I can't see why this should be included. @iceman1001: if you comment out the #include <intrin.h> , does it work?

iceman1001, 2017-02-24:
Tried that, but then the linker complains about **GetModuleHandleEx** 

```
g++ -I/qt/include -I/qt/include/QtCore -I/qt/include/QtGui -I/qt/include/QtWidgets  -I/mingw/include obj/proxmark3.o obj/uart.o ob
j/util.o obj/sleep.o obj/mifarehost.o obj/nonce2key/crapto1.o obj/nonce2key/crypto1.o obj/nonce2key/nonce2key.o obj/nonce2key/cryp
to1_bs.o obj/loclass/cipher.o obj/loclass/cipherutils.o obj/loclass/des.o obj/loclass/ikeys.o obj/loclass/elite_crack.o obj/loclas
s/fileutils.o obj/whereami.o obj/parity.o obj/crc.o obj/crc16.o obj/crc64.o obj/iso14443crc.o obj/legic_prng.o obj/iso15693tools.o
 obj/data.o obj/graph.o obj/ui.o obj/cmddata.o obj/lfdemod.o obj/cmdanalyse.o obj/cmdhf.o obj/cmdhf14a.o obj/cmdhf14b.o obj/cmdhf1
5.o obj/cmdhfepa.o obj/cmdhflegic.o obj/cmdhficlass.o obj/cmdhfmf.o obj/cmdhfmfu.o obj/cmdhfmfhard.o obj/cmdhfmfdes.o obj/cmdhftop
az.o obj/cmdhfemv.o obj/cmdhw.o obj/cmdlf.o obj/cmdlfio.o obj/cmdlfhid.o obj/cmdlfawid.o obj/cmdlfem4x.o obj/cmdlfhitag.o obj/cmdl
fti.o obj/cmdparser.o obj/cmdmain.o obj/cmdlft55xx.o obj/cmdlfpcf7931.o obj/cmdlfviking.o obj/cmdlfpresco.o obj/cmdlfpyramid.o obj
/cmdlfguard.o obj/cmdlfnedap.o obj/cmdlfjablotron.o obj/cmdlfvisa2000.o obj/cmdlfnoralsy.o obj/cmdlffdx.o obj/cmdlfcotag.o obj/pm3
_binlib.o obj/scripting.o obj/cmdscript.o obj/pm3_bitlib.o obj/aes.o obj/protocols.o obj/sha1.o obj/sha256.o obj/cmdcrc.o obj/reve
ng/preset.o obj/reveng/reveng.o obj/reveng/cli.o obj/reveng/bmpbit.o obj/reveng/model.o obj/reveng/poly.o obj/reveng/getopt.o obj/
tea.o obj/prng.o obj/bucketsort.o obj/proxgui.o obj/proxguiqt.o obj/proxguiqt.moc.o -L/opt/local/lib -L/usr/local/lib -lreadline -
lpthread -lm -L/mingw/lib -lgdi32 ../liblua/liblua.a -L/qt/lib  -lQt5Core -lQt5Gui -lQt5Widgets  -o proxmark3
obj/whereami.o: In function `wai_getModulePath':
C:\Projects\ProxMark3\ProxSpace-130613\ProxSpace\iceman\client/whereami.c:150: undefined reference to `GetModuleHandleEx'
collect2.exe: error: ld returned 1 exit status
make[1]: *** [proxmark3] Error 1
make[1]: Leaving directory `/pm3/client'
```

iceman1001, 2017-02-24:
I have Mingw32,  but I'm guessing you have Mingw64

pwpiwi, 2017-02-24:
Correct guess. I have mingw64 compiling 32Bit executables. Weird.

pwpiwi, 2017-02-24:
@iceman1001: you seem to compile and link your own branch with this commit applied. To rule out any side effects can you please try with a clean fresh master clone instead?

iceman1001, 2017-02-24:
That has nothing todo with it. Its a mingw32 issue more likely.

However since you insists,
PM3 master without uncomment 
http://pastebin.com/RH96QhYH

PM3 master with comment out.
http://pastebin.com/rFnTFfCb

marshmellow42, 2017-02-24:
Also strange is I have mingw32  and no issues.  Might be a version issue.  I'll get version #s when I'm next at my CPU.

iceman1001, 2017-02-24:
My mingw is the old dev-env (2012?) which has been upgraded a few times.
I belive this is not an issue if you download and install mingw fresh.  Not sure where on earth this replacement for intrin.h should be.

pwpiwi, 2017-02-24:
I was right: compiles and links fine without this include. It defines intrinsics for SSE, MMX and alike, which are not required here.

iceman1001, 2017-02-24:
Nop, compiles with a warning and links breaks as mentioned before missing **GetModuleHandleEx**
However I haven't tried a fresh making mingw-env,  which I will not make for this issue.

pwpiwi, 2017-02-24:
I meant: still compiles and links fine on my environment even without intrin.h

I.e. the real issue is that you are lacking GetModuleHandleEx(). In my environment it is defined in libloaderapi.h, which is included by winbase.h, which is included by windows.h, which is included by whereami.c. 

The set of WinAPI functions supported by mingw has improved over the last years.

marshmellow42, 2017-02-24:
works on mingw gcc 4.9.2
but fails on mingw gcc 4.4.0 (which is the old proxspace environment)

marshmellow42, 2017-02-24:
might be time to retire that old proxspace environment and publish a new one to encourage all to upgrade..

iceman1001, 2017-02-24:
...and why do you think I made a docker container?!?...

I'm testing to exclude the call to  GetModuleHandleEx and replace with a hardcoded call to GetModuleHandle instead...

iceman1001, 2017-02-24:
```
//  if (GetModuleHandleEx(GET_MODULE_HANDLE_EX_FLAG_FROM_ADDRESS | GET_MODULE_HANDLE_EX_FLAG_UNCHANGED_REFCOUNT, (LPCTSTR)WAI_RETURN_ADDRESS(), &module))
	module = GetModuleHandle("proxmark3.exe");
```

This hack compiles and when running client from "/" with "client/proxmark3" running scripts and listing them works.

pwpiwi, 2017-02-24:
Thanks. So GetModuleHandle() is available on old Proxspace? This should allow an easy fix because we don't need the extended functionality of GetModuleHandleEx().

iceman1001, 2017-02-24:
Yes, GetModuleHandle() works.   Feel free to make a nicer patch than my hardcoded.  Almost ready to close this one now.

pwpiwi, 2017-02-24:
Even more pragmatic: we don't need GetModulePath(), so we can comment out this part.

closed 2017-02-24 by iceman1001