Proxmark/proxmark3 #218: Segmentation Fault - Kali2 Rolling

kevthehermit, 2017-02-24:
New Proxmark3 V3
New Kali build

```
root@TechKali:~/proxmark3/client# uname -a
Linux TechKali 4.6.0-kali1-amd64 #1 SMP Debian 4.6.4-1kali1 (2016-07-21) x86_64 GNU/Linux
```
Follow the install guide for Kali. When i get to running the proxmark file i get a Segmentation Fault.

```
root@TechKali:~/proxmark3/client# ./flasher /dev/ttyACM0 ../armsrc/obj/fullimage.elf 
Loading ELF file '../armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002ab48->0x0002ab48) [R X] @0x94
1: V 0x00200000 P 0x0012cb48 (0x00001a34->0x00001a34) [RW ] @0x2abdc
Note: Extending previous segment from 0x2ab48 to 0x2c57c bytes

Waiting for Proxmark to appear on /dev/ttyACM0. Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to reappear on /dev/ttyACM0...... Found.

Flashing...
Writing segments for file: ../armsrc/obj/fullimage.elf
 0x00102000..0x0012e57b [0x2c57c / 355 blocks]................................................................................................................................................................................................................................................................................................................................................................... OK

Resetting hardware...
All done.

Have a nice day!
root@TechKali:~/proxmark3/client# ./proxmark3 /dev/ttyACM0 
Segmentation fault

```

kevthehermit, 2017-02-24:
Just noticed an error in the dmesg output

```
[ 1730.720068] usb 3-1: USB disconnect, device number 12
[ 1782.986059] usb 3-1: new full-speed USB device number 13 using xhci_hcd
[ 1783.710052] usb 3-1: new full-speed USB device number 14 using xhci_hcd
[ 1788.822161] usb 3-1: device descriptor read/64, error -110
[ 1789.055313] usb 3-1: New USB device found, idVendor=2d2d, idProduct=504d
[ 1789.055320] usb 3-1: New USB device strings: Mfr=1, Product=0, SerialNumber=0
[ 1789.055323] usb 3-1: Manufacturer: proxmark.org
[ 1789.055951] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
```

Axlbuld, 2017-02-25:
Same over here. Worked with the clone from mid of the week. Cloning and compiling today in Kali brings up the Same error.

pwpiwi, 2017-02-26:
What is the latest commit working for you?

Axlbuld, 2017-02-26:
Hi,
the last time I pulled a clone was an February 22nd at about 8 PM German time. That was working. There were two commits that day. 23rd I did not try.

Axlbuld, 2017-02-26:
Just pulled the files of the commit 4c16ae80f0bffae2092e208c46a5005356c24ce1 - should be the last from 23rd (if I did not make a mistake - clicked on the dot and downloaded as ZIP) - complied and works.

pwpiwi, 2017-02-26:
Any compiler and/or linker warnings/errors?

Axlbuld, 2017-02-26:
Not that I am aware of.

Just pulled the last one on Feb. 23 = 4197a3f6ff9420673cc2eeaeb3eeedeb4a9c0dc9. That has issues: can not compile. Looks the code cleanup was ok and the issues came on later.

I get compiler errors on that version:

whereami.c: In function ‘wai_getExecutablePath’:
whereami.c:179:15: error: ‘PATH_MAX’ undeclared (first use in this function)
   char buffer[PATH_MAX];
               ^~~~~~~~
whereami.c:179:15: note: each undeclared identifier is reported only once for each function it appears in
whereami.c:185:16: warning: implicit declaration of function ‘realpath’ [-Wimplicit-function-declaration]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
                ^~~~~~~~
whereami.c:179:8: warning: unused variable ‘buffer’ [-Wunused-variable]
   char buffer[PATH_MAX];
        ^~~~~~
whereami.c: In function ‘wai_getModulePath’:
whereami.c:244:19: error: ‘PATH_MAX’ undeclared (first use in this function)
       char buffer[PATH_MAX < 1024 ? 1024 : PATH_MAX];
                   ^~~~~~~~
whereami.c:249:12: warning: unused variable ‘path’ [-Wunused-variable]
       char path[PATH_MAX];
            ^~~~
whereami.c:244:12: warning: unused variable ‘buffer’ [-Wunused-variable]
       char buffer[PATH_MAX < 1024 ? 1024 : PATH_MAX];
            ^~~~~~
Makefile:158: recipe for target 'obj/whereami.o' failed
make[1]: *** [obj/whereami.o] Error 1
make[1]: Leaving directory '/root/proxmark/proxmark3-4197a3f6ff9420673cc2eeaeb3eeedeb4a9c0dc9/client'
Makefile:12: recipe for target 'client/all' failed
make: *** [client/all] Error 2

Axlbuld, 2017-02-26:
Compiling the latest version I get some warnings:
blua -Wall  -g -O4 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/whereami.o whereami.c
whereami.c: In function ‘wai_getExecutablePath’:
whereami.c:189:16: warning: implicit declaration of function ‘realpath’ [-Wimplicit-function-declaration]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
                ^~~~~~~~
whereami.c:189:14: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
              ^
whereami.c: In function ‘wai_getModulePath’:
whereami.c:266:20: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
           resolved = realpath(path, buffer);
                    ^

But i complies without errors.

The images work (installed and tested with older version), but the programm on the computer brings up the Segmentation fault.
System is:
Linux esrnb03154 4.9.0-kali2-amd64 #1 SMP Debian 4.9.10-1kali1 (2017-02-20) x86_64 GNU/Linux

pwpiwi, 2017-02-26:
This is most probably the issue. Please try #define _DEFAULT_SOURCE in line 166 in client/whereami.c (before stdlib.h is included).

Axlbuld, 2017-02-26:
changed and testing on the current version.

Warnings are still there:
 -o obj/whereami.o whereami.c
whereami.c: In function ‘wai_getExecutablePath’:
whereami.c:189:16: warning: implicit declaration of function ‘realpath’ [-Wimplicit-function-declaration]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
                ^~~~~~~~
whereami.c:189:14: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     resolved = realpath(WAI_PROC_SELF_EXE, buffer);
              ^
whereami.c: In function ‘wai_getModulePath’:
whereami.c:266:20: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
           resolved = realpath(path, buffer);

Complied.
I still get the Segmentation fault trying to start the programm :( - ./proxmark3 /dev/ttyACM0.

The help, however (./proxmark3 -h) runs.

Axlbuld, 2017-02-26:
Heres is a part from messages:
Feb 26 13:52:19 esrnb03154 kernel: [ 5485.687166] proxmark3[6235]: segfault at ffffffffad42c6e0 ip 00007eff4aed4486 sp 00007ffead42c6d8 error 5 in libc-2.24.so[7eff4ae54000+195000]
Feb 26 13:52:46 esrnb03154 kernel: [ 5511.952063] proxmark3[6240]: segfault at ffffffff88263d30 ip 00007facea375486 sp 00007ffd88263d28 error 5 in libc-2.24.so[7facea2f5000+195000]

Axlbuld, 2017-02-26:
That the part aroung line 166 in whereami.c:

  return length;
}
 */

#elif defined(__linux__)
#define _DEFAULT_SOURCE
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
// #include <limits.h>     // not all linux distri ...

Have to leave for couple of houres now. Check later. Thank you very much for the help.

pwpiwi, 2017-02-26:
OK, I have downloaded and installed Kali-Linux to review this issue. 

The define needs to come before the very first include. It works for me when I add this define before line 13. Please check.

Axlbuld, 2017-02-26:
Works. Have included the command in line 12. Now complies without issue and works. I have connection to the proxmark with the tool again.

Thank you very much for your help.

Axlbuld, 2017-02-26:
root@esrnb03154:~/proxmark/proxmark3/client# ./proxmark3 /dev/ttyACM0
Prox/RFID mark3 RFID instrument          
bootrom: master/v2.2.0-310-g5a6a7af-suspect 2017-02-25 19:28:45
[...]

Axlbuld, 2017-02-26:
Well, one little issue now. proxmark can't find the scripts directory:

cannot open /root/proxmark/proxmark3/client/�����Uscripts/legic.lua: No such file or directory

Somehow it seems some strange characters are included ...

pwpiwi, 2017-02-26:
Well, legic.lua doesn't exist. But nevertheless, I don't get those funny characters:

```
proxmark3> script run legic
--- Executing: legic.lua, args ''
cannot open /root/proxmark3/client/scripts/legic.lua: No such file or directory

-----Finished
proxmark3> 
```

With an existing lua file it works as expected:
```
proxmark3> script run htmldump
--- Executing: htmldump.lua, args ''
Wrote a HTML dump to the file 2017-02-26_093214.html

-----Finished
proxmark3> 
```

What do you get with existing lua files?

Axlbuld, 2017-02-26:
Well, I copied a version of legic.lua there ... ok, I am not programmer and might be sort of a messy user. Most likely I messed up something today.

Independent of what I am doing, I get strange characters. Don't know why. Never happened before. Even listing or just "run" give these characters. Seems, 5 char are taken from random ...

Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> script list
Couldn't open the scripts-directory          
proxmark3> script run
--- Executing: .lua, args ''
cannot open /root/proxmark/proxmark3/client/�&F��Uscripts/.lua: No such file or directory

-----Finished
proxmark3> script run test.lua
--- Executing: test.lua, args ''
cannot open /root/proxmark/proxmark3/client/�&F��Uscripts/test.lua: No such file or directory

-----Finished
proxmark3>

Axlbuld, 2017-02-26:
proxmark3> script run htmldump.lua
--- Executing: htmldump.lua, args ''
cannot open /root/proxmark/proxmark3/client/�&F��Uscripts/htmldump.lua: No such file or directory

-----Finished

Axlbuld, 2017-02-26:
I probably clone again when the current changes have been committed. Might be I messed some code up (not knowing what and where, but might have happend). I was using gedit to make the changes.

Axlbuld, 2017-02-26:
Tested with old version - nothing to do the my computer. It works there.
Cloned a new version - does not work. However, coding for the scripts-part has been changed.

@iceman1001 Line 163 seems to be a bit late, right?

pwpiwi, 2017-02-26:
@kevthehermit or any other Kali user: does it work for you with the change described above (add `#define _DEFAULT_SOURCE` in line 12 of client/whereami.c)?

kevthehermit, 2017-02-26:
Will be home from the weekend trip in a couple of hours. Will check then.

iceman1001, 2017-02-26:
Row 12,  works (still) on mingw, ubuntu.

Axlbuld, 2017-02-26:
Done for today. Somehow, get_my_executable_directory() already gives back the additional 5 to 6 characters. For test, I changed the part in cmdscripts.c to a manual code "    dp = opendir ("./scripts/");". At least it works for the time being. I have to reanimate my coding knowledge (died years ago ...).

Thanks a lot, there was a lot to learn today.

exploitagency, 2017-02-27:
> @kevthehermit or any other Kali user: does it work for you with the change described above (add #define _DEFAULT_SOURCE in line 12 of client/whereami.c)?

@pwpiwi this works for my setup

>Linux kali 4.9.0-kali1-amd64 #1 SMP Debian 4.9.6-3kali2 (2017-01-30) x86_64 GNU/Linux

closed 2017-03-06 by pwpiwi