Proxmark/proxmark3 #222: #db# unknown command:: 0x0399 error when ever I try and read from a iclass tag

thesle3p, 2017-03-04:
So when ever I try and read from a Iclass tag with the proxmark I get a ```#db# unknown command:: 0x0399``` error, tried with multiple cards and same issue, I can pull the CSN off the card using ``` hf iclass readblk b 0A k <key>```  but again get the ```#db# unknown command:: 0x0399 ``` error. Any ideas as to whats going on here?
```
 hf iclass dump k <key>
Authing with diversified key: <key>          
#db# unknown command:: 0x0399          
Waiting for a response from the proxmark...          
Don't forget to cancel its operation first by pressing on the button          
Auth Command execute timeout          
Authing with diversified key: <key>          
#db# unknown command:: 0x0399          
Waiting for a response from the proxmark...          
Don't forget to cancel its operation first by pressing on the button          
Auth Command execute timeout   
```

iceman1001, 2017-03-04:
and this would be because you broken the rules...  Never mix client / fullimage.elf from different forks, branches...   Only use the same client with the same fullimage from one compilation.

marshmellow42, 2017-03-06:
iceman is correct, you're arm/device firmware has not been flashed with the newer commands that your client is trying to use...  follow the proper flashing procedures and never run mismatched client and device code.

thesle3p, 2017-03-06:
Thanks that fixed it.

closed 2017-03-06 by thesle3p