Proxmark/proxmark3 #238: Install issue

lewis-elliott, 2017-03-22:
Steps carried out:

`brew tap proxmark/proxmark3`

`brew install proxmark3`

result:

```
Last 15 lines from /Users/Lewis/Library/Logs/Homebrew/proxmark3/02.make:
                 from /usr/local/Cellar/arm-none-eabi-gcc/20160926/arm-none-eabi/include/stdio.h:61,
                 from ../include/hitagS.h:13,
                 from apps.h:19,
                 from start.c:13:
/usr/local/Cellar/arm-none-eabi-gcc/20160926/arm-none-eabi/include/sys/_stdint.h:48:20: error: conflicting types for 'uint32_t'
 typedef __uint32_t uint32_t ;
                    ^
In file included from ../include/usb_cmd.h:21:0,
                 from ../include/proxmark3.h:17,
                 from start.c:12:
./stdint.h:21:24: note: previous declaration of 'uint32_t' was here
 typedef unsigned int   uint32_t;
                        ^
make[1]: *** [obj/start.o] Error 1
make: *** [armsrc/all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/proxmark/homebrew-proxmark3/issues
```

Second attempt:

`brew install --HEAD proxmark3`

result:

```
Last 15 lines from /Users/Lewis/Library/Logs/Homebrew/proxmark3/02.make:
gcc -o lua   lua.o liblua.a -lm -lreadline 
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_USE_MACOSX    -c -o luac.o luac.c
gcc -o luac   luac.o liblua.a -lm -lreadline 
gcc -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../tools -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/proxmark3.o proxmark3.c
gcc -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../tools -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/uart.o uart.c
gcc -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../tools -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/util.o util.c
util.c:44:27: error: use of undeclared identifier 'CMIN'
  Ntty.c_cc[VMIN]       = CMIN;    /* minimum time to wait      */
                          ^
util.c:45:27: error: use of undeclared identifier 'CTIME'
  Ntty.c_cc[VTIME]      = CTIME;   /* minimum characters to wait for */
                          ^
2 errors generated.
make[1]: *** [obj/util.o] Error 1
make: *** [client/all] Error 2
```

Latest version of OSX is installed and a fresh install of Hombrew was also installed.

Any ideas?

pwpiwi, 2017-03-22:
I don't have access to OS X so I am flying blind here. Try PR #241.

pwpiwi, 2017-03-23:
Should be fixed with commit 9b952b2.

And yes, you should use --HEAD. Otherwise brew will fetch the April 2016 release.

pwpiwi, 2017-03-29:
@lewis-elliott: Can this be closed?

closed 2017-03-29 by lewis-elliott