Proxmark/proxmark3 #247: mfkey make failes

iceman1001, 2017-03-25:
**System: 
Ubuntu 16.04
gcc v5.4
latest source from github**

> user@ubuntu:~/pm3_master/tools/mfkey$ make
> gcc -I../../common -I../../client -Wall -O4  -o mfkey32 crypto1.o crapto1.o util.o mfkey.o mfkey32.c
> In file included from /usr/include/i386-linux-gnu/bits/waitstatus.h:64:0,
>                  from /usr/include/stdlib.h:42,
>                  from mfkey32.c:4:
> ../../client/util.h:75:34: error: expected ‘)’ before ‘*’ token
>  extern uint32_t le32toh (uint8_t *data);
>                                   ^
> Makefile:17: recipe for target 'mfkey32' failed
> make: *** [mfkey32] Error 1
> 

> user@ubuntu:~/pm3_master/tools/mfkey$ gcc -v
> Using built-in specs.
> COLLECT_GCC=gcc
> COLLECT_LTO_WRAPPER=/usr/lib/gcc/i686-linux-gnu/5/lto-wrapper
> Target: i686-linux-gnu
> Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.4' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-i386/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-i386 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-i386 --with-arch-directory=i386 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-targets=all --enable-multiarch --disable-werror --with-arch-32=i686 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=i686-linux-gnu --host=i686-linux-gnu --target=i686-linux-gnu
> Thread model: posix
> gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 
>

pwpiwi, 2017-03-25:
That's an interesting one. Compiles fine on Kali and mingw. Could you please try to change
`extern uint32_t le32toh (uint8_t *data);` to `extern uint32_t le32toh(uint8_t *data);` (i.e. remove the space before the bracket).

iceman1001, 2017-03-26:
**user@ubuntu:~/pm3_master/tools/mfkey$ make all
gcc -I../../common -I../../client -Wall -O4  -o mfkey32 crypto1.o crapto1.o util.o mfkey.o mfkey32.c
In file included from /usr/include/i386-linux-gnu/bits/waitstatus.h:64:0,
                 from /usr/include/stdlib.h:42,
                 from mfkey32.c:4:
../../client/util.h:75:33: error: expected ‘)’ before ‘*’ token
 extern uint32_t le32toh(uint8_t *data);
                                 ^
Makefile:17: recipe for target 'mfkey32' failed
make: *** [mfkey32] Error 1
**

pwpiwi, 2017-03-26:
I should read more carefully.
There was a clash between the predefined le32toh and <endian.h> and the self defined le32toh in "util.h"

closed 2017-03-26 by iceman1001

iceman1001, 2017-03-26:
Maybe remove one in util.h,  to use the predefined instead?

pwpiwi, 2017-03-26:
Definitely to be considered. But if adding -std=c99 fixes the issue, the predefinition seems not to conform to C99.

iceman1001, 2017-03-26:
I have too little understanding in what c99 does when specified to gcc to comment further. 
Shouldn't  basic stuff  le32toh exist in c99 somewhere?

pwpiwi, 2017-03-26:
According to http://man7.org/linux/man-pages/man3/endian.3.html
>  These functions are nonstandard.